esp8266/Arduino #4663: Including <iostream> gives misleading error

mje-nz, 2018-04-21:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				N/A
- Core Version:			c12c9e21
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:              Generic ESP8266 Module
- Flash Mode:	      qio
- Flash Size: 	      1MB
- lwip Variant:       v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		serial
- Upload Speed:            115200

### Problem Description
Compiling a sketch with `#include <iostream>` for ESP8266 gives a long, irrelevant error. Googling the error finds people who have their build system set up incorrectly, which doesn't help. I discovered this after accidentally leaving some debugging code in a hardware-agnostic utility class.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>
#include <iostream>

void setup() {}
void loop() {}
```

### Compile error
```
Archiving built core (caching) in: /var/folders/9v/4bdlzlv544d3gn87rpg4kdd00000gn/T/arduino_cache_272822/core/core_esp8266_esp8266_generic_CpuFrequency_80,ResetMethod_ck,CrystalFreq_26,FlashFreq_40,FlashMode_qio,FlashSize_1M0,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,UploadSpeed_115200_f268ef1bf7b434efe7be82ffaf83fd6d.a
arduino.ar(libc_replacements.c.o): In function `_putc_r':
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/cores/esp8266/libc_replacements.c:96: multiple definition of `_putc_r'
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/libc/xtensa-lx106-elf/lib/libc.a(lib_a-putc.o):/Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/putc.c:97: first defined here
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/lib/libstdc++.a(locale-inst.o):(.literal._ZSt9use_facetISt5ctypeIcEERKT_RKSt6locale[std::ctype<char> const& std::use_facet<std::ctype<char> >(std::locale const&)]+0x4): undefined reference to `std::__throw_bad_cast()'
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/lib/libstdc++.a(locale-inst.o): In function `std::ctype<char> const& std::use_facet<std::ctype<char> >(std::locale const&)':
/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/locale_classes.tcc:114: undefined reference to `std::__throw_bad_cast()'
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/lib/libstdc++.a(locale-inst.o): In function `std::numpunct<char> const& std::use_facet<std::numpunct<char> >(std::locale const&)':
/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/locale_classes.tcc:114: undefined reference to `std::__throw_bad_cast()'
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/lib/libstdc++.a(locale-inst.o): In function `std::num_put<char, std::ostreambuf_iterator<char, std::char_traits<char> > > const& std::use_facet<std::num_put<char, std::ostreambuf_iterator<char, std::char_traits<char> > > >(std::locale const&)':
/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/locale_classes.tcc:114: undefined reference to `std::__throw_bad_cast()'
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/lib/libstdc++.a(locale-inst.o): In function `std::num_get<char, std::istreambuf_iterator<char, std::char_traits<char> > > const& std::use_facet<std::num_get<char, std::istreambuf_iterator<char, std::char_traits<char> > > >(std::locale const&)':
/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/locale_classes.tcc:114: undefined reference to `std::__throw_bad_cast()'
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/lib/libstdc++.a(locale-inst.o):/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/locale_classes.tcc:114: more undefined references to `std::__throw_bad_cast()' follow
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/lib/libstdc++.a(ios-inst.o):(.literal._ZNSt9basic_iosIcSt11char_traitsIcEE5clearESt12_Ios_Iostate[std::basic_ios<char, std::char_traits<char> >::clear(std::_Ios_Iostate)]+0x4): undefined reference to `std::__throw_ios_failure(char const*)'
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/lib/libstdc++.a(ios-inst.o): In function `std::basic_ios<char, std::char_traits<char> >::clear(std::_Ios_Iostate)':
/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_ios.tcc:122: undefined reference to `std::__throw_ios_failure(char const*)'
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/lib/libstdc++.a(ios-inst.o): In function `std::basic_ios<wchar_t, std::char_traits<wchar_t> >::clear(std::_Ios_Iostate)':
/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_ios.tcc:122: undefined reference to `std::__throw_ios_failure(char const*)'
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/lib/libstdc++.a(c++locale.o):(.literal._ZNSt6locale5facet18_S_create_c_localeERPiPKcS1_+0x4): undefined reference to `std::__throw_runtime_error(char const*)'
/Users/Matthew/Library/Arduino15/packages/esp8266/hardware/esp8266/master/tools/sdk/lib/libstdc++.a(c++locale.o): In function `std::locale::facet::_S_create_c_locale(int*&, char const*, int*)':
/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/src/c++98/c++locale.cc:207: undefined reference to `std::__throw_runtime_error(char const*)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.
```

For comparison, here is the error when compiled for Arduino Uno:

```
/Users/Matthew/Documents/Arduino/sketch_apr21a/sketch_apr21a.ino:2:20: fatal error: iostream: No such file or directory
 #include <iostream>
                    ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Uno.
```

devyte, 2018-04-22:
The error you see is not irrelevant, it explains what happens: the linker can't find the implementation for exceptions (among other things), and that is because exceptions are currently not supported on the ESP. However, the iostream.h file is in fact present. On the Uno, it seems that the iostream.h header is not present.
AFAIK, stdlibc++ support is not complete here, and iostream is among the parts not supported. I could be wrong, of course.
@igrr is my previous statement correct?

igrr, 2018-04-22:
Yes, you are right. I'll take a look re: missing symbols.

bestpika, 2018-07-10:
I got same error.
```
#include <iostream>
#include <string>
#include <regex>

void setup() {
  std::regex pattern("^GET ([^ ]+) HTTP.*$");
}

void loop() {

}
```

shineangelic, 2018-07-11:
same here with this sketch and libversion 2.4.1 on Arduino IDE

undefined reference to `std::__throw_bad_cast()'

with test sketch:
```
#include <sstream>
#include <Arduino.h>

std::stringstream s;

void setup()
{
    Serial.begin(9600);
    Serial.println("Test");
}

void loop()
{

}
```

shineangelic, 2018-07-11:
probably related/duplicate of #3358

igrr, 2018-08-14:
As mentioned above, this is combination of https://github.com/esp8266/Arduino/issues/3358 and https://github.com/esp8266/Arduino/issues/4630. Should be fixed via https://github.com/esp8266/Arduino/pull/5047. Closing as duplicate, please follow these issue links and click "Subscribe" to receive further updates.

closed 2018-08-14 by igrr