esp8266/Arduino #4665: Can i in any way shape or form get to the ARP table?

DasVoelkel, 2018-04-21:
I am trying to make a detector which can see when someone enters the wifi network and when (the device) leaves the network. 
First bit of research landed me a beacon sniffer.
https://github.com/kalanda/esp8266-sniffer 
Which is great in itself but doesn't help my issue since when going into flight mode or switching wifi off is undetectable. Also i can only detect the initial beacon from the phone which lets it receive wifi.

next up was pinging broadcast, but that is impossible (almost) since most phones (apple 100%) eat the ping command and don'T respond to it. OK that's not gonna work either. 

Next i tried the ARP commands. Saw that in principle the Packats can be UDP or TCP, they just have to have a specific port. So after opening udp port and rebuilding a package (from wireshark arp protocoll tests) i tried it and the overhead from the udp functions (which i suspected) made the package not recognizable in that state. OK, that already took ages.

Up next, try sending raw packages in a function with no overhead. Somehow found wifi_send_pkt_freedom in the SDK and after getting that to the normal arduino environment tried it and it just didn't respond at all. But that was to be expected, i'Ve never played with the sdk and know almost nothing about it. 

I've also looked at this:
https://github.com/spacehuhn/esp8266_deauther
and tried to understand how he sends packages. But the wifi_send_pk_freedom command seems to send it somewhere i don't know where ? And if a command like that ( so low layer ) exsists. is there one i just can'T see for the arp protocol ?

As i am already on the 3rd page of google and very near the end of hope, i wanted to ask.
Is there any way i can access the arp table or anything of the kind or the arp protocoll or the rarp protocoll frames (or just send raw frames) in the esp8266 ? 
From what i've seen systems don't eat up arp protocol requests as they are too important. 
Maybe there is another solution i just can't think of tight now? 

any help would be awesome and GREATLY appreciated as i'Ve already spend about 2 weeks on this issue i really want to make it work finally.

devyte, 2018-04-21:
I suggest you look into the Espressif api reference and other docs. There is a promiscuous mode that may help you, and I believe it is possible to hook into the interface between the SDK and lwip to capture and send raw packeets, but I don't know how.
This is not the right place to ask for help for your particular project. Closing per issue POLICY doc.

closed 2018-04-21 by devyte

DasVoelkel, 2018-04-21:
I've been searching for this kind of feature or documentation for a long time now and don't really know where to look or start. I've gone through the documentation and the library itself but didn't find what I was looking for, mainly because i don't know how something like that is phrased or where it's usually found.

But my main question, and that's not really specific for this project alone, can the Arduino library of the esp8266 send/receive raw packages you build yourself, or not?