esp8266/Arduino #4667: Will the ESP8266 respond to arp requests?

DasVoelkel, 2018-04-22:
Does the microcontroller respond to broadcast arp requests?
As in, if the esp tries to connect with a local address (let's say http of my gateway), and it succeeds.
If afterwards another user in the network tried to connect with that same ip, and they first send out an arp rerquest, will the esp answer? it should still have the address in the routing table and should theoretically send a response arp package, but does it?

I've searched in the docs and the references and in the library itself. Gone down to raw.h but really am not sure what i am looking at anymore. the documentation for the different structs and the explanation for some of them seems to fall short for this purpose. Or am i missing something completely different?
As i see it, the WiFi Classes (WifiClient, UDP / TCP ) can't send raw packages without adding something in addition to the data.

d-a-v, 2018-04-22:
You will find your answers in lwIP's reference.
You can also try to send requests, and use tcpdump on your PC to track the answers.
ESP does respond to ARP, but I don't think it will respond to RARP (as per lwIP sources).

devyte, 2018-04-23:
Not an issue => closing.

closed 2018-04-23 by devyte

DasVoelkel, 2018-04-23:
Where in the reference? the lwip v2 standard is huge and the implementation isn't 1:1. I don't really know what to look for. the only thing i've kinda found about was the raw_sendto command, which needs structs that are somewhere defined which are defined by other defines. 
e.g. what is a raw_pcb supposed to be? I only know a pcb in the sense of the physical board.

For a wifi module it's awkwardly unexplained how to just use the ipv4 standard to just send a package with data through an interface. Or to get to the interface that gets defined by itself by WiFi.begin(). 

I think it is an issue, just on transparency.

d-a-v, 2018-04-23:
From where I stand, I find it difficult to understand your concern.
There are standards which define things, and one need to know them to fit with standards. Reading examples then docs help learning and understanding.

You may try to explain in a more simple way what you are trying to achieve.
You may also try to achieve this using a linux setup for which you have a lot more accessible documentation and tools, then we can better understand and help you with this arduino environment.

This Wifi module's core library is designed to work and comply with arduino standards. There is indeed no easily accessible API to send or receive raw packets, at least not yet (#2330 has an example but only working with lwIP-v1.4).

If you stop your search at the lwIP's PCB not beeing a printed circuit board, that's bad spirit and your issues will be closed.