esp8266/Arduino #4668: SPISlave example not working

itexpertshire, 2018-04-23:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 lower]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

ESP Slave demo code not displaying data received from Master UNO.

Used below connection diagram for setting up UNO as master and nodemcu as Slave.
GPIO    NodeMCU   Name  |   Uno
   ===================================
     15       D8       SS   |   D10
     13       D7      MOSI  |   D11
     12       D6      MISO  |   D12
     14       D5      SCK   |   D13

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
Compiled and uploaded below code into UNO
https://github.com/esp8266/Arduino/tree/master/libraries/SPISlave/examples/SPISlave_SafeMaster

Compiled and uploaded below code into NodeMCU

https://github.com/esp8266/Arduino/tree/master/libraries/SPISlave/examples/SPISlave_Test

### Debug Messages

Debug info received from UNO serial monitor
[
Master: Are you alive?
Slave: 

Master: Are you alive?
Slave: 
]

Debug ino displayed in NodeMCU serial monitor,
[
connected with ASUS-2.4g, channel 2
dhcp client start...
ip:192.168.2.251,mask:255.255.255.0,gw:192.168.2.1
pm open,type:2 0
]

konosubakonoakua, 2019-12-30:
Maybe you need a wire to connent **GND** pins of the two boards.
A good **common ground** is important.

VakarisZ, 2020-01-19:
I'm also getting similar error, except on master end I get `Slave: ?????????????` and on ESP8266 I get the correct message `Are you alive?` once and after that I only get bogus messages. All my code is standard code (SPISlave_test on ESP8266-12 and SPISlave_testMaster on arduino yun).