esp8266/Arduino #4670: Creating HTTP Server: when the client answer slowly ESP-12E crashes and doesn't send data

calfizzi, 2018-04-23:
**Basic Infos**
[X ] This issue complies with the issue POLICY doc.
[ X] I have read the documentation at readthedocs and the issue is not addressed there.
[ X] I have tested that the issue is present in current master branch (aka latest git).
[ X] I have searched the issue tracker for a similar issue.
[ X] If there is a stack dump, I have decoded it.
[ X] I have filled out all fields below.

**Platform**
Hardware:	[ESP-12]
Core Version:	[latest git hash]
Development Env:	[Arduino IDE]
Operating System:	[Windows]

**Settings in IDE**
Module: [Nodemcu]
Flash Mode:	[qio]
Flash Size: [4MB]
lwip Variant: [v2 lower]
Reset Method:	[ck|nodemcu]
Flash Frequency:	[40Mhz]
CPU Frequency:	[80Mhz]
Upload Using:	[SERIAL]
Upload Speed: [115200] (serial upload only)

**Problem Description**
I tried to build an example for you of my problem.
I built a simple program that answer via HTTP showing the "Romeo And Juliet" romanze as web page. If you compile and run the code on your ESP-12E you can easily connect to its the web server on port 80. It will response the complete text of the R&J romanze as HTML page. But If you try to connect by a slow dispositive like a 3G mobile, while data transfer the wifi connection of ESP-12E go down. Some time it blocks as it runing in a  infinite loop. I also created a simple C# client to simulate a internet slow connection and the result is the same

**my example Sketch:**
[TEST_BUG_ESP8266.zip](https://github.com/esp8266/Arduino/files/1937686/TEST_BUG_ESP8266.zip)

**Simple C# Project:**
[HttpClient.zip](https://github.com/esp8266/Arduino/files/1937698/HttpClient.zip)

Please let me know if I'm doing something wrong or if the issue really exist!

Best Regards,
Mario Calfizzi

calfizzi, 2018-04-24:
Yes, I know the problem could be related to the send buffer for this reason I already tried what you suggested but it doesn't solve the problem. Sure, it reduce the effects of the problem but unfortunally it isn't the solution. For this reason I submitted de issue to the develop team.
After some analysis I suppose something in TCP communication doesn't work well because the ESP library continues to send data, filling the internal buffer, without waiting the client feedback. 
I'm not an expert in TCP protocol but I hope that exists a feedback during communication it can inform the server if the data packets are or not received by the client!
If this feedback exists could mean the ESP library isn't using it correctly.

earlephilhower, 2019-08-21:
Closing due to the fact that this looks like the same issue as was seen with the pre-SDK version, which was removed from the tree after discovering it would freeze for 100s of milliseconds, randomly.

If GIT head has the same problem, please open another bug so we can track it.

closed 2019-08-21 by earlephilhower