esp8266/Arduino #4672: not doing :ur 1, WS:dis and :del after :rcl and :abort

mamama1, 2018-04-23:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.4.1]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu|]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|Higher Bandwidth] <- doesn't matter which one of those two i use
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL] <-- doesn't matter
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I've got a pretty big (around 4k lines) software written for the ESP8266 (WiFi LED dimmer) which seems to compile differently between Arduino IDE and PIO.
The firmware has two TCP servers. One is a webserver and based on the ESP8266WebServer.h and the second one is a simple tcp server which listens to incoming bytes and is based on the ESP8266WiFi.h (class WiFiServer and WiFiClient).

When compiling the software using the Arduino IDE, everything works as expected. When compiling the same software with PIO, the tcp server which is based on ESP8266WiFi.h stops working after serving one request. The webserver and the rest of the devices continues working. However, also Wifi connection issues occur, like SoftAP not working (SSID doesn't show up).

I have reproduced this many times, it occurs using any ESP8266 core version beginning with 2.2.0 when I use PIO and does NOT occur on any Arduino compiled ESP8266 core, also beginning with 2.2.0 including 2.4.1

**What I have tracked down so far:**
i have enabled all debugging and it seems that when compiling with arduino and all is working, output looks like this:

```
WS:ac
:ref 1
WS:av
:ref 2
:ur 2
:ref 2
:ur 2
:rn 18
wlan received cmd type 10, cmd 97
:wr 1 1 0
:wrc 1 1 0
:wr 1 1 0
:wrc 1 1 0
:sent 1
:sent 1
:rcl
:abort
:ur 1
WS:dis
:del
WS:ac
:ref 1
WS:av
:ref 2
:ur 2
:ref 2
```
... and so on. everything fine.

when compiling using platformio, the 

```
:ur 1
WS:dis
:del
```

is missing, and the TCP server stops responding. the remaining firmware keeps running (including the http webserver) without any issues.

so when compiling using platformio, the output looks like this and it stops after that:
```
WS:ac
:rn 18
:ref 1
WS:av
:ref 2
:ur 2
:ref 2
:ur 2
wlan received cmd type 10, cmd 97
writing TCP_SC_BRIGHTNESS_STEP_GET...
:wr 1 1 0
:wrc 1 1 0
:wr 1 1 0
:wrc 1 1 0
writing TCP_SC_BRIGHTNESS_STEP_GET done.
:sent 1
:sent 1
:rcl
:abort
WS:ac
:rn 18
```

if i file another request towards the tcp server, i only get this (and i can repeat it as often as i want):
```
WS:ac
:rn 18
```

so it seems that somehow the tcp server socket is not closed, not deleted, or whatever - i do not exactly know what WS:dis and :del means or does but according to ClientContext.h, :del seems to do a cleanup jobby:

```
    void unref()
    {
        DEBUGV(":ur %d\r\n", _refcnt);
        if(--_refcnt == 0) {
            discard_received();
            close();
            if(_discard_cb) {
                _discard_cb(_discard_cb_arg, this);
            }
            DEBUGV(":del\r\n");
            delete this;
        }
    }
```

I'd appreciate any help or suggestion how to fix this. I am not sure if this is even related to the arduino core or to platformio, that's why i created an issue also with platformio: https://github.com/platformio/platform-espressif8266/issues/83

Thanks!

devyte, 2018-04-25:
@ivankravets Looking at the other referenced issue, I see you are already looking at it there. Should I leave this one open (I.e.: is it possible something in this repo needs fixing), or should I just close it?

ivankravets, 2018-04-25:
Duplicate of https://github.com/platformio/platform-espressif8266/issues/83

closed 2018-04-25 by ivankravets

mamama1, 2018-04-25:
hayo

i am/was not sure where the issue comes from and who would/should be able to fix this, also I don't know who collaborates on which projects. that's why i opened the issue in both repos.

however it seems that this is platformio related, as this issue has been closed?

hope that this will get fixed soon, love the esp8266 and platformio but unfortunately my skills aren't high enough to fully understand/debug/fix what is going on here.

thanks