esp8266/Arduino #4673: Problem using Client Secure with two hosts (CACert method)

Mark8869, 2018-04-24:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E]
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Hello,
I'm having a problem with the "client secure" library (CACert method).
In practice, in my sketch I have to use the secure client to communicate with two different hosts, so I have to use two different root certificates.
I have compiled the sketch as attached, but after 4 loops I get an error on the serial monitor.
I was able to understand that the cause of the problem is the command client.setCACert_P, but more than that I can not understand.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <WiFiClientSecure.h>

WiFiClientSecure client;

[... 192 lines cut ...]
3fff0810:  3fff0850 3fff1664 40204e2c 3fff7034  
3fff0820:  3fff701c 00003a98 00003a98 4020136f  
3fff0830:  3ffef5c8 00000000 3fff6bf4 3ffef5c8  
3fff0840:  3ffe851c 3fff271c 3fff15f4 402157d7  
3fff0850:  00000001 3ffef5c8 3ffe8e78 3ffef5c8  
3fff0860:  000001bb 3ffef5c8 3fff15f4 40203990  
3fff0870:  3ffe8e78 56c5cc12 40107180 56c5cc12  
3fff0880:  3ffe851c 3ffef5c8 3ffef8cc 402022d5  
3fff0890:  00000010 3fff08f0 3fff08f0 4020498f  
3fff08a0:  3ffe8d38 00000004 3fff08f0 402049db  
3fff08b0:  3ffeef80 5adf08d8 3fff08f0 40204a0d  
3fff08c0:  3fff08f0 00000000 3fff08f0 40204a5a  
3fff08d0:  3ffe8d38 00001388 00001388 3ffef8f8  
3fff08e0:  3fffdad0 00000000 3ffef8f0 4020240e  
3fff08f0:  3fff1dbc 0000000f 00000004 40204e4d  
3fff0900:  3fffdad0 00000000 3ffef8f0 40204e78  
3fff0910:  feefeffe feefeffe 3ffef900 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Mark8869, 2018-04-25:
UPDATE:
I have tried to monitor free heap size, and as I was suspecting, every loop it decreases of 256 byte, so I think that this the cause of fatal exception 29.
But... I'm not able to understand why happens this, and above all how the problem can be fixed.

Thank you in advance all those who want to get interested.

devyte, 2018-04-28:
I understand that there was a memleak fixed with WiFiCLient|Secure after 2.4.1 release, which could be relevant. 
You have marked in the issue checklist above that you tested with latest master branch (aka latest git), and that the issue is still present. Is that really true?

Mark8869, 2018-04-28:
I'm using latest release 2.4.1.

devyte, 2018-05-29:
BearSSL is merged in #4273 , with alternate BearSSL::WiFi* classes. Although axtls-based classes are still available and even the default, they are planned for deprecation and then retirement, hence won't be fixed. Any issues with BearSSL-based classes should be reported in new issues.
Also, there have been several bug fixes related to connection stability merged post 2.4.1. There is a reason why the issue template has a checkbox that says:
> I have tested that the issue is present in current master branch (aka latest git).

which you have ticked.
Closing.

closed 2018-05-29 by devyte