esp8266/Arduino #4682: promicous mode and wifi class , intended behaviour?

DasVoelkel, 2018-04-25:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.    <- my core code is like 12 files.

#### Platform

- Hardware:					[ESP8285 device
- Core Version:			[atest git hash 
- Development Env:		Arduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			?
- Flash Size: 		4MB
- lwip Variant:         v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	?
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

Working on a node MCU E-12 .

I've been trying out promiscuous mode, as from the mini sniffer or spacehuhns project. I tried if it was possible to connect wifi while using this mode.
(after disabling promicous_mode, setting a new callback, and then reenabling it)
WiFi.status() kept responding 0 although that the connection didn't seem to be successful -> because, client.connect(WiFi.gatewayIP(); ,80); //tried it with my router first, both respond 0.

So, is this intended behavior? shouldn't wifi.status change it's status if promiscuous mode cancels it?

devyte, 2018-04-25:
@DasVoelkel the behavior of promiscous mode is specific to the SDK, and not of the core libs in this repo. That means that it is currently completely unsupported. In addition, I am not aware of any developer with plans to support it in the foreseeable future.
If it is your intent to just gain information about how promiscuous mode works and how to use, this is likely not the right place to ask. In this case
If it is your intent to figure it out, and contribute your findings here, e.g.: a working example, it would be appreciated. In this case, I can provide some info on the internals of this core, e.g.: point you to specific parts of the source code, or provide some insight about why some things are done the way they are.
Be aware that, for the most part, the WiFi handling code is a not-too-thick wrapper around the SDK api, so if you find weirds things, they are most likely in the SDK itself rather than in this core.

DasVoelkel, 2018-04-25:
ah ok.
I didn't know there were no further plans to support it. I did know that it was specified in the SDK and i read through what i thought was important. 
If the promiscuous mode is of no concern to the development of this projects (library) then you've answered my question in that both code parts where never considered to be run in the same place?
Meaning SDK's promiscuous mode and the WiFi functions were never considered to be run at the same time?

devyte, 2018-04-28:
That is a PR (pull request) with a proposal of a new lib, which can capture ethernet packets. You can dump the packets (print them out), or do whatever you want.
So the question is valid: do you need to observe at wifi (phy) level, or at ethernet packet level?
For phy you likely need the SDK promiscuous mode, which we don't currently support here.
For ethernet packet traffic, you should test the PR and check if it is enough for you. In this case, feedback is welcome.

DasVoelkel, 2018-04-28:
For the purpose of capturing beacon packets. I think i need phy level. 
I tried observing on ethernet level, sadly in an encrypted network this isn't much use. If i wanted more information out of the network i'd need access to the arp and ICMP protocols ,which is currently not supported.
So for my test case currently i'd need the physical switch. 
In this case "not supported" regarding the promiscuous mode, means that it won't at all work as intended, or that some features may work and some may not because of no official support.

d-a-v, 2018-04-28:
With #4678 you can monitor arp packets if you are inside/connected to the WiFi network.
Regarding WLAN promiscuous "not supported", it is not supported here in esp8266/Arduino, we don't even use it. You need to ask your questions about that in espressif/esp8266_nonos_sdk which is our closed-sources vendor firmware.

RudyFiero, 2018-04-29:
> **If monitoring arp packets is supported on here**, how come i've not found anything in those regards in the library source code ?

And who said that? You are being a Dickhead. d-a-v has been very generous with his time in regard to your request. He has pointed you in the correct directions. He explained that the access you want is not part of the Arduino/ESP8266 framework but may be available from a lower level (that is not the code here) that you need to explore on your own. If you want to see what information is here regarding arp requests then use the search box. Have you even tried that? Because that is what is here. Nothing else.

[image]

devyte, 2018-04-29:
Alright, let's keep things.civil here.
@DasVoelkel as I explained above, the packet capture PR is currently proposed, and not yet merged. It is meant for ethernet traffic which means packets exchanged between the ESP phy and lwip.
I am still unclear about whether you need the sdk promiscuous mode, as stated in the title, or if ethernet packet capture is enough for you.

d-a-v, 2018-04-29:
Testing this PR is welcome. Assuming you are using git, the instructions are
```
git fetch origin/pull/4678/head:netdump
git checkout netdump
```
then restart the arduino IDE, and you will find two examples in `Files>Examples>esp8266/netdump>`.

Just run the `netdump.ino` one, and you'll see all packets from and to the esp. It's of course better if these few lines are integrated into a proper arduino sketch doing something with the network.

DasVoelkel, 2018-05-01:
Thanks a lot for the help! If i find issues with that build should i report them in a new issue?

devyte, 2018-05-01:
@DasVoelkel no, if you find issues in the PR, please report them in that same PR.

DasVoelkel, 2018-05-02:
Thanks a lot fot the help! Sorry for not understanding everything right away!

closed 2018-05-02 by DasVoelkel