esp8266/Arduino #4686: Hi,

YashKhokale, 2018-04-26:
Soft WDT reset

ctx: cont 
sp: 3fff09f0 end: 3fff0bf0 offset: 01b0

>>>stack>>>
3fff0ba0:  00000004 00000000 3ffefbbc 40202808  
3fff0bb0:  00000000 00000000 00000000 00000004  
3fff0bc0:  00000000 00000000 feefeffe 3ffefbc4  
3fff0bd0:  3fffdad0 00000000 3ffefbbc 402090c4  
3fff0be0:  feefeffe feefeffe 3ffefbd0 40100710  
<<<stack<<<

Soft WDT reset 
ctx: sys 
sp: 3ffffd80 end: 3fffffb0 offset: 01b0 

>>>stack>>>
3fffff30: 00000018 0000000c 00000000 40212450
3fffff40: 40104d7e 0272014f 00000000 40212500 
3fffff50: 00000000 3ffee188 00000000 00000000 
3fffff60: 40239a60 3ffee188 3ffef0f0 60000600 
3fffff70: 4023c6c1 3ffee188 3ffef0f0 07d3ab85 
3fffff80: 4023c706 3fffdab0 00000000 3fffdcb0 
3fffff90: 3ffef110 3fffdab0 00000000 40209033 
3fffffa0: 40000f49 40000f49 3fffdab0 40000f49 
<<<stack<<<

YashKhokale, 2018-04-26:
Is there any way i could switch off Soft WDT

devyte, 2018-04-27:
No, and I'm pretty sure you don't want to do that.
You probably need to use yield() every so often in your code.
Be aware there are timing constraints to observe on the ESP. If you don't, the wifi stack can blow up.
Closing.

closed 2018-04-27 by devyte

YashKhokale, 2018-04-27:
Sorry for late reply....
so when i call yield() in an infinite while() loop, it will allow my esp8266 to perform background operation without causing the above stack error
@devyte

YashKhokale, 2018-04-28:
Ok, but can you suggest me some example sketches?
Btw thanks for helping me.
@devyte