esp8266/Arduino #4688: wdt reset at use pinMode+analogWrite

bstsoftorg, 2018-04-27:
### Basic Infos

#### Platform

- Hardware:				ESP-12
- Core Version:			2,4,1
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		4MB/3MB
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:		ck
- Flash Frequency:	80Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

I tried to switch to 2.4.1 and found a strange reboot. There are no such problems with version 2.3.0-rc2.
If in 2.4.1 to perform before pinMode analogWrite no matter what speed, this is a "wdt reset". This problem is manifested when you call to pinMode+analogWrite times per second and the day of these reboots was to 20. But it's tests when using 1 port, if the ports will be greater, the stability at all. Also, tests have shown that very frequent falls occur during rapid circulation. See log.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

uint32 qqq = 0;

void loop() {
  HTTP.handleClient();
Serial.print(F("*1*="));
Serial.print(qqq);
//if (qqq == 0)
    pinMode(13, OUTPUT);
    analogWrite(13, qqq);
Serial.println(F("*3*"));
qqq++;
}
```

### Debug Messages

```

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

*1*=8293*3*
*1*=8294
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

*1*=3173*3*
*1*=3174*3*
*1*=3175
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

*1*=8292*3*
*1*=8293*3*
*1*=8294
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

*1*=17601*3*
*1*=17602*3*
*1*=17603
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

*1*=8292*3*
*1*=8293*3*
*1*=8294
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

*1*=11457*3*
*1*=11458*3*
*1*=11459
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

```

devyte, 2018-04-28:
@bstsoftorg please edit your issue and fix the sketch. The C in MCVE is for Complete.

bstsoftorg, 2018-04-28:
Everything is standard FS Browser and replace a single procedure. Here's the updated sketch for you, only to sense no the same error:
`
*1*=2153*3*
*1*=2154*3*
*1*=2155
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
`

msilenius, 2018-05-10:
analogwrite is currently bugged, cf issue 4321

https://github.com/esp8266/Arduino/issues/4321 the workaround described there is working

bstsoftorg, 2018-05-10:
`
...
void setup(void){
...
  pinMode(15, OUTPUT);
  analogWrite(15, 0);
  analogWrite(15, 10);
}
 
/*void loop(void){
  server.handleClient();
}*/

uint32 qqq = 0;

void loop() {
  server.handleClient();
Serial.print(F("*1*="));
Serial.print(qqq);
//if (qqq == 0)
    pinMode(15, OUTPUT);
    analogWrite(15, qqq);
Serial.println(F("*3*"));
qqq++;
}
`
result
`
*1*=11457*3*
*1*=11458*3*
*1*=11459
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
`

This is not a workaround since all ports are used by the system and are not free.
16 the port is generally the port is connected through a resistor 471 with Rosatom.
At any time, any of the ports 0, 12, 13, 14, 15 can have values from 0 to 1023
Free ports simply aren't present.
Also, at any time, the port can become digital or analog. The system is non-static and fully dynamic depending on the situation.
Such problems there is no in 2.3.0rc2, so you need to see what changes were made after this version in PWM functions.

msilenius, 2018-05-10:
you misunderstand the workaround. Choose a GPIO port that you don't use anywhere in your scketch and put in the setup:

pinMode(your pin you don't use, OUTPUT);
analogWrite(your pin you don't use, 0);
analogWrite(your pin you don't use, 10);

devyte, 2018-12-05:
Closing due to lack of feedback and age.

closed 2018-12-05 by devyte

ateeb327, 2019-07-20:
Hello, I was getting the same wdt reset error while trying to use pinMode with digitalWrite() function. 
I solved it by figuring out how the Wemos Pin mapping works. 
Actually, you need to refer the pin in 'Dx' notation. 
e.g digitalWrite(D15,LOW);
or 
pinMode(D15, OUTPUT); 

Also, make sure to select "Wemos D1 R1" in Tools > Boards so that Dx constants will match
 the labels.

HejaSverige, 2020-01-09:
@ateeb327:
Thank you man, you really made my day. :-) I was close to madness because of that problem. I'm using the NodeMCU 1.0 (ESP-12E) and as soon as I used the pinMode command, the wdt reset error occured. Now everything works fine, so thank you once again.

slellis, 2020-03-31:
Hi all,

I faced this problem recently.
For future reference, what happened to me was an error in the definition of the board. 
I am using a Nodemcu and had set ESP8266 generic 1.0