esp8266/Arduino #4689: ESP - Exception (29) - WiFiClient?

mojgithub, 2018-04-27:
### Basic Infos

- [yes  ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [yes  ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [yes  ] I have tested that the issue is present in current master branch (aka latest git).
- [yes ] I have searched the issue tracker for a similar issue.
- [yes  ] If there is a stack dump, I have decoded it.
- [yes  ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12e]
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL + OTA]
- Upload Speed:   [115200)

### Problem Description

hello

After some time ESP randomly(few minutes, hours) loses connection. It does not reconnect. 
I do not know how to reproduce this exception. I just wait.

Below is exception code.

I need some direction what is wrong in the code.

Exception was triggered with v2 Lower Memory. 
So now i am testing also with 1.4, will post results if any error.

btw i am not good in decoding and sorry if my code looks bad. :) (not a programmer)

tnx

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Wire.h>
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <TimeLib.h>
#include <WidgetRTC.h>
#include <ArduinoOTA.h>

// WiFi network info.
const char* ssid[] = {"XXXXXX", "YYYYY", "CCCCCCC", "VVVVVVVV"};
const char* password[] = {"DDDDDDDDDDDD", "FFFFFFFFF", "HHHHHHHHHHHH", "TTTTTTTTTTTT"};
#define sizeSSID (sizeof(ssid)/sizeof(char *))

// Blynk
char auth[] = "XXXXXXXXXXXXXXXXX";
const char* blynkSSID;
const char* blynkPASSWORD;

// THINGSPEAK SETTINGS
// ********************
const char* server = "api.thingspeak.com";
[... 1178 lines cut ...]
3fff0ea0:  00000008 3ffef33c 3ffef45c 40206fc4  
3fff0eb0:  3fffdad0 00000000 3ffefeb0 402051c2  
3fff0ec0:  3fffdad0 00000000 3ffefeb0 4020addc  
3fff0ed0:  feefeffe feefeffe 3ffefec0 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

Decoding 11 results
0x4020678d: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src/WiFiClient.cpp line 371
0x4021271b: tcp_close_shutdown at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp.c line 1779
0x4020633c: WiFiClient::connect(char const*, unsigned short) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src/WiFiClient.cpp line 371
0x40107248: sntp_get_current_timestamp at ?? line ?
0x40107248: sntp_get_current_timestamp at ?? line ?
0x402054aa: push_ThingSpeak_1() at E:\Projects\Arduino\Water-Irrigation\WI-ESP/WI-ESP.ino line 906
0x402054fe: push_ThingSpeak_2() at E:\Projects\Arduino\Water-Irrigation\WI-ESP/WI-ESP.ino line 910
0x40206fc4: BlynkTimer::run() at E:\Projects\Arduino\examples\libraries\Blynk\src\utility/BlynkTimer.cpp line 108
0x402051c2: loop at E:\Projects\Arduino\Water-Irrigation\WI-ESP/WI-ESP.ino line 168
0x4020addc: loop_wrapper at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/cont.S line 109
```

mojgithub, 2018-04-27:
> Exception was triggered with v2 Lower Memory.
> So now i am testing also with 1.4, will post results if any error.

Update: After 1h online it is down again with 1.4 setting.

I gonna switch back to v2 Lower Memory and try to fix with this settings. 

Any clue what should i try next?

mojgithub, 2018-04-27:
@tbdltee 

So i have tried ESP.getFreeHeap(); 

See below.

```
struct tcp_pcb;
extern struct tcp_pcb* tcp_tw_pcbs;
extern "C" void tcp_abort (struct tcp_pcb* pcb);

void tcpCleanup (void) {
  while (tcp_tw_pcbs)
    tcp_abort(tcp_tw_pcbs);
}

```

and then i am calling this function before client.connect and after client.stop();

```

void push_ThingSpeak_1() {
  Serial.print(String("FreeHeap Thingspeak_1_before: "));
  Serial.println(ESP.getFreeHeap(),DEC);
  tcpCleanup();
  Serial.print(String("FreeHeap Thingspeak_1_before_tcpCleanup "));
  Serial.println(ESP.getFreeHeap(),DEC);
  if (client.connect(server,80)) {
    String postStr = apiKey_1;
           if (!isnan(deviceOnline)) {
              postStr +="&field1=";
              postStr += deviceOnline;
           }
           if(masterStatus) {
              postStr +="&field2=";
              postStr += 1;
           }
           else {
              postStr +="&field2=";
              postStr += 0;
           }
           if (!isnan(flowMilliLitres_1)) {
              postStr +="&field3=";
              postStr += flowMilliLitres_1;
           }
           if (!isnan(flowTotalLitres_1)) {
              postStr +="&field4=";
              postStr += flowTotalLitres_1;
           }
           if (!isnan(flowMilliLitres_2)) {
              postStr +="&field5=";
              postStr += flowMilliLitres_2;
           }
           if (!isnan(flowTotalLitres_2)) {
              postStr +="&field6=";
              postStr += flowTotalLitres_2;
           }
           if (!isnan(flowTotalLitres_1) || !isnan(flowTotalLitres_2)) {
              postStr +="&field7=";
              postStr += flowTotalLitres_1 + flowTotalLitres_2;
           }
           
           //postStr += "\r\n\r\n";
     client.print("POST /update HTTP/1.1\n"); 
     client.print("Host: api.thingspeak.com\n"); 
     client.print("Connection: close\n"); 
     client.print("X-THINGSPEAKAPIKEY: "+apiKey_1+"\n"); 
     client.print("Content-Type: application/x-www-form-urlencoded\n"); 
     client.print("Content-Length: "); 
     client.print(postStr.length()); 
     client.print("\n\n"); 
     client.print(postStr);    
  }
  client.stop();
  Serial.print(String("FreeHeap Thingspeak_1: "));
  Serial.println(ESP.getFreeHeap(),DEC);
  tcpCleanup();
  Serial.print(String("FreeHeap Thingspeak_1_tcpCleanup: "));
  Serial.println(ESP.getFreeHeap(),DEC);
}

```

Results after few minutes:

```
FreeHeap Thingspeak_1_before: 37392
FreeHeap Thingspeak_1_before_tcpCleanup 37392
FreeHeap Thingspeak_1: 36384
FreeHeap Thingspeak_1_tcpCleanup: 36384

FreeHeap Thingspeak_2_before: 37840
FreeHeap Thingspeak_2_before_tcpCleanup 38008   **(+~170)**
FreeHeap Thingspeak_2: 36328
FreeHeap Thingspeak_2_tcpCleanup: 36328

FreeHeap Thingspeak_3_before: 37112
FreeHeap Thingspeak_3_before_tcpCleanup 37280  **(+~170)**
FreeHeap Thingspeak_3: 36272
FreeHeap Thingspeak_3_tcpCleanup: 36272

FreeHeap Thingspeak_1_before: 37056
FreeHeap Thingspeak_1_before_tcpCleanup 37224  **(+~170)**
FreeHeap Thingspeak_1: 36216
FreeHeap Thingspeak_1_tcpCleanup: 36216

FreeHeap Thingspeak_2_before: 37672
FreeHeap Thingspeak_2_before_tcpCleanup 37840  **(+~170)**
FreeHeap Thingspeak_2: 36160
FreeHeap Thingspeak_2_tcpCleanup: 36160

FreeHeap Thingspeak_3_before: 36944
FreeHeap Thingspeak_3_before_tcpCleanup 37112  **(+~170)**
FreeHeap Thingspeak_3: 36104
FreeHeap Thingspeak_3_tcpCleanup: 36104

.
.
.
.
.
.
.

FreeHeap Thingspeak_1_before: 34984
FreeHeap Thingspeak_1_before_tcpCleanup 35320  **(+~340)**
FreeHeap Thingspeak_1: 34312
FreeHeap Thingspeak_1_tcpCleanup: 34312

FreeHeap Thingspeak_2_before: 35096
FreeHeap Thingspeak_2_before_tcpCleanup 35264  **(+~170)**
FreeHeap Thingspeak_2: 34256
FreeHeap Thingspeak_2_tcpCleanup: 34256

FreeHeap Thingspeak_3_before: 34312
FreeHeap Thingspeak_3_before_tcpCleanup 34312
FreeHeap Thingspeak_3: 33304
FreeHeap Thingspeak_3_tcpCleanup: 33304
```

**Overall it is going down ...**

Should i call it somewhere else?

tbdltee, 2018-04-28:
no need. it tells you that your memory is getting lower and lower. To prove that it's not from your code, recompile the same code with 2.3.0 and see the memory.

mojgithub, 2018-04-28:
@tbdltee 

core 2.3.0. works normally, no leaks!

@devyte 

i have trouble with installation "Using git version". Never done this before. I am doing this steps https://arduino-esp8266.readthedocs.io/en/2.4.1/installing.html

1) I have removed Additional Board URLs from Arduino IDE.
2) I made the same directory structure and downloaded binary tools as in instruction.
3) I reopened Arduino.
4) Is this ok? I dont see any core dropdown option in board manager, should i see it?

devyte, 2018-04-28:
No, it doesn't show up in the board manager. You should see the ESP boards, though. With that, you should be able to rebuild.