esp8266/Arduino #4690: WiFiTelnetToSerial Serial1 & Serial not working ???

doudousha, 2018-04-27:
void setup() {
  **Serial1**.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  **Serial1**.print("\nConnecting to "); Serial1.println(ssid);
  uint8_t i = 0;
  while (WiFi.status() != WL_CONNECTED && i++ < 20) delay(500);
  if(i == 21){
    Serial1.print("Could not connect to"); Serial1.println(ssid);
    while(1) delay(500);
  }
  //start UART and the server
  **Serial**.begin(115200);
  server.begin();
  server.setNoDelay(true);
  
  **Serial1**.print("Ready! Use 'telnet ");
  **Serial1**.print(WiFi.localIP());
  **Serial1**.println(" 23' to connect");
}

closed 2018-04-28 by devyte

ancich, 2019-03-28:
I agree

d-a-v, 2019-03-28:
Please elaborate