esp8266/Arduino #4693: ESP8266 and firebase Firebase.setFloat() problem

YashKhokale, 2018-04-27:
Hi,
i am unable to upload my sensor data into my firebase database through arduino code and <FirebaseArduino.h> library .
Firebase.set("xyz", 0);
Firebase.set("abc", 0);
the above commands work perfectly fine in my setup() function.
But only one these below commands work in my loop() functon i.e, only only one variable in updated into my firebase database.
Firebase.setFloat("xyz", x1);     // x1 is a variable. 
Firebase.setFloat("abc", x2);   //  x2 is a variable.

Can anyone help me out here?

devyte, 2018-04-28:
If there is an issue with using Firebase on the ESP, then it should be brought up with Firebase first. If during that investigation an issue is found with the core libs here, then an issue should be opened here.
Closing.

closed 2018-04-28 by devyte

YashKhokale, 2018-04-28:
Yes, i think so there is a problem with the firebase lib....because i am able to update only one of the above variables only once i.e. when my esp8266 resets one of the above variable is updated only once initially
@devyte