esp8266/Arduino #4697: HTTP Post REQ: Send data failed error

Sushantnaik21, 2018-04-28:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description
I am trying to use HTTP POST request, using esphttpclient, but i am unable to send data. below is my modified code from sample example, 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266HTTPClient.h>
#include <ESP8266WiFi.h>

void setup() {
  Serial.begin(115200); 
  Serial.setDebugOutput(1);//Serial connection
  WiFi.begin("Guren Mark 2", "abrakadabra");   //WiFi connection
  
  while (WiFi.status() != WL_CONNECTED) {  //Wait for the WiFI connection completion
  
  delay(500);
  Serial.println("Waiting for connection");
 
  }
}

void loop() {
 if(WiFi.status()== WL_CONNECTED){   //Check WiFi connection status

  int httpCode;
 
  HTTPClient http;    //Declare object of class HTTPClient

  httpCode = http.begin("www.sappush.com", 3000, "/api1/devices/10/setWifi");  //Specify request
  http.addHeader("Content-Type", "application/json");  //Specify content-type header

  httpCode = http.POST("token=1234567890abc");   //Send the request
 // httpCode = http.POST("\"token\":\"1234567890abc\"");   //Send the request

 // httpCode will be negative on error
        if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            Serial.printf("[HTTP] POST... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                Serial.println(payload);
            }
        } else {
            Serial.printf("[HTTP] POST... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }
   http.end();  //Close connection
} // wifi connected IF
delay(30000);  //Send a request every 30 seconds
}

```

### Debug Messages

```
[HTTP-Client][begin] host: www.sappush.com port: 3000 uri: /api1/devices/10/setWifi
[HTTP-Client] connected to www.sappush.com:3000
[HTTP-Client][returnError] error(-3): send payload failed
[HTTP] POST... failed, error: send payload failed
[HTTP-Client][end] tcp is closed
```

RudyFiero, 2018-04-29:
@Sushantnaik21 you checked the following box.

[ x] This issue complies with the issue POLICY doc.

It does not comply. You are asking for programming help. You are asking for someone to tell you where you went wrong. You are not bringing up an issue with the Arduino core. And that is what this issue tracking system is for. Go back and read sections 3 and 4 as they apply to your request. 

> I am trying to use HTTP POST request, using esphttpclient, but i am unable to send data. below is my modified code **from sample example**,

The sample **sample example** you are using does not come from this repository. It comes from https://circuits4you.com/2018/03/10/esp8266-nodemcu-post-request-data-to-website/
There are two parts to the example. The ESP8266 code and the server side code. Use both as they are published in the example. If they then don't work then go ask for assistance from the person who has made that example. It has nothing to do with this issue system. 

Close this issue because it is not a valid issue.

Sushantnaik21, 2018-04-29:
> It does not comply. You are asking for programming help. You are asking for someone to tell you where you went wrong. You are not bringing up an issue with the Arduino core. And that is what this issue tracking system is for. Go back and read sections 3 and 4 as they apply to your request.
Sorry for inconvenience.

closed 2018-04-29 by devyte

toviddd, 2019-09-24:
> 
> 
> @Sushantnaik21 you checked the following box.
> 
> [ x] This issue complies with the issue POLICY doc.
> 
> It does not comply. You are asking for programming help. You are asking for someone to tell you where you went wrong. You are not bringing up an issue with the Arduino core. And that is what this issue tracking system is for. Go back and read sections 3 and 4 as they apply to your request.
> 
> > I am trying to use HTTP POST request, using esphttpclient, but i am unable to send data. below is my modified code **from sample example**,
> 
> The sample **sample example** you are using does not come from this repository. It comes from https://circuits4you.com/2018/03/10/esp8266-nodemcu-post-request-data-to-website/
> There are two parts to the example. The ESP8266 code and the server side code. Use both as they are published in the example. If they then don't work then go ask for assistance from the person who has made that example. It has nothing to do with this issue system.
> 
> If you think your code is good, because it worked at https://circuits4you.com/2018/03/10/esp8266-nodemcu-post-request-data-to-website/ then there is nothing wrong with the Arduino code. It worked at that example so the code here in this repository is good. You have other problems.
> 
> Close this issue because it is not a valid issue.

this worked like a charm