esp8266/Arduino #4698: network not found trying with WPA2-enterprise at my university network

kapyaar, 2018-04-28:
#### Platform

- Hardware:				[ESP-12]
- Core Version:			[latest git/2.4.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu 0.9]
- Flash Mode:			[qio]
- Flash Size: 		        [4MB/1MB]
- lwip Variant:                  [v1.4|v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:              [921600]

### Problem Description
I am trying to connect an ESP to wpa2 network at my university. Debug shows the module cannot find the network. If  I do a wifi scan, it does find the network though. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

extern "C" {
#include "user_interface.h"
#include "wpa2_enterprise.h"
}

static u8 ent_username[] = "username@buffalo.edu";
static u8 ent_password[] = "ubitpassword";
char username[] = "username@buffalo.edu";
void setup() {
  char a[100];
  ip_info info;
  wifi_get_ip_info(0, &info);

  Serial.begin(115200);

  wifi_station_disconnect();
  wifi_set_opmode(STATION_MODE);

  char ssid[32] = "UB_Secure";//Tried eduroam too and same error - 'no eduroam found'
  char password[64] = {0x00};
  struct station_config stationConf;
  stationConf.bssid_set = 0;  //need not check MAC address of AP
  memcpy(&stationConf.ssid, ssid, 32);
  memcpy(&stationConf.password, password, 64);

  if (!wifi_station_set_config(&stationConf)) {
    Serial.print("\r\nset config fail\r\n");
  }

  do_wifiscan();
  // switch to WPA2 Enterprise
  wifi_station_set_wpa2_enterprise_auth(1);
  wifi_station_set_enterprise_identity((uint8*)username, strlen(username));
  //https://github.com/esp8266/Arduino/issues/2595#issuecomment-267809669
  if (wifi_station_set_enterprise_username (ent_username, strlen((char*)ent_username))) {
    Serial.print("\r\nusername set fail\r\n");
  }
  if (wifi_station_set_enterprise_password (ent_password, strlen((char*)ent_password))) {
    Serial.print("\r\npassword set fail\r\n");
  }

  //do_wifiscan();
  if (!wifi_station_connect()) {
    Serial.print("\r\nconnect fail\r\n");
  }

  //do_wifiscan();
  Serial.print("\r\ntrying to connect...");
  
  while (info.ip.addr == 0) {
    //ESP.wdtFeed();
    Serial.print(".");
    delay(1000);
    wifi_get_ip_info(0, &info);
    
  }

  sprintf(a, "%"PRIu32, info.ip.addr);
  Serial.print("\r\nip addr: ");
  Serial.print(a);
  Serial.print("\r\n");
}

/*==========================Generate list of available wifi networks================================*/
void do_wifiscan( void ) {
  int netCount;
 Serial.println( F("scan start") );

  // WiFi.scanNetworks will return the number of networks found
  netCount = WiFi.scanNetworks();
  //  DEBUG_PRINTLN("scan done"); // no need if Serial.setDebugOutput(true)
  if ( netCount == 0 ) {
    Serial.println("no network found");
  } else {
    Serial.print(netCount);
   Serial.println(" network(s) found");

  }
  Serial.println("");
  String aplist = "<?xml version=\"1.0\" encoding=\"UTF-8\"?><aplist>";
  for ( int i = 0; i < netCount; ++i ) {
    aplist += "<network><ssid>" + (String(WiFi.SSID(i))) + "</ssid>";
    aplist += "<auth>" + ((WiFi.encryptionType(i) == ENC_TYPE_NONE) ? " " : String(WiFi.encryptionType(i))) + "</auth>";
    aplist += "<str>" + String(WiFi.RSSI(i)) + "</str>";
   aplist += "<chn>" + String( WiFi.channel(i)) + "</chn></network>";
  }
  aplist += "</aplist>";
 
  Serial.println(aplist);
}
void loop()
{
  
}
```

### Debug Messages

```
-é2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
scandone
scan start
V*ÙÑé 8
scandone
17 network(s) found

<network><ssid>UB_Connect</ssid><auth> </auth><str>-81</str><chn>1</chn></network><network><ssid>UB_Connect</ssid><auth> </auth><str>-70</str><chn>1</chn></network><network><ssid>UB_Secure</ssid><auth>255</auth><str>-81</str><chn>1</chn></network><network><ssid>eduroam</ssid><auth>255</auth><str>-70</str><chn>1</chn></network>....

WPA2 ENTERPRISE VERSION: [v2.0] enable

trying to connect......scandone
no UB_Secure found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconn..scandone
no UB_Secure found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
reconnect
..scandone
no UB_Secure found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
recon...scandone
no UB_Secure found, reconnect after 1s
wifi evt: 1
STA disconnect: 201
recon..scandone

and keeps going with same mesage repeat
```

kapyaar, 2018-05-08:
OK, I went a step further, got a TP Link EAP access point, and configured it with an external radius server, created a testUser, etc. Then, First, I connected my iphone to this network. It connected, asked for my username and password, and successfully authenticated. I could browse the internet, no issues. Then, I tried with ESP, with various minor code changes, and no success. But the good thing is, now I can see what is happening on the radius server. ESP does connect to the AP, but it is the radius auth that is failing. I belive this has to do with the way ESP core is sending info? Anyone who is familiar with the core might be able to help.

Log when Iphone connects successfully.

`Access Request:	Packet-Type = Access-Request
Packet-Src-IP-Address = 172.101.117.178
Packet-Dst-IP-Address = 10.142.0.4
Packet-Src-Port = 51481
Packet-Dst-Port = 8315
User-Name = "testUser"
NAS-IP-Address = 192.168.0.254
NAS-Port = 0
Called-Station-Id = "84-16-F9-88-81-62"
Calling-Station-Id = "24-F0-94-0A-86-91"
Framed-MTU = 1400
NAS-Port-Type = Wireless-802.11
Connect-Info = "CONNECT 0Mbps 802.11"
EAP-Message = 0x0200000e0161766172616368616e
Message-Authenticator = 0x38078a459ea243868a46981130743784
Event-Timestamp = "May  8 2018 18:00:50 UTC"
Tmp-Integer-3 = 8315
Tmp-String-1 = "local"
Called-Station-SSID = "UB_Secure"
Request-Session = "A8EtE6twSP"
Timestamp = 1525802450
Request Reply:	Packet-Type = Access-Accept
User-Name = "testUser"
MS-MPPE-Recv-Key = 0xd5cbe4473c46d30e890fe3c3924d523e6be032b3094bf14b60e05af5ba9b519e
MS-MPPE-Send-Key = 0xc73ef4774c2323653421aa7f4c84edd981bdbbd5112528c5d67f3ffff6176047
EAP-MSK = 0xd5cbe4473c46d30e890fe3c3924d523e6be032b3094bf14b60e05af5ba9b519ec73ef4774c2323653421aa7f4c84edd981bdbbd5112528c5d67f3ffff6176047
EAP-EMSK = 0xbcfd626b8b288217d499fcdfeab443e447c3b249f9f40534ce118b27195fcb726a5bfcb8d4daf828ac7fc5b197cae480f10d3269b95671ac1f2b55a4ec7afa25
EAP-Session-Id = 0x195af1e5d304697d2ba6c623125bbde0d6d1840641b961bab216fa17cb09a377c85af1e5d3a10361a0bfaa66c6476aa301b1e6ac09448148ccdf10a057fd9b8cab
EAP-Message = 0x030a0004
Message-Authenticator = 0x00000000000000000000000000000000
Timestamp = 1525802456
authdate:	2018-05-08 14:00:50
called_station_id:	84-16-F9-88-81-62
calling_station_id:	24-F0-94-0A-86-91
inner_username:	testUser
port:	8315
result:	Access-Accept
username:	testUser`

Now, with ESP conecting,

`Access Request:	Packet-Type = Access-Request
Packet-Src-IP-Address = 172.101.117.178
Packet-Dst-IP-Address = 10.142.0.4
Packet-Src-Port = 51481
Packet-Dst-Port = 8315
User-Name = "testUser"
NAS-IP-Address = 192.168.0.254
NAS-Port = 0
Called-Station-Id = "84-16-F9-88-81-62"
Calling-Station-Id = "5C-CF-7F-19-D7-AE"
Framed-MTU = 1400
NAS-Port-Type = Wireless-802.11
Connect-Info = "CONNECT 0Mbps 802.11"
EAP-Message = 0x0200000e0161766172616368616e

Message-Authenticator = 0xb4e7aff218206bc6a65477bc7dd91c98
Event-Timestamp = "May  8 2018 19:15:04 UTC"
Tmp-Integer-3 = 8315
Tmp-String-1 = "local"
Called-Station-SSID = "UB_Secure"
Request-Session = "dx6I5LuWl0"
Timestamp = 1525806904
Request Reply:	Packet-Type = Access-Reject
EAP-Message = 0x04010004
Message-Authenticator = 0x00000000000000000000000000000000
Timestamp = 1525806905
authdate:	2018-05-08 15:15:04
called_station_id:	84-16-F9-88-81-62
calling_station_id:	5C-CF-7F-19-D7-AE
port:	8315
result:	Access-Reject
username:	testUser`

kapyaar, 2018-05-09:
#More updates

Playing around more on this, I tried the following. I tried to connect my iphone to the enterprise network with the correct username but wrong password. This time, Two interesting finds.

1. The radius server log for iphone matches more or less with that of the ESP
2. The radius server log for iphone shows a line called 'inner_usernamme' which is missing from ESP. 

Does this light any bulbs, any one? :)

Radius log for iphone with correct username and wrong password

```

Access Request:	Packet-Type = Access-Request
Packet-Src-IP-Address = 172.101.117.178
Packet-Dst-IP-Address = 10.142.0.4
Packet-Src-Port = 51481
Packet-Dst-Port = 8315
User-Name = "testUser"
NAS-IP-Address = 192.168.0.254
NAS-Port = 0
Called-Station-Id = "84-16-F9-88-81-62"
Calling-Station-Id = "24-F0-94-0A-86-91"
Framed-MTU = 1400
NAS-Port-Type = Wireless-802.11
Connect-Info = "CONNECT 0Mbps 802.11"
EAP-Message = 0x0200000e0161766172616368616e

Message-Authenticator = 0x73ed9adc6cc07fc6452963f2a4a3bc3e
Event-Timestamp = "May  9 2018 18:57:46 UTC"
Tmp-Integer-3 = 8315
Tmp-String-1 = "local"
Called-Station-SSID = "UB_Secure"
Request-Session = "UOBQhUF1IQ"
Timestamp = 1525892266
Request Reply:	Packet-Type = Access-Reject
EAP-Message = 0x04090004
Message-Authenticator = 0x00000000000000000000000000000000
Timestamp = 1525892268
authdate:	2018-05-09 14:57:46
called_station_id:	84-16-F9-88-81-62
calling_station_id:	24-F0-94-0A-86-91
inner_username:	testUser
port:	8315
result:	Access-Reject
username:	testUser
```

Radius log for ESP with correct username and password

```
Access Request:	Packet-Type = Access-Request
Packet-Src-IP-Address = 172.101.117.178
Packet-Dst-IP-Address = 10.142.0.4
Packet-Src-Port = 51481
Packet-Dst-Port = 8315
User-Name = "testUser"
NAS-IP-Address = 192.168.0.254
NAS-Port = 0
Called-Station-Id = "84-16-F9-88-81-62"
Calling-Station-Id = "5C-CF-7F-19-D7-AE"
Framed-MTU = 1400
NAS-Port-Type = Wireless-802.11
Connect-Info = "CONNECT 0Mbps 802.11"

EAP-Message = 0x0201000e0161766172616368616e
Message-Authenticator = 0x66ae3c9a3d6a4306437f42d954911411
Event-Timestamp = "May  9 2018 18:32:16 UTC"
Tmp-Integer-3 = 8315
Tmp-String-1 = "local"
Called-Station-SSID = "UB_Secure"
Request-Session = "4lutlg4WNe"
Timestamp = 1525890736
Request Reply:	Packet-Type = Access-Reject
EAP-Message = 0x04020004
Message-Authenticator = 0x00000000000000000000000000000000
Timestamp = 1525890736
authdate:	2018-05-09 14:32:16
called_station_id:	84-16-F9-88-81-62
calling_station_id:	5C-CF-7F-19-D7-AE

port:	8315
result:	Access-Reject
username:	testUser
```

d-a-v, 2019-06-23:
[https://github.com/espressif/ESP8266_RTOS_SDK/issues/635](http://comment/if/you/can)

Flole998, 2019-07-17:
@d-a-v Your link is messed up

d-a-v, 2019-07-17:
https://github.com/esp8266/Arduino/issues/3442#issuecomment-504954666

bkrajendra, 2019-09-28:
Any update on this for ESP8266 WPA2-enterprise PEAP?

Vbansal1996, 2019-12-13:
Did we have any success with ESP8266 WPA2-enterprise PEAP?

bkrajendra, 2019-12-13:
not possible with ESP8266. Got success with ESP32 but only works with few network, not with all.
Finally shifted to RPI3B+