esp8266/Arduino #4700: ESP8266 mini-NodeMCU it does not appear in Arduino IDE tools/port/

felox13, 2018-04-29:
Hello, I bought this ESP8266 mini-NodeMCU : https://www.banggood.com/Mini-NodeMCU-ESP8266-WIFI-Development-Board-Based-On-ESP-12F-p-1054209.html?cur_warehouse=CN and when I plug it into my Mac, it does not appear in Arduino IDE tools/port/? I installed the USB driver ( https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers) I tried to change the USB cable and added the ESP8266 cards to the IDE card library. However, it does not always appear in tools/port/. Could someone help me please?

Translated with www.DeepL.com/Translator

WereCatf, 2018-04-29:
The USB-UART-chip is not a Silabs-chip, it's a CH340, so those drivers won't work. You need to find a CH340-driver for your Mac.

felox13, 2018-04-29:
Okay, thanks. I'll try to find one.

felox13, 2018-04-29:
no sorry but still nothing I installed this driver : http://ouilogique.com/ch340_driver/ and the ESP8266 does not appear in tools/port

WereCatf, 2018-04-29:
I do not use OSX, so I can't help you any further. Besides, this place is for issues with the ESP8266-core, not for user-issues; go to esp8266.com and ask for help there.

felox13, 2018-04-29:
ok thanks i watched it and it works thanks

closed 2018-04-29 by devyte