esp8266/Arduino #4701: Memory leak 60 bytes at a time

AbitEncrypted, 2018-04-30:
[image]

### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			Wemos D1 Mini
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE & Visual Studio for Arduino]
- Operating System:	[Windows]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[?]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

A small memory leak of ~60bytes occurs after making a call to UniversalTelegramBot::getUpdates()
See attached graph.
This coincides with the following Serial output when the bot debug is set to true.
See debug messages below.
I am using board manager ESP8266 v2.4.1.  I was using v2.3.0 prior to the latest because the memory leak was even worse. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch
My sketch is too large to go here.  
### Debug Messages

Checking for bot messages
GET Update Messages
[BOT]Connecting to server
.... connected to server
{"ok":true,"result":[]}
incoming message length23
Creating DynamicJsonBuffer
no new messages 

Eventually the memory falls to 4k and the Wemos throws -
[BOT]Connecting to server

Exception (29):
epc1=0x4021f9fa epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont
sp: 3fff2070 end: 3fff28f0 offset: 01a0

>>>stack>>>
3fff2210:  00820000 00000082 3fff7b5c 4022cd12
.......
<<<stack<<<

ets Jan  8 2013,rst cause:2, boot mode:(1,6)

5chufti, 2018-05-01:
probably related to  #1923 et al.
there seem to be several ways to "exploit" the wifi.client() memory leak with more or less "leakage".
So I doubt your "[x ] I have searched the issue tracker for a similar issue."

AbitEncrypted, 2018-05-01:
devyte Merge branch 'master' into bugfix/wificlient_connect_leak 
Has fixed my issue.  Thanks.

closed 2018-05-01 by AbitEncrypted