esp8266/Arduino #4706: No way to resolve mDNS hostnames

G-glop, 2018-05-03:
The ESP8266 can do everything else with mDNS (advertise services, query services, set it's hostname), apart from the very important thing of querying hostnames.

ESP32 can do with the MDNS.queryHost function, but the function only wraps one from the sdk, which lives inside a compiled binary. So it's no use for backporting it to the ESP8266.

RudyFiero, 2018-05-04:
From [https://github.com/esp8266/Arduino/issues/4591#issuecomment-377728021](https://github.com/esp8266/Arduino/issues/4591#issuecomment-377728021)

> Be aware that I am currently rewriting the mdns responder/query based on the idf implementation. It is very slow going, though, so it will take quite some time.

palmerabollo, 2018-06-03:
I'm also interested in this feature.
@G-glop do you mean that mDNSResolver suddenly stops working?

G-glop, 2018-06-04:
@palmerabollo It stops working after about 20 resolutions, and it also blocks all other connections. Probably some kind of resource leak.

kosso, 2018-06-07:
Glad to find this thread. 

I found myself wondering why there was no ESP8266 equivalent to the `MDNS.queryHost()` method, which makes life a whole lot easier. 

Going to try mDNSResolver for now.

RudyFiero, 2018-06-07:
I assume that someone made their own library and it had some basic functionality. And because it had useful features it was included in this framework.  I don't believe it was ever claimed to be a full implementation.

devyte, 2018-12-05:
Closing in view of #5442 with a full rewrite of mdns.

closed 2018-12-05 by devyte