esp8266/Arduino #4707: Sonoff basic ESP http update

lapgoch, 2018-05-04:
I am trying to update a sketch on a Sonoff Basic using Esp Http Update.
My sketch binaries are on a RPI web server.
The problem is that it looks like the sketch size is too big for the update , but I can't see why.
The Sonoff Basic is a 1MB and the sketch size is 321Kb.
I am using Spiffs which is set to 64Kb in the Arduino 1.8.5 IDE.
Here is a snippet from the com port output when the update fails.

Update Check
IDE size = 1048576
Flash size = 1048576
Sketch size = 321536
Space size = 638976
Sketch url = http://192.168.0.22/php/updateCheck.php
Update status = 0
HTTP_UPDATE_FAILED Error (-107): New Binary Does Not Fit Flash Size

Does anyone know why a sketch size 321Kb with free space 639Kb fails to update?

Eszartek, 2018-05-04:
in  ESP8266httpUpdate.cpp you'll see:
```
uint32_t bin_flash_size = ESP.magicFlashChipSize((buf[3] & 0xf0) >> 4);

// check if new bin fits to SPI flash
if(bin_flash_size > ESP.getFlashChipRealSize()) {
  DEBUG_HTTP_UPDATE("[httpUpdate] New binary does not fit SPI Flash size\n");
  _lastError = HTTP_UE_BIN_FOR_WRONG_FLASH;
  http.end();
  return HTTP_UPDATE_FAILED;
}
```
So one of the 2 variables in there is wrong, either:
   ESP.magicFlashChipSize((buf[3] & 0xf0) >> 4) 
or
   ESP.getFlashChipRealSize()

My guess is that ESP.getFlashChipRealSize() is probably correct.

ESP.magicFlashChipSize is clearly taking the 4th byte from your update.bin file that was served by the RPI and converting it to an equivelent size value. If your RPI server is encoding the update.bin in any way, this could cause a corruption. It could also be that your IDE was set to the wrong flash size for generating the update.bin file.

From reading the source code, all the size values listed in the serial outout you gave are derived from the existing running app, which are all fine. So that makes me believe your issue is an external one.

lapgoch, 2018-05-05:
Hi.
Thanks for reply.
I am really at a loss with this one.

The IDE size of the existing sketch is 1048576 as shown in my original post.
The new sketch is an exact copy of the existing sketch with just a variable changed for the sketch version number. 
Eg. String sketchVersion = "RELAY-100";  would be changed to "RELAY-101" in the new sketch.
So the IDE size of the new sketch is also 1048576.

But when I try the update I get the following
Sketch url = http://192.168.0.22/php/updateCheck.php
ESP Magic size = 4194304
Update status = 0
HTTP_UPDATE_FAILED Error (-107): New Binary Does Not Fit Flash Size

All that the php code does is look at the existing sketch version number and serves up a newer sketch version if one is available.

Here is the updateCheck.php code ..... can you see any problem with it ?

//	check if php script has been called by esp8266HttpUpdate
if(!check_header('HTTP_USER_AGENT', 'ESP8266-http-Update')) {
    header($_SERVER["SERVER_PROTOCOL"].' 403 Forbidden', true, 403);
    echo "Invalid USER AGENT. This is only for ESP8266 updater!\n";
    exit();
}

//	Confirm existance of HTTP headers from ESP8266
if(
    !check_header('HTTP_X_ESP8266_STA_MAC') ||
    !check_header('HTTP_X_ESP8266_AP_MAC') ||
    !check_header('HTTP_X_ESP8266_FREE_SPACE') ||
    !check_header('HTTP_X_ESP8266_SKETCH_SIZE') ||
    !check_header('HTTP_X_ESP8266_SKETCH_MD5') ||
    !check_header('HTTP_X_ESP8266_CHIP_SIZE') ||
    !check_header('HTTP_X_ESP8266_SDK_VERSION')
) {
    header($_SERVER["SERVER_PROTOCOL"].' 403 Forbidden - Missing header value.', true, 403);
    exit();
}

// get sketch filename and version passed by esp8266HttpUpdate ... Eg "RELAY-100"
$currentVersion = $_SERVER['HTTP_X_ESP8266_VERSION'];
if($currentVersion != "") {
	$currentVerLength = strlen($currentVersion);
	$minusChr = strpos($currentVersion,"-");
	$currentVerName = substr($currentVersion,0,$minusChr);
	$currentVerVersion = substr($currentVersion,$minusChr+1,$currentVerLength);
	$intCurrentVersion = (int)$currentVerVersion;
}
else {
	echo "Requires sketch version name-version \n";
    exit();
}

// get dir list of binaries folder
$sketchPath = "../binaries/";
$sketches =  scandir($sketchPath);
$i = 2;
while($sketches[$i] != '') {
	// get each sketch name and version number from dir list
	$sketchLength = strlen($sketches[$i]);
	$minusChr = strpos($sketches[$i],"-");
	$sketchName = substr($sketches[$i],0,$minusChr);
	$sketchVersion = substr($sketches[$i],$minusChr+1,3);
	$intSketchVersion = (int)$sketchVersion;
	
	// if sketch version is newer then upload it
	if($sketchName == $currentVerName) {
		if($intSketchVersion > $intCurrentVersion) {
			sendFile($sketchPath . $sketches[$i]);
		}
	}
	$i++;
}

//	Function to test existance of header value
function check_header($name, $value = false) {
    if(!isset($_SERVER[$name])) {
        return false;
    }
    if($value && $_SERVER[$name] != $value) {
        return false;
    }
    return true;
}

//	Function to send file to ESP8266
function sendFile($path) {
    header($_SERVER["SERVER_PROTOCOL"].' 200 OK', true, 200);
    header('Content-Type: application/octet-stream', true);
    header('Content-Disposition: attachment; filename='.basename($path));
    header('Content-Length: '.filesize($path), true);
    header('x-MD5: '.md5_file($path), true);
    readfile($path);
}
	
?>

earlephilhower, 2019-01-22:
Closing as this is a Sonoff environment issue and seems to have been solved by @Eszartek .

closed 2019-01-22 by earlephilhower