esp8266/Arduino #4709: 2.4.1 seems to be SLOW

GeorgeDLake, 2018-05-05:
Hi.
I just reinstalled my PC and am using Arduino IDE 1.8.5 and the ESP8266 2.4.1 board lib.
I recompiled an app and loaded it to the LoLin board. 
All the code seems to bee running very SLOW.
I noticed a few new options under tools and not sure if this is the problem.
Board: NodeMCU 1.0 ESP-12E
Flash Size 4M (1M SPIFFS)
Debug Port: "Disabled"
Debug Level: None
IwIP Varieant "v2 Higher Bandwidth"
CPU Frequency 160Mhz
Upload Speed 115200

Thanks.

WereCatf, 2018-05-05:
"It feels slow" isn't a quantifiable assessment. Provide some actual data that shows the difference, if there really is any.

GeorgeDLake, 2018-05-05:
what took less then 1 second now takes up to 2. 
A simple Mqqt message that on reception would set D3 to High, do an alarm.delay(250) and then set D3 to Low, now the Led is on for more than 1 second. 
I downgraded to 2.3 and all is back to normal.

GeorgeDLake, 2018-05-06:
yes. 
once WiFi is connected and all the setup function as ran, it's slow. I am guessing its some lib that I have.
#include <TimeLib.h>
#include <TimeAlarms.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <WiFiUDP.h>
#include <Nextion.h>
#include <NextionHotspot.h>
#include <NextionNumber.h>
#include <NextionText.h>
#include <NextionVariableString.h>
#include <NextionVariableNumeric.h>
#include <NextionPicture.h>
#include <NextionHotspot.h>
#include <SoftwareSerial.h>
#include "DHT.h"
#include <PubSubClient.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>

GeorgeDLake, 2018-05-06:
One has to be the problem. Once I have time.... I will make some app that will just do some random thing that runs slow on 2.4.1 and start to eliminate lib by lib until I find what it is.

TD-er, 2018-05-06:
You may also want to add some logging to indicate wifi status and if MQTT is reconnecting.
MQTT trying to connect may slow down the system quite a bit and if that's frequently being done, it may slow down all.
Also trying to resolve a hostname or making a connection that will result in a timeout will also take quite some resources.

devyte, 2018-05-06:
I haven't seen any slowness in the core code, except for establishing secure connections, which is expected.
I don't see any evidence of a core issue here. I suggest discussing at a forum until you reach a conclusion, and if you find a core issue, then open an issue here.
Closing.

closed 2018-05-06 by devyte