esp8266/Arduino #4711: WiFiClient client.connect fails when switching from one SSID to another

royseberg, 2018-05-06:
Two separate ESP 12F's are set up as servers to provide temperatures from two different locations. They can both be accessed indepently from any web browser using the url 192.168.4.1 Now I try using a 3rd ESP 12F to display the data from the other two on a Nokia 5110. My sketch uses a switching scheme to access the first ESP 12F then connect with url 192.168.4.1, display the data temperature, which works fine, then disconnect from this one, access the second SSID which works up to this point, but when the client.connect is repeated on the second pass to get the other temperature, it fails, no matter what. Any suggestions?

//++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
//  To display temperatures from 2 servers in basement and
//  kitchen on Nokia 5110
//------------------------------------------------------ Apr 30 2018
#include <ESP8266WiFi.h>
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_PCD8544.h>
Adafruit_PCD8544 display = Adafruit_PCD8544(2, 5, 4); //DC, CE, RST (MOSI=DIN on pin 13, SCLK on pin 14)
char* ssid;
bool flip = false; //toggle flag to switch from one SSID to the other

void setup() {
  display.begin();
  display.setContrast(60);
  display.setTextSize(1);
  display.setTextColor(BLACK);
  display.setCursor(0,0);
  display.clearDisplay();
  Serial.begin(9600);
  Serial.setDebugOutput(true);
  WiFi.mode(WIFI_STA);//  station mode
}

unsigned long startWait;
const char* host  = "192.168.4.1";
const int httpPort = 80;
String url;

void loop() {
if(!flip)ssid= "ESP 01";//if the flag is 0 start with basement server
else ssid = "ESP 12F";  //if flag is 1 switch to kitchen
WiFi.begin(ssid);// connect to either one of the two ESP8266 servers (no password)
if(!flip){
  display.setCursor(0,0);
  display.clearDisplay();
  }
startWait = millis();
while ((WiFi.status() != WL_CONNECTED) && (millis() - startWait) < 5000 ){
    delay(500);
     Serial.print(".");   
    }
    url=(String)WiFi.localIP();
    if(flip) display.print("Ktch ");
    else  display.print("Bsmt ");
    display.display();
    WiFiClient client;
     if(!client.connect(host, httpPort)){
         display.print("fail");//this always fails on second pass
        display.display();
        flip=!flip;
        return;//back to begining of loop()
     }
   ////  send the request to the server (it works without url also)
    client.print(String("GET /")  + url+ " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Connection: close\r\n" + "r\n");
    startWait=millis();
    while(!client.available() && (millis()-startWait) < 5000);
    if(client.available()){
       String line = client.readStringUntil('\r');
       String val=line.substring(0,2);//get only the first 2 characters 0,1
       display.print(val);
       display.println(" F");
       display.display();
       client.stop();
      }
flip=!flip;
WiFi.disconnect();
}

P.S. I tried a scheme where I save to EEPROM the temperature data from the 1st Server  then do a ESP.restart() to access the second one and get the other temperature, recall the first one from EEPROM and then display them both. But is there any way to solve the problem without rebooting?

aerlon, 2018-05-13:
If you are using the latest master branch you should try with lwIP v1.4. (can be changed in the Tools menu of Arduino IDE)
If that does not work you may also want to try using the latest release version (2.4.1) instead of the latest master branch.
If that does not work you may want to add `WiFi.mode(WIFI_OFF); WiFi.mode(WIFI_STA);` to the client code just before whenever your client switches AP.
I'm reporting a similar problem, so let us know if the steps above solves things for you (then we probably have the same bug).

royseberg, 2018-05-15:
Thank for your help. The solution that works is the first one you suggested; changing to IwIP Variant v1.4 Prebuilt. It did the trick.Thanks once again. I spent countless hours trying to make it work.

aerlon, 2018-05-16:
You're welcome! Glad it helped.

aerlon, 2018-08-25:
@royseberg I think core release 2.4.2 should fix your problems, even when using lwIP v2. Would it be possible for you to verify this and close this issue if it is fixed? That would help keep the number of open issues down.

closed 2018-09-04 by royseberg