esp8266/Arduino #4712: HTTP Server: when the client answer slowly ESP-12E crashes and doesn't send data

calfizzi, 2018-05-07:
**Basic Infos**
- [X] This issue complies with the issue POLICY doc.
- [X] I have read the documentation at readthedocs and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

**Platform**
Hardware:	[ESP-12E]
Core Version:	[latest git hash]
Development Env:	[Arduino IDE]
Operating System:	[Windows]

**Settings in IDE**
Module: [Nodemcu]
Flash Mode:	[qio]
Flash Size: [2MB]
lwip Variant: [v2 lower]
Reset Method:	[ck|nodemcu]
Flash Frequency:	[40Mhz]
CPU Frequency:	[80Mhz]
Upload Using:	[SERIAL]
Upload Speed: [115200] (serial upload only)

**Problem Description**
I tried to build an example for you of my issue.
I built a simple program that answer via HTTP showing the "Romeo And Juliet" romanze as web page. If you compile and run the code on your ESP-12E you can easily connect to its web server on port 80. It will response the complete text of the R&J romanze as HTML page. 

To better understand the issue, I created a _**SendSubChunk_P**_ function that send the HTML chunks of the exact size of available write buffer at the moment.

If you try to connect by a slow dispositive like a 3G mobile, while data transfer the wifi connection of ESP-12E goes down. Some time it blocks as it is runing in a  infinite loop. I also created a simple C# client to simulate a slow internet connection and the result is the same

**my example Sketch:**
[HTTP_Issue_ESP8266.zip](https://github.com/esp8266/Arduino/files/1980427/HTTP_Issue_ESP8266.zip)

**Simple C# Project:**
[HttpClient.zip](https://github.com/esp8266/Arduino/files/1980429/HttpClient.zip)

Please let me know if I'm doing something wrong or if the issue really exists!

Best Regards,
Mario Calfizzi

d-a-v, 2018-05-07:
> If you try to connect by a slow dispositive like a 3G mobile, while data transfer the wifi connection of ESP-12E goes down. Some time it blocks as it is runing in a infinite loop.

Can you please be a little more specific ?

calfizzi, 2018-05-07:
Sure!
My english isn't perfect.. 
I'm using Client.write_P([bytes], [bytes_size]) method it returns number of bytes sent.
During the write_p  command execution the ESP 12E stops, some time for few seconds and some time forever. 
If you are in a good case and it stops only for few second checking the write available buffer it is 0; some seconds after the WiFi connection goes down ( the WiFi.status() responds WL_DISCONNECTED).

This issue happens only when the client browser has a slow internet connection as 3G. 

To verify my hypothesis I created a simple C# test program that connect directly the ESP board downloading the HTML and it simulate a slow connection. 

If you compile and run it, you will find on the dialog the delay field (default = 10ms). 
If you set it to 70 or 80ms and click on connection to download the html, looking on the esp serial terminal you will check what is happening... In the 80% of the cases it stops and the WiFi connection goes down.
The same thing happens connecting the esp by mobile phone using 3G connection.

**I reported here my log showed on esp serial terminal:**
 
~~~
[Client connected]
GET / HTTP/1.1
Host: 192.168.0.254
Accept: text/html, application/x-ms-application, image/jpeg, application/xaml+xml, image/gif, image/pjpeg, application/x-ms-xbap, */*
Accept-Language: en-US
User-Agent: Self Agent
Accept-Encoding: gzip, deflate
Connection: Keep-Alive
  Status= 4  Avail For Write= 877 (36D) HTML_ChunkSizeSent = 89  Count => 89/89
Sent 89 On 89 (59) 
  Status= 4  Avail For Write= 977 (3D1) HTML_ChunkSizeSent = 977  Count => 977/4499
  Status= 4  Avail For Write= 624 (270) HTML_ChunkSizeSent = 624  Count => 1601/4499
  Status= 4  Avail For Write= 970 (3CA) HTML_ChunkSizeSent = 970  Count => 2571/4499
  Status= 4  Avail For Write= 624 (270) HTML_ChunkSizeSent = 624  Count => 3195/4499
  Status= 4  Avail For Write= 970 (3CA) HTML_ChunkSizeSent = 970  Count => 4165/4499
  Status= 4  Avail For Write= 624 (270) HTML_ChunkSizeSent = 334  Count => 4499/4499
Sent 4499 On 4499 (1193) 
  Status= 4  Avail For Write= 724 (2D4) HTML_ChunkSizeSent = 724  Count => 724/15844
  Status= 4  Avail For Write= 877 (36D) HTML_ChunkSizeSent = 877  Count => 1601/15844
  Status= 4  Avail For Write= 717 (2CD) HTML_ChunkSizeSent = 717  Count => 2318/15844
  Status= 4  Avail For Write= 1065 (429) HTML_ChunkSizeSent = 1065  Count => 3383/15844
  Status= 4  Avail For Write= 534 (216) HTML_ChunkSizeSent = 534  Count => 3917/15844
  Status= 4  Avail For Write= 1060 (424) HTML_ChunkSizeSent = 1060  Count => 4977/15844
  Status= 4  Avail For Write= 1070 (42E) HTML_ChunkSizeSent = 1070  Count => 6047/15844
  Status= 4  Avail For Write= 531 (213) HTML_ChunkSizeSent = 531  Count => 6578/15844
  Status= 4  Avail For Write= 1063 (427) HTML_ChunkSizeSent = 1063  Count => 7641/15844
  Status= 4  Avail For Write= 531 (213) HTML_ChunkSizeSent = 531  Count => 8172/15844
  Status= 4  Avail For Write= 527 (20F) HTML_ChunkSizeSent = 527  Count => 8699/15844
  Status= 4  Avail For Write= 0 (0) HTML_ChunkSizeSent = 0
  Status= 4  Avail For Write= 0 (0) HTML_ChunkSizeSent = 0
...
...
  Status= 4  Avail For Write= 0 (0) HTML_ChunkSizeSent = 0
ERROR: Sent 0 Of 15844
ERROR [SendChunk_P]: Client Lost Connection
ERROR [SendChunk_P]: WiFi Lost Connection
ERROR [SendChunk_P] Code: 6
[Client disconnected]
ERROR [HandleConnection]: WiFi Lost Connection
ERROR [HandleConnection] Code: 6
Wifi NOT Connected
~~~
If you need more help please contact me again.

Mario

d-a-v, 2018-05-08:
Thanks. I will try to reproduce to find out what is happening.

In the meantime you can try
* other lwIP options (v2 more bandwidth, v1.4 more bandwidth)
* enabling CORE debug options (you can watch packets flowing)

calfizzi, 2018-05-08:
> other lwIP options (v2 more bandwidth, v1.4 more bandwidth)
enabling CORE debug options (you can watch packets flowing)<

Please, 
may you explain how I can set the above choices?

Thanks and best regards,
Mario

d-a-v, 2018-05-12:
It's in the Arduno IDE's `Tool` menu. `lwIP variant` and `Debug Port/Level` entries.

calfizzi, 2018-05-12:
This is the configuration:
[image]

With the flag "v1.4 Higher bandwidth" I've got a lot of compiler errors...
So I used "v2 Higher bandwidth"!

Here the serial log (mixed with my log):
~~~
SDK:2.2.1(cfd48f3)/Core:2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
bcn 0
del if1
usl
mode : sta(ec:fa:bc:12:df:1c)
add if0
STUB: dhcp_stop
MYESP
MDNS started
wifi evt: 8
....wifi evt: 2
.scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with MONKEY, channel 13
ip:192.168.0.254,mask:255.255.255.0,gw:192.168.0.1
ip:192.168.0.254,mask:255.255.255.0,gw:192.168.0.1
wifi evt: 0
wifi evt: 3
:ur 1
.
Wifi Connected
Local IP  : 192.168.0.254
gateway   : 192.168.0.1
SubnetMask: 255.255.255.0
macAddress: EC:FA:BC:12:DF:1C
WS:ac
:ref 1
WS:av
:ref 2
:ur 2

[Client connected]
:rn 278
GET / HTTP/1.1
Host: 192.168.0.254
Accept: text/html, application/x-ms-application, image/jpeg, application/xaml+xml, image/gif, image/pjpeg, application/x-ms-xbap, */*
Accept-Language: en-US
User-Agent: Self Agent
Accept-Encoding: gzip, deflate
Connection: Keep-Alive
:wr 195 195 0
:wrc 195 195 0
  Status= 4  Avail For Write= 2725 (AA5) HTML_ChunkSizeSent = 89:wr 4 4 0
:wrc 4 4 0
:wr 89 89 0
:wrc 89 89 0
:wr 2 2 0
:wrc 2 2 0
  Count => 89/89
:sent 195
Sent 89 On 89 (59) 
  Status= 4  Avail For Write= 2825 (B09) HTML_ChunkSizeSent = 2048:wr 5 5 0
:wrc 5 5 0
:wr 2048 2048 0
:wrc 256 2048 0
:wrc 256 1792 0
[... 396 lines cut ...]
ERROR [SendChunk_P]: Client Lost Connection
ERROR [SendChunk_P]: WiFi Lost Connection
ERROR [SendChunk_P] Code: 6
[Client disconnected]
:ur 1
WS:dis
:del
ERROR [HandleConnection]: WiFi Lost Connection
ERROR [HandleConnection] Code: 6
Wifi NOT Connected
scandone
del if0
usl

 ets Jan  8 2013,rst cause:2, boot mode:(3,3)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
~~~

Mario

calfizzi, 2018-05-12:
in another case, sending complete HTML defined chunk, I got this error:
~~~
Status= 4  HTML_ChunkSize= 6685 
:wr 6 6 0
:wrc 6 6 0
:wr 1227 6685 0
:wrc 256 1227 0
:wrc 256 971 0
:wrc 256 715 0
:wrc 256 459 0
:wrc 203 203 0
:urn 77
:urd 9, 77, 13
:urd 4, 77, 23
:urd 11, 77, 28
:wr 0 5458 1227
:sent 1460
:sent 1460
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
:er -13 0x3fff15ac
wifi evt: 1
STA disconnect: 200
:ur 1
:wtmo
ERROR: Sent 1227 On 6685 (1A1D) 
ERROR [SendChunk_P]: WiFi Lost Connection
ERROR [SendChunk_P] Code: 6

[Client disconnected]
:ur 1
WS:dis
:del
ERROR [HandleConnection]: WiFi Lost Connection
ERROR [HandleConnection] Code: 6
Wifi NOT Connected
scandone
del if0
usl

 ets Jan  8 2013,rst cause:2, boot mode:(1,2)
~~~
it seems while it is sending big data something happen and it stops sending with this message:
~~~
:urn 77
:urd 9, 77, 13
:urd 4, 77, 23
:urd 11, 77, 28
:wr 0 5458 1227
~~~

Best Regards,
Mario

d-a-v, 2018-05-17:
I have tried your sketch in STA mode and it seems to work.

In fact it fails always the same way, but not like the way you describe (this seems to be only a sketch bug). My logs are below.

Regarding your setup, there is at least this `:wtmo` which is a message from our core and stands for "Write Timeout" (it is by default set to 5 seconds).
There are also lots of phy-rmware messages indicating poor WiFi connectivity, and that explains the core-write-timeout.
```
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
wifi evt: 1
STA disconnect: 200
```

You have or are simulating a slow link, but the logs show a poor-quality WiFi link.
Regarding the slow-link situation only, you have the opportunity to increase the timeout with WiFiClient::setTimeout(milliseconds) (5000 by default, in fact in Stream::).

Sketch logs here:
```
[...]
14:46:22.590 ->   Status= 4  Avail For Write= 1072 (430) HTML_ChunkSizeSent = 857  Count => 15844/15844
14:46:22.656 -> Sent 15844 On 15844 (3DE4) 
14:46:22.656 ->   Status= 4  Avail For Write= 1072 (430) HTML_ChunkSizeSent = 1072  Count => 1072/7064
14:46:22.656 ->   Status= 4  Avail For Write= 1072 (430) HTML_ChunkSizeSent = 1072  Count => 2144/7064
14:46:22.656 ->   Status= 4  Avail For Write= 1072 (430) HTML_ChunkSizeSent = 1072  Count => 3216/7064
14:46:22.689 ->   Status= 4  Avail For Write= 1072 (430) HTML_ChunkSizeSent = 1072  Count => 4288/7064
14:46:22.689 ->   Status= 4  Avail For Write= 1072 (430) HTML_ChunkSizeSent = 1072  Count => 5360/7064
14:46:22.722 ->   Status= 4  Avail For Write= 1072 (430) HTML_ChunkSizeSent = 1072  Count => 6432/7064
14:46:22.756 ->   Status= 4  Avail For Write= 1072 (430) HTML_ChunkSizeSent = 632  Count => 7064/7064
14:46:22.756 -> Sent 7064 On 7064 (1B98) 
14:46:22.756 -> ERROR [SendChunk_P]: Client Lost Connection
14:46:22.855 -> [Client disconnected]
```

on client's side `wget`'s tcp connection is closed everytime at offset 244683:
```
$ wget http://10.43.1.117
--2018-05-17 14:46:11--  http://10.43.1.117/
Connecting to 10.43.1.117:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: 'index.html.1'

index.html.1                   [                   <=>                ] 238.95K  64.3KB/s    in 3.8s    

2018-05-17 14:46:15 (63.7 KB/s) - Read error at byte 244683 (Connection reset by peer).Retrying.

--2018-05-17 14:46:16--  (try: 2)  http://10.43.1.117/
Connecting to 10.43.1.117:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: 'index.html.1'

index.html.1                   [                   <=>                ] 238.95K  62.8KB/s    in 3.8s    

2018-05-17 14:46:20 (62.8 KB/s) - Read error at byte 244683 (Connection reset by peer).Retrying.

--2018-05-17 14:46:22--  (try: 3)  http://10.43.1.117/
Connecting to 10.43.1.117:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: 'index.html.1'

index.html.1                   [  <=>                                 ]  24.66K  60.7KB/s
^C
```

calfizzi, 2018-05-17:
Hi @d-a-v ,
before all thanks!

it's very strange have a  "poor-quality WiFi link" because my netgear router is far only 2meters!
Releted to the "setTimeout" I already tried it but the only change was I had to wait more time to get the  "Connetcion Failure" message.

I tried also in AP mode and it works fine. The problem is releated only to the STA mode.

if you take look to this parts of my log:
~~~
  Status= 4  Avail For Write= 590 (24E) HTML_ChunkSizeSent = 262:
wr 5 5 0
:wrc 5 5 0
:wr 262 262 0
:wrc 256 262 0
:wrc 6 6 0
:wr 2 2 0
:wrc 2 2 0
~~~

and 

~~~ 
Sent 7622 On 7622 (1DC6) 
  Status= 4  Avail For Write= 321 (141) HTML_ChunkSizeSent = 321
:wr 5 5 0
:wrc 5 5 0
:wr 316 321 0
:wrc 256 316 0
:wrc 60 60 0
:wr 0 5 316
:wr 0 5 316
:wr 0 5 316
:wr 0 5 316
:wr 0 5 316
:wr 0 5 316
:wr 0 5 316
~~~
If I understood how the debug log works
:wrc 5 5 0 means the ESP is sending 5 bytes of residue 5 bytes
in the first case the SW want send 262bytes
and the ESP send it in the following order
```
Status= 4  Avail For Write= 590 (24E) HTML_ChunkSizeSent = 262
wrc 5 5 0  --> I think are tcp bytes
wrc 256 262 0 --> chunk of SW bytes (256 bytes of residue 262)
wrc 6 6 0  --> chunk of SW bytes (6 bytes of residue 6)
                       so 256 + 6 = 262 bytes sent.... For me it makes sense
:wrc 2 2 0 --> the last  wrc indicates other 2 bytes sent i believe are for TCP communication
```
but in second case looking to the SW bytes to send:

```
Status= 4  Avail For Write= 321 (141) HTML_ChunkSizeSent = 321
:wrc 5 5 0 --> I supposed it was tcp bytes but it subtrats them from the SW bytes count
:wrc 256 316 0 --> chunk of SW bytes should be 256 bytes of residue 321 and not 316!
:wrc 60 60 0
:wr 0 5 316
```
always after the above situation the ESP lose connection!

I still believe is an esp issue!

Looking to second log I posted:

```
Status= 4  HTML_ChunkSize= 6685 
:wr 6 6 0
:wrc 6 6 0
:wr 1227 6685 0
:wrc 256 1227 0
:wrc 256 971 0
:wrc 256 715 0
:wrc 256 459 0
:wrc 203 203 0
:urn 77
:urd 9, 77, 13
:urd 4, 77, 23
:urd 11, 77, 28
:wr 0 5458 1227
:sent 1460
:sent 1460
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
pm close 7
:er -13 0x3fff15ac
wifi evt: 1
STA disconnect: 200
```
Analyzing each wrc:

:wrc 6 6 0   --> I think are tcp Bytes (but in before example ware 5bytes probably is different because the esp is chunking the total bytes)
:wr 1227 6685 0 --> it wants send 1227bytes of 6685
:wrc 256 1227 0 --> each block is about 256 bytes
:wrc 256 971 0  --> each block is about 256 bytes
:wrc 256 715 0 --> each block is about 256 bytes
:wrc 256 459 0 --> each block is about 256 bytes
:wrc 203 203 0 --> and finally all 1227 bytes are sent, now sholud start with the next residued bytes but it crashes
:urn 77
:urd 9, 77, 13
:urd 4, 77, 23
:urd 11, 77, 28
:wr 0 5458 1227

I still believe is an esp issue!

Did you try to use my c# example to simulate slow connection? 
(because if I access to the ESP using fast connection it works fine!)

Mario

d-a-v, 2018-05-17:
> it's very strange have a "poor-quality WiFi link" because my netgear router is far only 2meters!

I agree. This is my interpretation of `rm 0` (= for sure the STA "netif" interface is removed from lwIP by the phy-rmware) and also `STA disconnect: 200`. Neither your sketch nor the core causes that. WiFi is lost at that point.

What I would try is to disable power management mode with `WiFi.setSleepMode(WIFI_NONE_SLEEP);` and see if things improve. The last firmware became more sensitive with power management and espressif is apparently working [on this](https://github.com/espressif/ESP8266_NONOS_SDK/commit/236b6d0adb80eb52ab4f5d86a62cdb4d58f0daae#diff-3d8510b5f68b1ce4e40ea4515fff75e5).

I could not try your c# example (maybe because I don't have any c# interpreter right under my hand).

Regarding the communication trace itself, you noticed that data are sent by chunks of 256 bytes. That's the way WiFiClient is currently working. lwIP is fed by these chunk and buffers them. When lwIP has the right to send all that gathered data, then a real packet is sent in the air. You don't see the real packets, but you can (with #4678 or) with tcpdump/wireshark on your PC.

Regarding `:wrc`, `:wr` and others, they live in `libraries/ESP8266WiFi/src/include/ClientContext.h`. `:wr` is what we want to send, `:wrc` is what is given to lwIP. Every single byte here is yours (not TCP management). `:u*` is for UDP. This piece of code is old, solid and has proven working well. It handles all TCP data from the core.

calfizzi, 2018-05-18:
> I could not try your c# example (maybe because I don't have any c# interpreter right under my hand).

Do you that want I give you the ".exe" of my test program to better verify the issue?
I can build it for you!

---
> Every single byte here is yours (not TCP management). 

Also your affirmation is very strange for me, because in the log below:
```
Status= 4  Avail For Write= 590 (24E) HTML_ChunkSizeSent = 262
wrc 5 5 0  --> I think are tcp bytes
wrc 256 262 0 --> chunk of SW bytes (256 bytes of residue 262)
wrc 6 6 0  --> chunk of SW bytes (6 bytes of residue 6)
                       so 256 + 6 = 262 bytes sent.... For me it makes sense
:wrc 2 2 0 --> the last  wrc indicates other 2 bytes sent i believe are for TCP communication
```
what my SW send are 262 bytes; take look to the second line: wrc 256 262 0...
In change, the previous line sends 5 bytes (wrc 5 5 0) and the latest line sends ther 2 bytes more (wrc 2 2 0) ; I think aren't mine because the total of bytes I'm sending is 262 as wrote in my log (HTML_ChunkSizeSent = 262) so if you count the total amount of bytes sent is 269, 7 bytes more of my amount

Let me know if my doubt is justified!

Mario

calfizzi, 2018-05-18:
Ok @d-a-v ,
> WiFiClient does not add more data to yours, every single bytes you can see with the debug info are yours. 

about bytes sent, I'll check the total amount of bytes sent starting from the beginning!

> because TCP cannot be working if the radio link goes down

I don't beleave the WiFi connection goes down for signal losing, because this case happen only when the client download slowly. If the client is fast I never loose the WiFi connection.
I will try using WIFI_NONE_SLEEP  as you suggested....

...And I'll try to create a c/c++ program to be used by command row to simulate a slow connection...
(I hope I'm able to do it, I never have worked on C/C++ sockets :-) )

Mario

d-a-v, 2018-05-18:
I agree that loosing WiFi because of a slow connection is rather strange. From what I can remember I have not seen that before.
About your tester, anything that is not exclusively microsoft will do :)

calfizzi, 2018-05-18:
I'm at office and I cannot try the esp here but looking at my code... You were right, the bytes more are sended by my code. Is the HTML chunked protocol... before sending a chunk need to send the size of following block in Hex format followed by Cariage return and Line Feed. At the end of the block need to send the Cariage return and Line Feed.
When I'l go back home I'll try your suggestions...

Mario

calfizzi, 2018-05-19:
Hi @d-a-v ,
using WIFI_NONE_SLEEP  as you suggested the esp doesn't crash any more but not all data are sended...
 
I created the ESP web connection test program for **linux**.
I created a makfile for build it. You have to type "make" in the Test_Linux folder.

To use it you have to write:
`./test http://(your IP or your Address [/d(dalay  you want]`

using  /d20 (20ms) as delay the ESP work fine but
using  /d80 (80ms) as delay the ESP for long time cannot send information (available buffer too small) and the last data of my html page are lost; my page should end with the following lines:
```
----------------------- END -----------------------<br>
----------------------- END -----------------------<br>
----------------------- END -----------------------<br>

12
```

This is the file:
[Test_Linux.zip](https://github.com/esp8266/Arduino/files/2019666/Test_Linux.zip)

Regards,
Mario

calfizzi, 2018-05-20:
Hi @d-a-v ,
This is the ESP source code modified that doesn't crash.
In that way you can execute the test as the mine:
[TEST_BUG_ESP8266.zip](https://github.com/esp8266/Arduino/files/2020183/TEST_BUG_ESP8266.zip)

d-a-v, 2018-05-23:
I have been trying to reproduce today, with no luck.

I could trigger again the `:wtmo` by reducing the PC receive window and I saw the `win 0` in tcpdump, but this window receive size later increased and the connection could go on. I have no explanation why it did not work yesterday (PC's `win` staying at 0).

What happened when I reproduced your bug is not the ESP fault. The PC receive window is controled by the PC (or your 3G phone) and the ESP is complying with the TCP protocol by not sending data because it has not the right to do so.

Are you able to run wireshark on your windows PC and observe `win` field of packets going from your PC to the ESP ? It reflects the amount of data that the ESP is authorized to send in the next packet.
Another test would be to use #4678 (tcpdump inside the ESP, not merged yet) and observe the same field when using your 3G phone.

About WiFi disconnecting, solved by `WIFI_NONE_SLEEP`, you can also remove it and try instead [#2330(comment)](https://github.com/esp8266/Arduino/issues/2330#issuecomment-387709161).

skarcha, 2018-11-27:
I've also noted this issue using ESPurna on my ESP at home and connecting to it over an OpenVPN connection from my phone. If the connection is slow, something goes wrong and WD reset the ESP.

devyte, 2020-11-25:
Closing due to age and lack of feedback.

closed 2020-11-25 by devyte