esp8266/Arduino #4713: consider this printf implementation with stdio fopencookie function

jandrassy, 2018-05-08:
```
ssize_t adapterWrite(void* p, const char *buf, size_t n) {
  return ((Print*) p) -> write(buf, n);
}

FILE* openAdapter(void *p) {
  cookie_io_functions_t fncs; 
  fncs.write = adapterWrite; 
  fncs.close = NULL; 
  return fopencookie(p, "w", fncs); 
}

size_t Print::printf(const char *fmt, ...) {
  va_list args;
  va_start(args, fmt);
  FILE* adapter = openAdapter(this);
  size_t len = vfprintf(adapter, fmt, args);
  fclose(adapter);
  va_end(args);
  return len;
}

size_t Print::printf(const __FlashStringHelper *fmt, ...) {
  size_t fmtLen = strlen_P((PGM_P) fmt);
  char format[fmtLen + 1];
  strcpy_P(format, (PGM_P) fmt);
  va_list args;
  va_start(args, fmt);
  FILE* adapter = openAdapter(this);
  size_t len = vfprintf(adapter, format, args);
  fclose(adapter);
  va_end(args);
  return len;
}
```

jandrassy, 2019-01-17:
compare it to current implementation

devyte, 2019-01-17:
@jandrassy I also don't understand the intent here. If you provide a more detailed explanation and examples we can discus it.

u48, 2019-02-09:
Hi.Ф

The solution is correct, it will work fine.

I have already added a completely similar way of printf() in arduino STM32.
Works on STM32, ESP8266, DUE, Raspbery.

In my profile there is a fork Arduino Stm32. In the branch for Stm32F1 there is already a ready source print.с

Bonus, for Arduino Uno, too, there is a similar solution.