esp8266/Arduino #4714: My new esp8266/cp2102 doesn't work.

AndrewPDK, 2018-05-08:
I have recently orderdered the board from: https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20180508074019&SearchText=1pcs+Wireless+module+NodeMcu+Lua+WIFI+Internet+of+ 
I have tried to reflashed it twice. The firmware was: ESP8266_Deauther_v2.0.5_4MB.bin   I noticed that the module is very hot during operation. It has worked only 2 days.  I have alreday tried to connect it to my computer but I can't program my module. It seems to me that the module died due to the overheating. 
The current source was power bank and it  gives an output of 5v, while is recommended 3.3 v. 
Can you help to find the problem. 

#### Platform

- Hardware:					 ESP8285 device
- Core Version:			                 [2.0.5]
- Development Env:		                 [Arduino IDE]
- Operating System:		                 [Windows]

### Settings in IDE

- Module:  			ESP8266 Module | Nodemcu|
- Flash Mode:			[dio]
- Flash Size: 		        [4MB]
- lwip Variant:                   [Higher Bandwidth]
- Reset Method:		nodemcu
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		[OTA|SERIAL]
- Upload Speed:                [230400]

kapyaar, 2018-05-08:
Module getting hot in itself is a red flag. I would recommend
1. Check if the 3.3V reg output is indeed 3.3V. If there is a burned unit that draws current, it is likely this voltage will be lower. This would also explain the heat.

2. If 3.3v looks ok, then is the board being recognized when you plug in to pc? It is not unusual that you may burn a board or two in the process. Got scar tissues :)

AndrewPDK, 2018-05-08:
Yes, the board is recognized when I plug in to pc. I have already tried to program the board in Arduino IDE also in XTCOM_UTIL, unfortunately, the programs doesn't see the board. Futhermore, the diods don't blink.

kapyaar, 2018-05-08:
Hmm, it has all the signs of a burnt board :( What is the VDD status?

AndrewPDK, 2018-05-09:
I have 3 new board. I have an idea to connect the voltage regulator to the
board with a capacitor. How I can prevent the overheating and board death?

8 мая 2018 г. 23:33 пользователь "kapyaar" <notifications@github.com>
написал:

Hmm, it has all the signs of a burnt board :( What is the VDD status?

—
or mute
the thread