esp8266/Arduino #4715: Wrong #define LED_BUILTIN for NodeMCU board

TridentTD, 2018-05-09:
### Basic Infos
   ESP8266 Arduino Core 2.4.1 has a mistake (?) #define LED_BUILTIN
for NodeMCU board  to pin gpio16 , then when using BLINK example ,
it can't blink the LED_BUILTIN

https://github.com/esp8266/Arduino/blob/master/variants/nodemcu/pins_arduino.h#L35

It should be defined to pin gpio 2  ?

#### Platform

- NodeMCU 1.0 (ESP12F)
- Core version 2.4.1
- ArduinoIDE
- Windows10

### Settings in IDE

- NodeMCU
- 4MB
- lwip V2 & V1.4

NobleAscent, 2018-06-30:
This issue is previously covered but I guess they haven't decided to do anything about it. :cry:

lrmoreno007, 2018-07-01:
The official Nodemcu has the integrated led connected to the GPIO16 pin (with inverse logic by the way). 

The led in GPIO2 to which you refer, is integrated into the ESP12e and this is integrated into Nodemcu. But be careful because not all ESP12e cards have this led. 

Therefore, LED_BUITIN in GPIO16 is correct.

https://encrypted-tbn0.gstatic.com/images?q=tbn:ANd9GcQMdY72MOGomZsb1h_nGYNivWbEVOq_hBC34VSR3-S0n3MWpIws

tablatronix, 2018-08-22:
@lrmoreno007  is correct, the led on an esp12e is tied to io2/txd1 and is used for com ind, meaning it will blink if you use serial2. io16 is the correct definition for nodemcu.

5chufti, 2018-08-23:
@devyte since defines seem to be correct, shouldn't this be closed?

devyte, 2018-08-23:
I'm confused. My understanding is that for all boards based on the esp12 reference hw the builtin led is gpio2. Yes, it blinks when using serial2, but it also blinks e.g. with a dht22 shield on a wemos d1 r2.
I also understand that some nodemcu devices have an extra led, which is NOT called builtin, and in the case.of noddemcu it's on gpio16.
Then, the builtin define was changed from gpio2 to gpio16, which is the current state. And now there is a request to change it back,  because the blink example doesn't work on that board.
So which is it? Please don't make me dig out my old nodemcu boards to answer this myself, they're somewhere behind 15 months worth of storage.

tablatronix, 2018-08-24:
Hmm it was changed at some point?

tablatronix, 2018-08-24:
blink works fine on nodemcu (amica), and a wemos v3 pro
maybe they are using some lolin nodemcu without a second led and using the wrong board definition.

Either way it looks fine to me ( unless we add LED_BUILTIN_2 at some point )

IMHO boards with an aux led should be mapped to LED_BUILTIN and ones without should be mapped to esp led. Sounds reasonable to me

d-a-v, 2018-08-24:
@TridentTD @NobleAscent Could you elaborate ?
Does your nodemcu board have a led on pin 16 ?

tablatronix, 2018-08-24:
Fyi same goes for adafruit huzzah it has 2 leds builtin defined for aux led io 0

5chufti, 2018-08-24:
I have only cheap "replicas" of nodemcu boards but even the cheapest without ESP-12 module (esp directly on pcb w/o metal can) has both leds...

devyte, 2018-08-25:
Are there any of these boards that don't have a led on gpio2?

tablatronix, 2018-08-25:
Every dev board I know of uses 12e or 12f so no, but there is a lot of boards. I can start looking

devyte, 2018-08-25:
@tablatronix If all esp12-based boards have a led on gpio2, then I'd think that led should be the builtin one. That makes it consistent and assures that the blink example will always work, putting this recurring topic to rest.
If some boards have another additional led, then those variants should have an extra define. Call it AUX, per your comment above.

If there is any variant that does not have a led on gpio2, then we can discuss further what to do.

@d-a-v any thoughts here?

DaleSchultz, 2019-02-19:
LED_BUILTIN does **not** work on my **NodeMCU** boards
LED_BUILTIN is defined as 16 and does not activate an LED, neither HIGH nor LOW
D4 (defined as 2) is connected to a second(?) LED on the NodeMCU next to the antenna, using reverse logic. LOW turns the LED on.

d-a-v, 2019-02-20:
https://github.com/nodemcu/nodemcu-devkit
there is a led on gpio16

DaleSchultz, 2019-02-20:
LED_BUILTIN is defined as 16 and does **not** activate an LED, neither HIGH nor LOW

This is what people keep reporting. 

GPIO16 does NOT activate an on board LED on Multiple NodeMCU boards that I bought recently. I don't know how to tell what version they are but they use the CH340 drivers and have "HW-625" silkscreened near the antenna. The chip is marked ESP8266MOD.

d-a-v, 2019-02-20:
Can you propose a pull request to change led definition to 2 ?

DaleSchultz, 2019-02-20:
I am not sure how to do that, nor do I feel confident about submitting a change. I was confirming what has been reported and seemingly dismissed.

tablatronix, 2019-02-20:
The official nodemcu boards have a led on gpio 16, maybe you are using some lolin verison or something.

dsyleixa, 2019-10-20:
I can confirm that LED_BUILIN for ESP12E nodeMCU is GPIO 16 == D0:

```
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(115200);
  delay(2000);
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.println();
  Serial.print("LED_BUILTIN=");Serial.print(LED_BUILTIN);
  Serial.println();
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  Serial.print(LED_BUILTIN); Serial.print("="); Serial.println("HIGH");
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  Serial.print(LED_BUILTIN); Serial.print("="); Serial.println("LOW");
  delay(1000);                       // wait for a second
}

```

output:

> 
> LED_BUILTIN=16
> 
> 16=HIGH
> 16=LOW
> 16=HIGH
> 16=LOW
> 16=HIGH
> 16=LOW
> 16=HIGH
> 16=LOW
> 16=HIGH
> 16=LOW
> 16=HIGH
> 16=LOW
> 16=HIGH
> 
>

closed 2019-11-10 by devyte

dsyleixa, 2019-11-10:
I have to disagree:
as stated, LED_BUILTIN for ESP-12E nodeMCU is GPIO 16 == D0, and this is true for all of my 6 ESP-12E Boards!
By GPIO 2 nothing happens, neither LED is flashing.

devyte, 2019-11-10:
There are boards with a led on 2, boards with a led on 16, and boards with both, all called nodemcu. I'm defining the builtin as 2 to match other boards based on the esp12 that have the led. If your board has a led on 16, please use the aux.

tablatronix, 2019-11-10:
That should work , although there is only one real official nodemcu.

mhightower83, 2019-11-10:
An observation of what I think I am reading here and what I have seen, there appears to be an ESP-12E _development board_ containing an ESP-12E _module_. The ESP-12E _module_ (and its various clones) sometimes have an LED on GPIO-2. The LED would appear to be optional at least on some of the cloned _modules_.  The [schematic reference above of a _development board_](https://github.com/nodemcu/nodemcu-devkit) has an LED on GPIO-16. Last, and frustratingly the [schematic](https://components101.com/sites/default/files/component_datasheet/ESP12E%20Datasheet.pdf#page=18) in the datasheet for the ESP-12E _module_ fails to show there is an LED on the _module_; however, a [reversed engineered schematic](http://ejenn.free.fr/images/Esp8266_esp_12e_schematics.jpg) shows it on GPIO-2.

d-a-v, 2019-11-10:
There are two possibilities:
- two board entries for nodeMCU-v1.0, one for gpio2 and the other for gpio16
- a single entry for nodeMCU-v1.0, but a led menu like with generic board

dsyleixa, 2019-11-10:
IMO there already far too many entries in the boards men, so I would be fine with option b)

devyte, 2019-11-10:
@dyslexia there is a filtering capability now to only show boards you're interested in.

dsyleixa, 2019-11-10:
@devyte 
thanks, neither heard of it - nor found after having searched for it

mhightower83, 2019-11-10:
@dsyleixa, devyte is referring to this [How to create an abridged boards.txt file](https://arduino-esp8266.readthedocs.io/en/latest/faq/a05-board-generator.html#how-to-create-an-abridged-boards-txt-file) it is in the FAQ. 

Also for the non-ESP8266 boards.txt files, that are too long, you can manually edit those down to the boards you use. Note, you cannot delete all of them. There must be at least one board in a boards.txt file. Don't forget to make a backup copy.

dsyleixa, 2019-11-10:
@mhightower83 
thank you, nice to see that but too cumbersome for general Arduino users

mhightower83, 2019-11-10:
@dsyleixa ?? only the 1st two commands are needed for what you want. You can do copy-paste and edit the favorites file. The rest are additional things that can be done. Or are you referring to the lake of a GUI interface?

dsyleixa, 2019-11-10:
well, this is not the topic here, but I doubt that me other common Arduino users would ever do that.

PR #6743 (2019-11-10): Puts to rest the confusion about the builtin leds for nodemcu boards
Given that most nodemcu boards are based on the ESP12 boards, the definition of the builtin led is now changed to 2.
In addition, for those nodemcu boards that have an additional LED on the board connected to gpio16, an additional define is added called LED_BUILTIN_AUX.
Fixes #4715