esp8266/Arduino #4716: ESP.getFreeHeap() cause wdt reset

rkoptev, 2018-05-09:
### Basic Infos

- [yes] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [yes] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [no] I have tested that the issue is present in current master branch (aka latest git).
- [yes] I have searched the issue tracker for a similar issue.
- [yes] If there is a stack dump, I have decoded it.
- [no] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[NodeMCU 1.0 (ESP-12E Module)]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600]

### Problem Description

In my project in loop section I have a snippet that responsible for periodical log of free RAM to monitor its usage in the runtime. But when I make a request to ESPAsyncWebServer to get a static html from SPIFFS - almost always this leads to an exception and wdt reset when ESP.getFreeHeap() gets called.

I noticed mentions about yield in stack trace and I guess this may be somehow related to this issue: https://github.com/me-no-dev/ESPAsyncWebServer/issues/289

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <FS.h>
#include <DNSServer.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include "AudioFileSourceSPIFFS.h"
#include "AudioGeneratorWAV.h"
#include "AudioOutputI2S.h"
#include "IPAddress.h"
#include "painlessMesh.h"
#include "Hash.h"
#include <ESPAsyncTCP.h>
#include "ESPAsyncWebServer.h"
#include "Base64_user.h"
#include "FastLED.h"

...

void setup() {
  ...
}

void loop() {
...
  if (millis() - lps_request_time > 3000) {
    //Serial.printf("LPS = %d\n", loops_count);
    Serial.printf("Available RAM: %d\n", ESP.getFreeHeap()); // Exception appears here
    request_for_lps = false;
    loops_count = 0;
    lps_request_time = millis();
  }
...
}

```

### Debug Messages

```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
Decoding 12 results
0x4023fd75: umm_info at /Users/rkoptev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1012
0x4010020c: _umm_free at /Users/rkoptev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1291
0x4021a574: esp_yield at /Users/rkoptev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 57
0x4023fe88: umm_free_heap_size at /Users/rkoptev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/umm_malloc/umm_malloc.c line 1745
0x40206f0c: xPortGetFreeHeapSize at /Users/rkoptev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/heap.c line 162
0x40249851: system_get_free_heap_size at ?? line ?
0x40218688: EspClass::getFreeHeap() at /Users/rkoptev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/Esp.cpp line 418
0x40218b65: HardwareSerial::available() at /Users/rkoptev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/HardwareSerial.cpp line 188
0x40207bd9: loop at /Users/rkoptev/rkoptev/upwork/HealthCare Futurists GmbH/Hotspot/Firmware/Firmware.ino line 192
0x4021a5c0: loop_wrapper at /Users/rkoptev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at /Users/rkoptev/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/cont.S line 109

```

d-a-v, 2018-05-10:
Your code is probably modifying memory outside from what is allocated by malloc/new to you.
So the allocator (`umm_*`) gets garbage and goes to hell.
Try to disable portions of your code, one by one, in order to isolate and fix the error.
You can also enable the CORE debug option, this enables umm sanity checks so the error would happen sooner for easier debugging.

devyte, 2018-05-11:
My app uses the Async libs and ESP.getFreeHeap(), and I don't have this issue.
I agree with @d-a-v : this is likely something in the user app, not in the core.
The sketch is incomplete, and can't reproduce. Closing.

closed 2018-05-11 by devyte

proddy, 2019-06-08:
for the record, I have the same issue with 2.5.2. Calling ESP.getFreeHeap() randomly causes a reset. I'll see if I can trace the root cause.

dxzl, 2020-06-08:
Do not call ESP.getFreeHeap() between Preferences.h begin and end statements or you will get continuous watchdog timer resetting. (I am using the Arduino IDE for ESP32)

Pablo2048, 2020-06-08:
There is no library like `Preferences.h` for ESP8266 AFAIK...