esp8266/Arduino #4719: BasicOTA example does not work

samlewis02, 2018-05-10:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					Wemos D1 mini]
- Core Version:			[latest]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]
- Upload Speed:   [921600] (serial upload only)

### Problem Description

After loading BasicOTA from examples the esp8266 port appears in ports and I can ping the device at its IP address. However when I try uploading the same file over wifi I just get the message
19:21:29 [ERROR]: No response from device
When I enable  OTA debugging and use Putty to monitor debug, the debug shows that the Wemos is seeing the update request but I cannot see why the other errors occur

### Sketch

BasicOTA,ino from ArduinoOTA examples

### Debug Messages
THIS IS Putty OUTPUT

 OTA server at: esp8266-53d41e.local:8266
                                                                        Ready
IP address: 192.168.1.101
pm open,type:2 0
                sleep disable
                             Start updating sketch
Connect Failed
              Error[2]: Connect Failed
Error[4]: End Failed
ERROR[0]: No Error
**************** the following repeats each time I try upload over wifi
sleep disable
             Start updating sketch
Connect Failed
              Error[2]: Connect Failed
Error[4]: End Failed
ERROR[0]: No Error

Pablo2048, 2018-05-10:
Can You please check if there is "hole" in the firewall? IMHO this kind of errors is because of Windows firewall block.

Pablo2048, 2018-05-10:
Ping is ICMP protocol (nothing to do with OTA), and Putty is just TCP client. Can You please try OTA with disabled firewall?

WereCatf, 2018-05-10:
@samlewis02 Did you remove power from the ESP after flashing the OTA-sketch? You have to remove power, then reapply it before OTA starts to work -- it's because something gets stuck in the hardware whenever you flash a sketch via serial and resetting the hardware fixes it.

samlewis02, 2018-05-11:
I edited the following table which originally showed Firewall On for PPPoE (all others were off). Still get exactly the same response when I try wifi update. I tried another Wemos and same result.
I think you must be right about it being related to router setting as the problem only occurred when my broadband company installed a new router. Is there another setting which might be relevant?
Sam

網際網路 - 網際網路設定

Wide Area Network (WAN) Setup
 
 
VLAN ID
Index
Service
Interface
Connection Type
Igmp
NAT
Firewall
State
IPv6
MLD
Remove
Edit
Off
1
NMS
ptm0.1
IPoE
Off
Off
Off
On
Disabled
Disabled

Off
2
br-1
ptm0.2
Bridge
Off
Off
Off
On
Disabled
Disabled

Off
4
ipoe-1
ptm0.4
IPoE
Off
On
Off
On
Disabled
Disabled

Off
3
pppoe-1
ppp0.3
PPPoE
Off
On
Off
On
Enabled
Disabled

 

From: Pablo2048
Sent: 10 May 2018 23:08
To: esp8266/Arduino
Cc: samlewis02; Author
Subject: Re: [esp8266/Arduino] BasicOTA example does not work (#4719)

Ping is ICMP protocol (nothing to do with OTA), and Putty is just TCP client. Can You please try OTA with disabled firewall?

samlewis02, 2018-05-11:
Yes, tried reset, power off/on but same result

From: WereCatf
Sent: 10 May 2018 23:15
To: esp8266/Arduino
Cc: samlewis02; Mention
Subject: Re: [esp8266/Arduino] BasicOTA example does not work (#4719)

@samlewis02 Did you remove power from the ESP after flashing the OTA-sketch? You have to remove power, then reapply it before OTA starts to work -- it's because something gets stuck in the hardware whenever you flash a sketch via serial and resetting the hardware fixes it.

samlewis02, 2018-05-11:
I think the table in last email might have been messed up. I will try again.

網際網路 - 網際網路設定Wide Area Network (WAN) Setup  
VLAN IDIndexServiceInterfaceConnection TypeIgmpNATFirewallStateIPv6MLDRemoveEditOff1NMSptm0.1IPoEOffOffOffOnDisabledDisabledOff2br-1ptm0.2BridgeOffOffOffOnDisabledDisabledOff4ipoe-1ptm0.4IPoEOffOnOffOnDisabledDisabledOff3pppoe-1ppp0.3PPPoEOffOnOffOnEnabledDisabled
 

From: Pablo2048
Sent: 10 May 2018 23:08
To: esp8266/Arduino
Cc: samlewis02; Author
Subject: Re: [esp8266/Arduino] BasicOTA example does not work (#4719)

Ping is ICMP protocol (nothing to do with OTA), and Putty is just TCP client. Can You please try OTA with disabled firewall?

Pablo2048, 2018-05-11:
Can You please check AP Isolation mode on Your AP (https://www.techwalla.com/articles/what-is-ap-isolation) ?

devyte, 2018-05-11:
Not a core issue, closing.

closed 2018-05-11 by devyte

buzzy, 2018-10-12:
Doing a OTA update will open a port on your computer. Then it sends a request to the esp8266 to download the image from your computer and flash it. That you are able to ping the esp8266 is not enough. The esp8266 also needs to be able to reach your computer on the temporary opened port to download the image. Probably you have a firewall that prevents the esp8266 from doing that download. Disable your firewall (and maybe antivirus) and try again.

petisoft, 2019-01-05:
A common problem is the selected network profile when joining the network on a Windows PC, i.e. as "private" or "public" as different firewall settings are applied. If connected as "public", the firewall is (typically) configured more restrictive and may not allow the ESP8266 OTA to connect back to the PC.  This is also true for different Wifi networks, e.g. 2.4GHz and 5GHz.

balassy, 2019-01-11:
In my case it was my antivirus software, Eset Internet Security that blocked the traffic of python.exe and ESP8266. After turning that blocking off upload works like a charm.

Dishendramishra, 2019-05-28:
In my case also, I have to disable firewall, on xubuntu i used
`sudo ufw disable`
then log out and login again

LeshaVH, 2019-07-19:
Just allow Pyhon.exe to use network in Firewall Settings

GrzegorzKaczan, 2020-02-23:
> Ping is ICMP protocol (nothing to do with OTA), and Putty is just TCP client. Can You please try OTA with disabled firewall?

It was the firewall for me, disabled and works like a charm.