esp8266/Arduino #4720: MDNSResponder crashes ESP on malformed MDNS packet

drschlaumeier, 2018-05-12:
ESP8266 generic, 1M/128k, lwip v2 lower mem, 80Mhz, DIO
core 2.4.0
DevEnv Arduino IDE or Vmicro
OS Win10

### Problem Description
I installed TPLINK KASSA APP on my smartphone to control my LB130 smart bulp.
Suddently (some seconds or latest 1-2min), all my ESP8266 on my network went into reboot.
I can reproduce: when Kassa App is open my esp's reboot after a short while.

I checled with wireshark analysis that TPLINK KASSA APP sends malformed MDNS packets according to wireshark:
0000   00 1a 11 00 00 02 00 1a 11 00 00 01 08 00 45 00  ..............E.
0010   00 20 05 56 40 00 01 11 89 73 0a 08 00 01 e0 00  . .V@....s......
0020   00 fb 8b 92 14 e9 00 0c 59 9c **0a ba 10 00**        ........Y.....

I have following code in my project:

```
#include <ESP8266mDNS.h>
MDNSResponder mdns;
...
if (mdns.begin(DEVICENAME.c_str(), WiFi.localIP())) {
  Serial.println(F("MDNS responder started"));
}
...
mdns.addService(F("http"), F("tcp"), 80); 
...
```

The malformed packets let mdns responder crash.
When I sending this into my network the esp's with mdns responder will reboot...try yourself:
`while true; do echo -ne "\x0a\xba\x10\x00" | ncat -u 224.0.0.251 5353; done`

Everything is fine (no reboot) if I disable mdns responder.

Is it a BUG or something wrong on my code?

Thanks, Rgds

arihantdaga, 2018-05-22:
I am facing the same with TP Link Kasa app :o

devyte, 2018-12-05:
Closing in view of #5442 with a full rewrite of mdns.

closed 2018-12-05 by devyte