esp8266/Arduino #4722: ESP-12 & Arduino IDE 1.8.5 ESP8266 2.4.1 & Mac: software is not starting anymore

tzahari, 2018-05-12:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).

*I tried, but steps from the documentation are not working with Arduino IDE 1.8.5 and Mac OS 10.13.4*

- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12E
- Core Version:			2.4.1
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio or dio
- Flash Size:                     4MB/1MB
- lwip Variant:                  v2 Lower Memory
- Reset Method:		none,ck
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

I wrote a program but this was not starting anymore. Flashing is no problem. 
Then I tried so simplify everything. Now I trired it with this small test wich is also not starting. 
After that I tried it with the esptool.py and the helloworld-esp8266.bin wich was then starting.  
So I think that something is wrong with the binary or the flashing process.  

I tried to use ESP8266 version 
- 2.4.0
- 2.3.0

The documentation about the git version is not working with the Mac. No ESPBoards are present then.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
int i = 0;

void setup() {
 Serial.begin(74880);
  
 Serial.println("\n\nESP12E Modul Test"); 
}

void loop() {
  i++;
  Serial.println(i);
  delay(1000);
}
```

### Debug Messages

```

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

```

tzahari, 2018-05-18:
Testet now with a second ESP-12 and also with Arduino 1.6.13

d-a-v, 2018-05-18:
Have you tried the "erase flash"  option in tools menu ?

tzahari, 2018-05-18:
Yes. Also with the esptool.py.
The hello world example from the esptools.py also works.

d-a-v, 2018-05-18:
> The documentation about the git version is not working with the Mac. No ESPBoards are present then.

Can you elaborate ?
Does it mean that you are not able to try with current git version ?

It looks like you have an installation issue. Reinstalling from scratch, removing everything including `.arduino15/`, preferably the git version, will help.

tzahari, 2018-05-18:
Hey d-a-v,
Thanks for the help. 
I deleted everything and also the ~/Library/Arduino15 folder and reinstalled everything.
Also I installed the git version of esp8266.
But still with the same issue.

d-a-v, 2018-05-18:
Then there is a number of possible reasons. Your power supply, your flash mode...
Check #2414

tzahari, 2018-06-08:
@d-a-v thanks a lot. 
I cleared the breadboard and started from scratch. Now it is working again.
But I knew it was working before and I did not changed anything. 
Very unusual.

closed 2018-06-08 by tzahari