esp8266/Arduino #4726: Can't be installed successfully

lukeweston, 2018-05-14:
Arduino IDE Version: 1.8.5
Mac OSX 10.12.6
Python: 2.7.10
Which ESP8266 hardware I'm using is irrelevant, I'm not really up to that point.
I've shown the specific git clone path and board manager JSON file URL, below.

I have inserted the following entry into the additional board manager URLs:
`http://arduino.esp8266.com/stable/package_esp8266com_index.json
`
The ESP8266 entry is now visible for installation in the Board Manager GUI.
I've attempted to install, with v 2.4.1 selected. (Selecting 2.4.0 does not fix the issue.)

Installation fails with the following error message in red text:
`Error downloading https://github.com/esp8266/Arduino/releases/download/2.4.1/esp8266-2.4.1.zip
`

OK, that doesn't work, so let's try manually installing it.
I don't mind manual installation, as long as it gets the job done in a usable way.
(Remember this is OSX, so we're inside /Applications/Arduino.app/)
`cd /Applications/Arduino.app/Contents/Java
`
`cd hardware
`
`mkdir esp8266com
`
`cd esp8266com
`
`git clone https://github.com/esp8266/Arduino.git esp8266
`
`cd esp8266/tools
`
So far so good.

`get.py` fails. I'm not sure what Python packages etc, if any, are prerequisites for this script. No such requirements, beyond Python 2.7 itself, are documented on the documentation page.

Python is installed. (2.7.10)

I have followed the instructions as they are given on the documentation page, for both the manual install-from-git and the Board Manager cases, and in neither case can I get a successful installation.

```
$ python get.py
Traceback (most recent call last):
  File "get.py", line 15, in <module>
    import tarfile
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/tarfile.py", line 52, in <module>
    import copy
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/copy.py", line 52, in <module>
    import weakref
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
```

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

tzahari, 2018-05-14:
Hello @lukeweston ,
I have also a Mac but with Mac OS X 10.13.4 and here works the installation via board manager.

Thanks to clarify the manual installation method. I tried it to start in the ~/Library/Arduino15/ folder but this was not working. 

Best Regards
Thomas

lukeweston, 2018-05-15:
Just because I did it that way, don't assume that it's correct or the best way to do it :)

d-a-v, 2018-05-18:
It seems that python 2.7.14 solves the import thing.

lukeweston, 2018-05-22:
After upgrading to python 2.7.14, I still get a failure, a different failure, of that download script.

`$ python ./get.py
Platform: x86_64-apple-darwin
Downloading esptool-0.4.13-osx.tar.gz
Traceback (most recent call last):
  File "./get.py", line 116, in <module>
    get_tool(tool)
  File "./get.py", line 75, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 443, in open_https
    h.endheaders(data)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1038, in endheaders
    self._send_output(message_body)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 882, in _send_output
    self.send(msg)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 844, in send
    self.connect()
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1263, in connect
    server_hostname=server_hostname)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 363, in wrap_socket
    _context=self)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 611, in __init__
    self.do_handshake()
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 840, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:661)
`

Projectsaathi01, 2018-06-09:
I tried but it can't be installed on window 
[image]

earlephilhower, 2019-05-01:
These errors seem to be completely SSL cert related (as if your specific system doesn't trust the CA that GitHub (now MSFT, I guess, so may be different from before) is using.  I've even tried building the very old Python 2.7.14 and am presently able to run the get.py without any errors at all.

Closing as wontfix since there's nothing we can do on our end.

closed 2019-05-01 by earlephilhower