esp8266/Arduino #4727: lwIP2 Network Switching Bug

aerlon, 2018-05-14:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			
SDK:2.2.1(cfd48f3)/Core:win-2.5.0-dev/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-7-g2b827f8)
SDK:2.2.1(cfd48f3)/Core:win-2.5.0-dev/lwIP:1.4.0rc2
SDK:2.2.1(cfd48f3)/Core:win-2.4.1/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1)
SDK:2.2.1(cfd48f3)/Core:win-2.4.1/lwIP:1.4.0rc2

- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU 1.0
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

This bug has two shapes depending on which Arduino core for ESP8266 version is used:

- Using the latest master branch and lwIP2: The station in lwIP2_NetworkSwitchingBug.ino (download below) becomes unable to connect to any WiFi network after the first it connects to. Only if `WiFi.mode(WIFI_OFF)` is called first will a network switch be possible. This is not a problem when using lwIP1.4.

- Using Arduino core for ESP8266 version 2.4.1 with lwIP2: The station in lwIP2_NetworkSwitchingBug.ino (download below) becomes unable to connect to any server except the first it connects to. It can however still connect to any WiFi network. `WiFi.mode(WIFI_OFF)` does not help with this situation. This is not a problem when using lwIP1.4.

**Instructions for reproducing bug:**
1. Download and unzip the .ino files in [lwIP2_NetworkSwitchingBug.zip](https://github.com/esp8266/Arduino/files/2000549/lwIP2_NetworkSwitchingBug.zip).
2. Make sure you are using either Arduino core for ESP8266 version 2.4.1 or the latest master branch.
3. Make sure you are using lwIP1.4.
4. Add AP1.ino to one ESP8266, AP2.ino to another and lwIP2_NetworkSwitchingBug.ino to a third.
5. Observe serial output from lwIP2_NetworkSwitchingBug.ino.
6. Power off the AP that lwIP2_NetworkSwitchingBug.ino connected to.
7. Observe serial output from lwIP2_NetworkSwitchingBug.ino.
8. Power up the disconnected AP.
9. Re-upload lwIP2_NetworkSwitchingBug.ino using lwIP2.
10. Observe serial output from lwIP2_NetworkSwitchingBug.ino.
11. Power off the AP that lwIP2_NetworkSwitchingBug.ino connected to.
12. Observe serial output from lwIP2_NetworkSwitchingBug.ino.
13. Optional: If the latest master branch is used, uncomment `WiFi.mode(WIFI_OFF); WiFi.mode(WIFI_STA);` at line 12 of lwIP2_NetworkSwitchingBug.ino and re-start from step 7.

Note: This may be a more detailed description of issue #4711 .

d-a-v, 2018-06-18:
Thanks for reporting, can you please try this:
```
cd tools/sdk/lwip2
make latestmaster
```
I will be doing more tests before making a PR for it.

(this is the same fix as for #4792)

aerlon, 2018-06-19:
I can confirm that this lwIP2 update seems to solve both the issue reported here and the related issue in PR #4718. WiFI connection did seem a bit unstable in PR #4718 with the lwIP2 fix applied, but since the fix is WIP I guess that may be expected.

d-a-v, 2018-06-19:
Instability is not expected with this fix, any detail is appreciated.

aerlon, 2018-06-26:
Closing this since issue #4843 is now fixed, so there are no more (known) problems with the bug fix.

closed 2018-06-26 by aerlon