esp8266/Arduino #4731: com port not found ESP8266

cheezzer, 2018-05-14:
Hi All,

Apologies if this issue has been queried and resolved as I couldn't find it on this forum. 

**Problem:** 
I've connected the (out of the box) ESP8266 directly to my PC. When checking the Arduino IDE. However, the Port is greyed out. I've confirmed in Device Manager as well. 

I'm running Win10 Build 1803
Arduino IDE 1.8.5

Any help would be E much appreciated.

5chufti, 2018-05-15:
this is not a forum. This is a issue tracker for problems within the esp8266 arduino IDE core.
your problem is with either your arduino IDE or OS (driver) installation.
So please close this issue and look at a more appropriate forum (e.g. www.arduino.cc www.esp8266.com)

devyte, 2018-05-16:
Previous comment is correct. Closing.

closed 2018-05-16 by devyte