esp8266/Arduino #4735: WiFi softAP mode crashes when Tenda Nova MW6 in range

chlous, 2018-05-16:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			ESP8266
- Core Version:			2.4.0
- Development Env:		Visual Micro
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 			2MB (256k Spiffs)
- lwip Variant:   		v1.4
- Reset Method:		ck
- Flash Frequency:		40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		Serial
- Upload Speed:   		921600

### Problem Description

After setting to softAP mode, the ESP crashes after a couple of seconds, whenever a specific mesh wifi ap is in range! We have hundreds of installations, but never seen anything alike before. 

This happens only with the following model is in range:
**Tenda Nova MW6 (Whole Home Mesh WiFi System)**

After removing the power of the access point, the softAP mode is running as excpected.

Any idea how to proceed with this problem? How can i deliver more information? It seems that this access point is sending specific packets that leads to the crash.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

void setup()
{
	WiFi.mode(WIFI_AP);

	IPAddress local_IP(10, 11, 0, 100);
	IPAddress gateway(10, 11, 0, 1);
	IPAddress subnet(255, 255, 255, 0);
	WiFi.softAPConfig(local_IP, gateway, subnet);
	WiFi.softAP("ESPsoftAP_01");
}

void loop()
{
	...
}

```

### Debug Messages

```
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
```

devyte, 2018-06-17:
@chlous this sounds ominously like an Espressif sdk bug. Unfortunately, this is the first I've ever even heard of that Tenda Nova device.
My suggestion to you: look into the sdk examples, and try to reproduce this in an application outside this core. Also, ou might want to open an issue in the NONOS espressif repo.
BTW, the reason I mention reproducing this in an example based on a pure sdk app is that I read somewhere that Espressif has a bounty on bugs found in their SDK code. I could be wrong, or that could no longer be true, but it's worth looking into.
Closing for now, because I don't think there's anything that can be done in this repo, but I'll reopen if further details prove that to be wrong.

closed 2018-06-17 by devyte

araczkowski, 2019-01-13:
I have the same problem with Tenda Nova **MW3**. I'm curious how it is possible that softAP on ESP is crashing by router? The router is sending something to AP and this is the reason of a crash?

Anybody has any idea how to bypass this?

PS
I'm facing this problem only on core version 2.3.0, Wemos D1 mini and Sonoff's devices with Tasmota