esp8266/Arduino #4738: BearSSL::WiFiClientSecure

artua, 2018-05-17:
#### Platform

- Hardware:				[ESP-12]
- Core Version:			[Mon May 14 20:46:47 2018 -0700]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

BearSSL::WiFiClientSecure does not connect. No debug messages at all (debug tls+ssl+http client). My custom logs attached.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <WiFiClientSecure.h>

void setup() {
 BearSSL::WiFiClientSecure client;
 if (!client.connect(host, httpsPort)) {
    res.code = 0; res.content = F("connection failed");
    return (res); // 1 = connection failed
  }
}

void loop() {

}

```

### Debug Messages

```
00:01:374 Free HEAP: 36264
00:01:375 Client Secure started 1 ms
00:01:375 Free HEAP: 31704
00:01:376 connecting to api.monitec.systems
00:01:969 Connection failed 593 ms
00:01:973 Request failed. Reason: (0) connection failed
```

artua, 2018-05-17:
Thanks, I already found client.setInsecure(). 
But it still won't work and I can't get any debug output.

### Debug Messages

```
0:01:284 connecting to api2.monitec.systems
00:02:235 Free HEAP: 10456
00:02:236 Connected 952 ms
00:02:237 fingerprint doesn't match
00:02:238 Certificate checked 1 ms
00:02:239 Free HEAP: 10456
00:02:240 requesting URL: /api/data with data: {"....."}
00:02:309 Request sent 14 ms
00:02:310 Free HEAP: 8840
00:07:312 Result length: 0 
```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch (logger lines removed)

```cpp

BearSSL::WiFiClientSecure client;
  client.setInsecure();
  client.setTimeout(API_TIMEOUT);
  client.connect(host, httpsPort);
  client.print(String("POST ") + url + " HTTP/1.0\r\n" +
               "Host: " + host + "\r\n"
               + "Content-Type: application/json\r\n"
               + "Content-Length: " + data.length() + "\r\n\r\n"
               + data);
  client.setTimeout(API_TIMEOUT);
  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line.startsWith("HTTP/1.")) {
      res.code = line.substring(9, 12).toInt();
    }
    if (line.startsWith("Content-Length: ")) {
      c_len = line.substring(15).toInt();
    }
    if (line == "\r") {
      break;
    }
  }
  if (client.available()) {
    bytes = client.readBytes(c_buf, c_len);
    c_buf[c_len] = '\0';
  }
  res.content = String(c_buf);
  client.stop();
  return (res);
}

artua, 2018-05-17:
Interesting, if I log bytes, it shows:
```
00:07:312 Result length: 0 | 1075938601
```
Like this:
```
  logger("Result length: " + String(res.content.length()) + " | " + String(bytes));

```

earlephilhower, 2018-05-21:
@artua, can you please give a full sketch example that doesn't work and can be publicly accessible?  I'm not able to reproduce it with my own testing...

earlephilhower, 2018-05-22:
Looks like a bug in `client.connected()` in BearSSL.  If you change that `while (client.connected())` to `while (TOUT)` it always succeeds.  Probably reporting client.connected() relative to the TCP link, which may go down before the entire SSL data has been read out.  Let me take a gander...

artua, 2018-05-22:
When it will be merged to master so I can check?

earlephilhower, 2018-05-22:
Please make a branch that has this PR in your own repo and try it out with your test...I'll be waiting on your response before it gets into master. :)

https://help.github.com/articles/checking-out-pull-requests-locally/

artua, 2018-05-22:
@earlephilhower yes, now it's working.
**But still takes 500ms more to connect, so it is unusable on battery-powered devices. :(**

Update: now it gives me timeouts with another host (powered by cloudflare, so using modern ciphers).
Try www.artua.com/bearssl.txt

```
01:22:644 Client Secure started 0 ms
01:22:649 Free HEAP: 40720
01:22:652 connecting to www.artua.com
01:23:986 Free HEAP: 18464
01:23:987 Connected 1330 ms
01:23:988 Free HEAP: 18464
01:23:988 requesting URL: /bearssl.txt with data: {}
01:24:001 Request sent 12 ms
01:24:002 Free HEAP: 16848
01:24:409 Got HTTP code 200
01:24:413 Got Content-length: 21
01:24:426 Headers received 423 ms
01:24:427 Free HEAP: 18464
01:24:428 Result length: 21 | 21
01:24:428 Content received 1 ms
01:24:429 Free HEAP: 18464
01:24:432 Client stop 2 ms
01:24:434 Free HEAP: 40912
01:24:438 
*** Result: {"result":"success"}

01:24:444 Client Secure started 0 ms
01:24:449 Free HEAP: 40976
01:24:453 connecting to www.artua.com
01:25:632 Free HEAP: 18464
01:25:633 Connected 1175 ms
01:25:634 Free HEAP: 18464
01:25:635 requesting URL: /bearssl.txt with data: {}
01:25:648 Request sent 12 ms
01:25:649 Free HEAP: 16848
01:40:649 
*** Timeout receiving headers
```

^^^ here I set timeout to 15000ms

earlephilhower, 2018-05-22:
Can you leave that URL accessible for a while?  I'll try tonight (12hrs or so).  In the meantime, can you add a call to the client.getLastSSLError() and dump that on "timeout" as well as the state of connected()/etc.  The code I saw kind of mushed up socket timeout with !connected() so it's hard to say what you're seeing from the dump.

` int getLastSSLError(char *dest = NULL, size_t len = 0);`

Best I can about the speed is tell you to use 160MHz, not 80MHz (avail from the menus).  There's no HW acceleration of any kind in the system.  Even 32-bit multiply is emulated.  For EC key exchange @Adam5Wu and I have seen multi-second handshakes even at 160MHz.

artua, 2018-05-22:
Sure, it will be accessible.
For your request:
```
*** Timeout receiving headers. SSLError: "0" Connected: 0
```
If I understood it right:
```
    logger("\n*** Timeout receiving headers. SSLError: \"" + String(client.getLastSSLError()) + "\" Connected: " + String(client.connected()));
```

earlephilhower, 2018-05-22:
That'll do.  Can you also run using axTLS and verify that it works?  Might be something related to other bits, that'd passing would help ensure it's related to BSSL.  Thx!

earlephilhower, 2018-05-22:
Try pull #4756 .  You found two bugs in the code!  This second one had to do with the way partial reads from the SSL buffer were handled when the server responds fast and closes down the connection.

I've run against your server and it seems to work all the time.

artua, 2018-05-24:
Now working perfectly. Thanks!

01:38:003 Cycles: 91 Head TOUTs: 0 Body TOUTs: 0
Conn ave: 514
Req ave: 11
Head ave: 513
Body ave: 0
Stop ave: 0
Overall ave: 1038

earlephilhower, 2018-05-24:
@artua, if you have a fixed system you're connecting to, you can use the client.setKnownKey(key) to hardcode the public key of the other system.  That should take connection time down dramatically as it ignores the x509 certificate completely and uses the server public key you code to encrypt things.

Check the BearSSL_Validation example for more info.

Closing this for now.

closed 2018-05-24 by earlephilhower

artua, 2018-06-03:
Hi @earlephilhower 
I was testing performance and found no difference between 
  client.setInsecure();
and 
client.setKnownKey(&key);

Is it ok? Because you told me the last method should dramatically increase TLS handshake...

setInsecure()
```
01:03:582 Cycles: 94 Head TOUTs: 0 Body TOUTs: 0
Conn ave: 567
Req ave: 12
Head ave: 34
Body ave: 1
Stop ave: 1
```
setKnownKey()
```
01:15:466 Cycles: 103 Head TOUTs: 0 Body TOUTs: 0
Conn ave: 565
Req ave: 12
Head ave: 34
Body ave: 1
Stop ave: 1
```

earlephilhower, 2018-06-03:
@artua There's no way in BearSSL of getting a detailed breakdown, but I misspoke so this is expected, actually.  The way a TLS handshake works is that you check the x509 and signing up through the cert chain until you find one you trust, then you use the public key to do some exchange of entropy.  So you still need to do the EC stuff if the public key is EC.

artua, 2018-06-03:
I got something like 120ms with curl on my macbook. I know it is
incomparable in terms of computing speed but maybe you know some way to
speed up the connection? I have access to server configuration also, so I
can set up any cipher/options on server side.

earlephilhower, 2018-06-04:
They're SSL sessions, like HTTP.  It's dependant on the server and client, of course.   As of now the BearSSL::WiFiClientSecure does NOT enable them, as they take some amount of RAM and we're not exactly drowning in available heap.  The session ID is stored as well as the last encryption params by the client, and then it sends it back to the server on a reconnect sometime later who either accepts it (and uses the last settings) or starts a new negotiation.

If there's a real demand I (or someone else!) can add a patch that adds a "bool enableSessions(uint count)" method.  Feel free to open an issue or pull request.  :)  First, though, I want to get the CertStore changes accepted, though, as that's a more common issue since PIO's default is unsafe but common and tripping up folks.

I think axTLS also has the option, but it's also not implemented in the wrapper class.

iamneo2416, 2018-06-12:
I also met same error with client.connect, but I use **connect(IPAddress, port)**.
Whenever I change connect to connect(host_name, port), it works fine!

host_name = "**google.com**", "**api.github.com**", that works, but when I change to 192.168.1.210, it always connect fail !!

I use dos command line to check port 443 with following
**C:\> telnet 192.168.1.210  443**   <== it works. means 443 is listening !

chegewara, 2018-08-28:
Hi, ive been told to use bearssl instead axtls, and here is a problem. BearSSL_Validation example is not connecting with certificates, only works insecure and fingerprints. I am assuming that certificates included should be valid for api.github.com (example host):
```
ip:192.168.43.11,mask:255.255.255.0,gw:192.168.43.1
.
WiFi connected
IP address: 
192.168.43.11

If there are no CAs or insecure options specified, BearSSL will not connect.
Expect the following call to fail as none have been configured.
Trying: api.github.com:443...*** Can't connect. ***
-------

This is absolutely *insecure*, but you can tell BearSSL not to check the
certificate of the server.  In this mode it will accept ANY certificate,
which is subject to man-in-the-middle (MITM) attacks.
Trying: api.github.com:443...Connected!
-------
HTTP/1.1 200 OK
-------

The SHA-1 fingerprint of an X.509 certificate can be used to validate it
instead of the while certificate.  This is not nearly as secure as real
X.509 validation, but is better than nothing.
Trying: api.github.com:443...Connected!
-------
HTTP/1.1 200 OK
-------

It is also possible to accept *any* self-signed certificate.  This is
absolutely insecure as anyone can make a self-signed certificate.
First, try and connect to a badssl.com self-signed website (will fail):
Trying: self-signed.badssl.com:443...*** Can't connect. ***
-------
Now we'll enable self-signed certs (will pass)
Trying: self-signed.badssl.com:443...*** Can't connect. ***
-------

The server certificate can be completely ignored and its public key
hardcoded in your application. This should be secure as the public key
needs to be paired with the private key of the site, which is obviously
private and not shared.  A MITM without the private key would not be
able to establish communications.
Trying: api.github.com:443...*** Can't connect. ***
-------

A specific certification authority can be passed in and used to validate
a chain of certificates from a given server.  These will be validated
using BearSSL's rules, which do NOT include certificate revocation lists.
A specific server's certificate, or your own self-signed root certificate
can also be used.  ESP8266 time needs to be valid for checks to pass as
BearSSL does verify the notValidBefore/After fields.
Try validating without setting the time (should fail)
Trying: api.github.com:443...*** Can't connect. ***
-------
Try again after setting NTP time (should pass)
Waiting for NTP time sync: .
Current time: Tue Aug 28 05:38:00 2018
Trying: api.github.com:443...*** Can't connect. ***
-------
pm open,type:2 0
```

chegewara, 2018-08-28:
@devyte 
Hi, ive been told to use bearssl instead axtls, and here is a problem. BearSSL_Validation example is not connecting with certificates, only works insecure and fingerprints. I am assuming that certificates included should be valid for api.github.com (example host):
```
ip:192.168.43.11,mask:255.255.255.0,gw:192.168.43.1
.
WiFi connected
IP address: 
192.168.43.11

If there are no CAs or insecure options specified, BearSSL will not connect.
Expect the following call to fail as none have been configured.
Trying: api.github.com:443...*** Can't connect. ***
-------

This is absolutely *insecure*, but you can tell BearSSL not to check the
certificate of the server.  In this mode it will accept ANY certificate,
which is subject to man-in-the-middle (MITM) attacks.
Trying: api.github.com:443...Connected!
-------
HTTP/1.1 200 OK
-------

The SHA-1 fingerprint of an X.509 certificate can be used to validate it
instead of the while certificate.  This is not nearly as secure as real
X.509 validation, but is better than nothing.
Trying: api.github.com:443...Connected!
-------
HTTP/1.1 200 OK
-------

It is also possible to accept *any* self-signed certificate.  This is
absolutely insecure as anyone can make a self-signed certificate.
First, try and connect to a badssl.com self-signed website (will fail):
Trying: self-signed.badssl.com:443...*** Can't connect. ***
-------
Now we'll enable self-signed certs (will pass)
Trying: self-signed.badssl.com:443...*** Can't connect. ***
-------

The server certificate can be completely ignored and its public key
hardcoded in your application. This should be secure as the public key
needs to be paired with the private key of the site, which is obviously
private and not shared.  A MITM without the private key would not be
able to establish communications.
Trying: api.github.com:443...*** Can't connect. ***
-------

A specific certification authority can be passed in and used to validate
a chain of certificates from a given server.  These will be validated
using BearSSL's rules, which do NOT include certificate revocation lists.
A specific server's certificate, or your own self-signed root certificate
can also be used.  ESP8266 time needs to be valid for checks to pass as
BearSSL does verify the notValidBefore/After fields.
Try validating without setting the time (should fail)
Trying: api.github.com:443...*** Can't connect. ***
-------
Try again after setting NTP time (should pass)
Waiting for NTP time sync: .
Current time: Tue Aug 28 05:38:00 2018
Trying: api.github.com:443...*** Can't connect. ***
-------
pm open,type:2 0
```

devyte, 2018-09-03:
@marcelstoer @artua please open a new issue to discuss axtls vs. bearssl performance, and add the perf numbers shown here.

marcelstoer, 2018-09-03:
Yep, wanted to do that anyhow.

d-a-v, 2018-09-03:
@earlephilhower Does "larger bandwidth" make a difference ?

artua, 2018-09-05:
@marcelstoer <https://github.com/marcelstoer> thanks! please post a link to
new thread here, so I can participate too.

earlephilhower, 2018-09-05:
@d-a-v Are you suggesting BW tests, too?  Sounds like a good idea.  But it's not really actionable (nor is the connect time) as I'm simply calling the BearSSL internal code to do the real work.

Also, @marcelstoer, it may *look* like BearSSL is using more heap as it allocates everything all at once.  But, in reality, axTLS does lots of heap allocations while it's working on a function call.  So your `getHeapFree()` before and after a `write()` may say report only ~20KB is used, but in reality there may be 4-5KB (I don't know exactly, just examples here) of heap allocation/deallocation going on during the `write()`.  That's the cause of a bunch of crashes w/axTLS, it does not handle OOM conditions (well or at all).

d-a-v, 2018-09-05:
@earlephilhower No, only wondering if you checked the impact of MSS.

JeanVanDenderFlume, 2020-04-18:
> Please check the examples. You have not set any CAs, nor have you asked it to be insecure (i.e. SSL connect but w/o checking certificates). `client.setInsecure()` will do the latter, before any connection.
> 
> Added #4739 to track adding some debug output beyond the GetLastError API.

Do you know or could you say if the line is secure encrypted if you use setInsecure to a backend with a not expited certificate?  I understand setInsecure will not validate the certtifcate at the backend but is it correct the data cannot be read  by anybody? Second how do i read the attributes from the backend certificate, i would like to check the common name and domeinname.

JiriBilek, 2020-04-18:
As to your first question: without checking a certificate the communication is vulnerable to MITM attack. It is encrypted, obviously, but the MITM is able to decrypt it.

PR #4754 (2018-05-22): Report connected when any bytes are available
The SSL pipeline is multi-stage, and the TCP connection can go down
even though there is still data waiting to be decrypted or in the
decryption buffer.

Explicitly check that there if there can be any data made available
to the app, and if so report that we are still connected().  When
there is no data and there is no TCP connection, report disconnected.

Fixes #4738