esp8266/Arduino #4742: WiFi.softAPConfig & WiFi.softAP return 0 but AP will be opened

Pfannex, 2018-05-17:
#### Platform

- Hardware:			ESP12e
- Core Version:			V1.7
- Development Env:		platformio
- Operating System:		WIN10

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			dio
- Flash Size: 		4MB/1MB
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		nodemcu
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Hi, I try to implement a switchable access point. WiFi debug is enabled. I'm wondering why WiFi.softAPConfig and WiFi.softAP returns false while AP ist going startet.
Sometimes WiFi.softAPConfig and WiFi.softAP returns true.

The STA-Mode is enabled an the ESP is connectet to my home network while the AP shoud be switched ON/OFF.

What is the right way to implement a switchable AP in one function, while STA-connection is established?

I simplyfied the Sketch to this function:

### Sketch

```cpp
String startAP(int state) {
  if (state) {
    
    //WiFi.mode(WIFI_AP_STA);
    //delay(1000);
    const char* apSSID     = "MyAccessPoint";
    const char* apPassword = "1234567890";
    IPAddress apIP(192,168,4,1);
    IPAddress gateway(192,168,4,1);
    IPAddress subnet(255,255,255,0);
 
    Serial.println(WiFi.softAPConfig(apIP, gateway, subnet));
    Serial.println(WiFi.softAP(apSSID, apPassword));
    Serial.println(WiFi.softAPIP());

    return "AP is on";
  } else {
    //WiFi.mode(WIFI_STA);
    WiFi.softAPdisconnect(true);
    return "AP is off";
  }
}
```

### Debug Messages

```
[APConfig] local_ip: 192.168.4.1 gateway: 192.168.4.1 subnet: 255.255.255.0
[APConfig] wifi_softap_dhcps_stop failed!
[APConfig] wifi_set_ip_info failed!
[APConfig] DHCP IP start: 192.168.4.100
[APConfig] DHCP IP end: 192.168.4.200
[APConfig] wifi_set_ip_info failed!
[APConfig] wifi_softap_set_dhcps_lease_time failed!
[APConfig] wifi_softap_dhcps_start failed!
[APConfig] IP config Invalid?!
0
[AP] set_config failed!
0
0.0.0.0
AP is on
mode : sta(a0:20:a6:16:a3:22) + softAP(a2:20:a6:16:a3:22)
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
wifi evt: 7
```

devyte, 2018-05-18:
This is not the right place to ask "how do I" questions. Please refer to a community forum.
As an off-the-cuff suggestion, take a look at the ESP8266WiFiAP.h|cpp files.
Closing per issue POLICY doc.

closed 2018-05-18 by devyte

Pfannex, 2018-05-18:
But returning "false" and open the AP anyway should be a issue, isn't it?
Also the Debug reports "[AP] set_config failed!", but I guess the configured values are valid.

tablatronix, 2018-09-05:
why was this closed?

This should clearly not emit "wifi_softap_dhcps_stop failed" errors.

devyte, 2018-09-05:
The issue is phrased as a question about the return type, so I considered it a how do I type issue. In addition, I wasn't able to reproduce this back then across a whole lot of esp12 based devices, which made me think it was a defective device. Nor have I seen the behavior since then in my own app (same sequence, if more complex).

@Pfannex do you still see this with latest release or latest git?

tablatronix, 2018-09-05:
I am not personally seeing this but have open issues with this same symtom , I am attempting to reproduce or create an example.

jeroenst, 2022-07-06:
I experience this issue with 2.4.7.