esp8266/Arduino #4747: Code Crash when trying to use sscanf from new libc

IMAN4K, 2018-05-20:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					`ESP-12`
- Core Version:			`SDK:2.2.1(cfd48f3)/Core:win-2.5.0-dev/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1`
- Development Env:		`Arduino IDE`
- Operating System:		`Windows`

### Settings in IDE

- Module:  			`Nodemcu`
- Flash Mode:			`DIO`
- Flash Size: 		`4MB/1MB`
- lwip Variant:   `v2 Lower Memory`
- Reset Method:		`nodemcu`
- Flash Frequency:	`40Mhz`
- CPU Frequency:		`160MHz`
- Upload Using:		`SERIAL`
- Upload Speed:   `115200`

### Problem Description

Crashing when using simple `sscanf` parsing

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
#include <Arduino.h>
#include <HardwareSerial.h>

extern "C" {
#include <stdio.h>
}

void setup() {
	Serial.begin(115200);

	uint8_t hh, mm, ss;
	if (sscanf("@SetTime:10:18:35$", "@SetTime:%d:%d:%d$", &hh, &mm, &ss))
		Serial.printf("H:%d, M:%d, S:%d\n", hh, mm, ss);
}

void loop() {

}
```

### Stack dump
<details>
<summary>Log</summary>
Fatal exception 9(LoadStoreAlignmentCause):
epc1=0x40208fc8, epc2=0x00000000, epc3=0x00000000, excvaddr=0x3fffffa2, depc=0x00000000

Exception (9):
epc1=0x40208fc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fffffa2 depc=0x00000000

ctx: cont
sp: 3ffff9d0 end: 3fffffd0 offset: 01a0

3ffffb70:  feefeffe feefeffe feefeffe feefeffe
3ffffb80:  feefeffe feefeffe feefeffe feefeffe
3ffffb90:  00000007 3ffe8308 7fffffff 3ffffc3d
3ffffba0:  3ffe85c0 3ffffc3c feefeffe feefeffe
3ffffbb0:  feefeffe feefeffe feefeffe 3ffffc20
3ffffbc0:  3ffffc3e 00000000 3ffffeb0 40208f69
3ffffbd0:  3ffe8e6d 3ffe8e70 3ffe8e73 feefeffe
3ffffbe0:  3ffffc3c fffffea3 3ffe8308 40205bf0
3ffffbf0:  00000002 3ffffef0 feefeffe feefeffe
3ffffc00:  feefeffe feefeffe feefeffe 3ffe8300
3ffffc10:  3ffe8308 3ffe8897 3ffffef0 4020880c
3ffffc20:  00000400 0000000a 0000015b 00000000
3ffffc30:  00000009 3ffffda4 00000003 fe003031
3ffffc40:  feefeffe feefeffe feefeffe feefeffe
3ffffc50:  feefeffe feefeffe feefeffe feefeffe
3ffffc60:  feefeffe feefeffe feefeffe feefeffe
3ffffc70:  feefeffe feefeffe feefeffe feefeffe
3ffffc80:  feefeffe feefeffe feefeffe feefeffe
3ffffc90:  feefeffe feefeffe feefeffe feefeffe
3ffffca0:  feefeffe feefeffe feefeffe feefeffe
3ffffcb0:  feefeffe feefeffe feefeffe feefeffe
3ffffcc0:  feefeffe feefeffe feefeffe feefeffe
3ffffcd0:  feefeffe feefeffe feefeffe feefeffe
3ffffce0:  feefeffe feefeffe feefeffe feefeffe
3ffffcf0:  feefeffe feefeffe feefeffe feefeffe
3ffffd00:  feefeffe feefeffe feefeffe feefeffe
3ffffd10:  feefeffe feefeffe feefeffe feefeffe
3ffffd20:  feefeffe feefeffe feefeffe feefeffe
3ffffd30:  feefeffe feefeffe feefeffe feefeffe
3ffffd40:  feefeffe feefeffe feefeffe feefeffe
3ffffd50:  3ffe9095 4010462f 3ffecaf8 feefeffe
3ffffd60:  40103579 00040000 7fffffff 00000000
3ffffd70:  00000022 40103576 00040000 000000f9
3ffffd80:  3ffed210 401024a2 4010457a 3ffed210
3ffffd90:  3ffed1e0 01366558 53000000 402083f8
3ffffda0:  40208488 00000000 00000000 00000000
3ffffdb0:  00000000 00000000 00000000 00000000
3ffffdc0:  00000000 00000000 00000000 00000000
3ffffdd0:  00000000 01010101 01010101 00000101
3ffffde0:  00000000 00000000 00000000 00000000
3ffffdf0:  00000000 00000000 00000000 00000000
3ffffe00:  00000000 00000000 00000000 00000000
3ffffe10:  00000000 00000000 00000000 00000000
3ffffe20:  00000000 00000000 00000000 00000000
3ffffe30:  00000000 00000000 00000000 00000000
3ffffe40:  00000000 00000000 00000000 00000000
3ffffe50:  00000000 00000000 00000000 00000000
3ffffe60:  00000000 00000000 00000000 00000000
3ffffe70:  00000000 00000000 00000000 00000000
3ffffe80:  00000000 00000000 00000000 00000000
3ffffe90:  00000000 00000000 00000000 00000000
3ffffea0:  00000000 00000a48 00000149 0000000e
3ffffeb0:  3fffff80 3fffff70 0000000c 3ffe8897
3ffffec0:  3ffe8896 00000060 00000060 40201b60
3ffffed0:  00000012 00000020 3fffff40 3ffeeac0
3ffffee0:  3fffdad0 3ffeeb00 3ffe888c 40204a5c
3ffffef0:  3ffe8883 00000007 3fffff70 ffff0204
3fffff00:  3ffe8878 00000012 00000145 3ffeeac0
3fffff10:  3fffdad0 40204b18 3ffeeb00 40202b70
3fffff20:  3ffe8ad0 00000000 3fffff4c 40202dbd
3fffff30:  3ffeeb58 00000157 00000000 401002ac
3fffff40:  3fffdad0 00000001 3ffefb94 40100a88
3fffff50:  00000000 00000000 3fffff80 3fffff70
3fffff60:  00000008 00000001 3ffeeb00 40202cb8
3fffff70:  00000000 00000000 3fffffa2 3fffffa1
3fffff80:  3fffffa0 000019e4 00000000 40202768
3fffff90:  3fffdad0 3ffeeb00 3ffee958 402026b3
3fffffa0:  feefeffe feefeffe feefeffe feefeffe
3fffffb0:  feefeffe 00000000 3ffeeab8 40202a20
3fffffc0:  feefeffe feefeffe 3ffe84fc 401000e5
</details>

### Debug Messages
```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 23 results
0x40208fc8: _scanf_i at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfscanf_i.c line 225 (discriminator 3)
0x40208f69: _scanf_i at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfscanf_i.c line 218
0x40205bf0: _strtol_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/strtol.c line 139
0x4020880c: __ssvfscanf_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfscanf.c line 462
0x4010462f: lmacRxDone at ?? line ?
0x40103579: lmacProcessTXStartData at ?? line ?
0x40103576: lmacProcessTXStartData at ?? line ?
0x401024a2: wDev_ProcessFiq at ?? line ?
0x4010457a: lmacTxFrame at ?? line ?
0x402083f8: _sungetc_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfscanf.c line 221
0x40208488: __ssrefill_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfscanf.c line 269
0x40201b60: realloc_loc at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/heap.c line 140
0x40204a5c: sscanf at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/sscanf.c line 431
0x40204b18: __seofread at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/stdio.c line 73
0x40202b70: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/HardwareSerial.cpp line 69
0x40202dbd: Print::write(char const*) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Print.cpp line 114
0x401002ac: _umm_free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1295
0x40100a88: free at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc/umm_malloc.c line 1755
0x40202cb8: HardwareSerial::begin(unsigned long, SerialConfig, SerialMode, unsigned char) at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/HardwareSerial.cpp line 50
0x40202768: VisualMicroDebug::outPacketEnd() at C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\IDE\Extensions\iv20ypi4.j1y\Micro Platforms\default\debuggers\VM_DBG/VM_DBG.cpp line 996
0x402026b3: setup at C:\Users\IMAN4K\Documents\Arduino\sscanf_test/sscanf_test.ino line 15 (discriminator 2)
0x40202a20: loop_wrapper at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/core_esp8266_main.cpp line 122
0x401000e5: cont_wrapper at C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/cont.S line 81
```

IMAN4K, 2018-05-22:
Different failure on `Arduino nano(avr-libc)` :(
```
#include <Arduino.h>
#include <HardwareSerial.h>
#include <stdio.h>

void setup() {
	Serial.begin(115200);

	uint8_t hh, mm, ss;
	if (sscanf("@SetTime:10:18:35$", "@SetTime:%d:%d:%d$", &hh, &mm, &ss))
		Serial.print("H: " + String(hh) + " M: " + String(mm) + " S: " + String(ss));
}

void loop() {

}
```
Which produce:
`H: 0 M: 0 S: 35`

earlephilhower, 2018-05-22:
You have a type error and passing in invalid pointers to store the results in sscanf.  If you turn on compiler warnings, gcc should barf on the code you've pasted.

`%d` wants `int*`, you're passing in a pointer to a `uint8_t*`.

Change `uint8_t` -> `unsigned int` and it will work on both.

closed 2018-05-22 by earlephilhower