esp8266/Arduino #4751: Loop on 8266 gets slower and slower the longer it runs.

ondabeach, 2018-05-21:
I was running 2.30 core and ran into this problem, so I updated to the latest (2.41) and the problem persists.

Anyone know what is going on here?

Running the simple sketch below demonstrates what seems to be a timing drift of some sort. As soon as the sketch starts the time taken to loop increases quite quickly. Then it stabilises for a while and only varies by a few loops second by second before dropping quickly by around 1000 iterations per second then stabilizing again for a while. It just seems to keep going in the same fashion as you can see by the output below the sketch.

The gaps in the output are where I have edited out most of the output from the stable periods so this post wasn't thousands of lines long.

SKETCH:
////////////////////////////////////////////////////
long lastTime;
long loops=0;

void setup(){
  Serial.begin(115200);
}
void loop(){
  loops++;
  if(millis()>=lastTime+1000){
    Serial.println(String(loops));
    loops=0;
    lastTime=millis();
  } 
}
////////////////////////////////////////////////////////////

OUTPUT:
//////////////////////////////////////////////////////
112312
112313
112311
112312
112312
112311
112312
112312
112311
112314
112312
112314
111547
110907
110911
110907
110908
110908
110909
110909
110908
110907
110909
110909
110904
110909
110909
110909
110907
110907
110910
110909
110907
110909
110908
110910
110908
110906
110910
110908
110907
110911
110907
110910

110908
110910
110909
110910
110909
110912
110910
110312
109539
109541
109540
109541
109537
109539
109541
109540
109539
109542
109540
109541
109540
109541
109541
109540
109539
109541
109540
109541

109541
109541
109539
109541
109541
109541
109541
109247
108206
108205
108204
108209
108204
108205
108205
108203
108205
108205
108203
108206
108204
108205
/////////////////////////////////////////////////////////////////////////////

Makuna, 2018-05-21:
not related specifically to your issue, but more to the error in how you testing for time passed. Millis returns an unsigned value, you don't want to check > or <. 
Your code does not work in all cases and includes time not being tracked  ...
```
if  (millis()>=lastTime+1000){ // this doesn't work when rollover happens
  // blah blah blah
    lastTime=millis();  // ddn't account for the time between the two millis() calls
}
```
the better way to do it...

```
uint32_t currentTime = millis(); // store away one millis to work with
if  (currentTime - lastTime > 1000) { // compare the delta works for unsigned
  // blah blah blah
    lastTime=currentTime;
}
```

tonton81, 2018-05-21:
you could keep currentTime above the condition by declaring it as static, keeps it clean and organized to the condition, using Makuna's code, just change "uint32_t" to "static uint32_t"

tonton81, 2018-05-21:
you could also do:

```
static uint32_t theTime = millis();
if ( millis() - theTime > 1000 ) {
  theTime = millis();
  // Do whatever here
}
```

It's simpler to use and understand

ondabeach, 2018-05-22:
@makuna and @tonton81 you didn't even bother to understand the issue. Stop making irrelevant, unhelpful posts.

The point is to count the number of loops the processor can get through in 1 second and check the number of iterations for consistency over time.

The result is that the number of iterations per second decreases with time. It happens quite quickly. Within a few minutes the iterations reduce by around 10,000 and it just keeps getting slower.

tonton81, 2018-05-22:
that was quite harsh, maybe you should brush up on your “community” skills

tonton81, 2018-05-22:
better yet, you have a problem, both in code and in mentality.

ondabeach, 2018-05-22:
@tonton81 your post had nothing to do with the issue I was reporting. 
You didn't try the sketch. 
You didn't even understand the problem.
How exactly is pointing out totally irrelevant coding niceties supposed to help?

tonton81, 2018-05-22:
the fact that your casting a long as a String to print, and using a long for millis() speaks alot about how your coding sucks. take that into consideration next time you tell others about making unhelpful posts. Goto a forum as this isn’t a core issue and learn some respect, and go learn basic programming while your at it.

Pablo2048, 2018-05-22:
To calm this down i have some questions:
1. is wifi on or off when testing?
2. if on - is Wifi connected?
Maybe it depend on how much wifi traffic goes around when measuring...

ondabeach, 2018-05-22:
@Pablo2048 a sensible question.
The same thing happens with wifi turned on or off.

torntrousers, 2018-05-22:
When you say you've tried it with Wifi off was that by booting with RF_DISABLED? Otherwise there's quite a lot going on on the background which will slow things down as it tries to connect to Wifi and DHCP etc. Also what about if you run it for longer - many minutes or hours? Does it settle down into a more steady loop time?

tonton81, 2018-05-22:
Correcting users on their grammer your/you’re shouldn’t be tolerated either, in fact, it’s worse than what happened earlier. If we speak our native language here (french) you can’t judge a person by their linguistics, otherwise it’s the same boat as earlier. If we’re going to start correcting users on their grammer provided they’re not speaking their native language for the benefit of the forums/github community, them we are dealt with racism as a whole. I can’t help but see users come on here every day who don’t speak english, who are we to tell them they are not speaking or talking your language or not google translating :) Would you like me to tell users to correct their grammar when they post their issue? I understand you try to keep “morals” in order, but correcting users on a language thats secondary to them is not a motivation/practice either for the public as a whole.

devyte, 2018-05-22:
@tonton81 Grammar correction has absolutely nothing to do with racism. My intent was to point out a recurring mistake to you, so that you may correct yourself and improve. And before you throw racism around in the context of native English, you should know that I'm latin american, I live in Chile, and my native tongue is Spanish. You have now needlessly insulted both a user and a developer.
Now please either move on, or help with the issue at hand. Ou si tu veux nous pouvons avoir une conversation privée et sérieusement insulter l'un a l'autre.

tonton81, 2018-05-22:
We could keep arguing but I’m not here for that, this is getting off topic and I agree enough is enough and no need to speak french, some people won’t understand it :)

ondabeach, 2018-05-23:
@torntrousers  
I used WiFi.mode(WIFI_OFF) which doesn't require putting the esp into deep sleep.
I confirmed that the WiFi was indeed off as there is no SSID visible and current consumption was way down.

ondabeach, 2018-05-23:
@devyte 
The longest I've let it run is 10 minutes and it continues to drop. 
**Yes, the drop pattern is monotonical**. 
It drops a couple of thousand iterations within a few seconds of starting.
Then it stabilises for about 30 seconds.
Then it drops another couple of thousand iterations over the space of only about 3 seconds.
Then it stabilises for around another 30 seconds.
Then it drops another couple of thousand iterations over the space of only about 3 seconds.
Then it stabilises for around another 30 seconds and keeps repeating that pattern for as long as I've run the test to this point.

If you look at the output in my original post, each line is the number of loop iterations which are reported every 1 second. where the blank lines are between the groups of numbers is where the iteration count was stable for the approx. 30 seconds so I deleted them to keep the post from being a mile long.

The behaviour is the same with a cold module first thing in the morning or after running all day. It does not seem to be temperature related.

devyte, 2018-05-23:
@ondabeach what about heap?
Could you please fix your sketch to account for millis overflow, and let it run e.g. overnight, or over a couple of days?

ondabeach, 2018-05-23:
@devyte what do you want to know or see in regard to the heap?

I will mod as suggested, start the sketch running and report back what happens in the long term.

WereCatf, 2018-05-23:
I am not seeing the issue here. I just tested this for ~10 minutes and the number of loop-iterations it does in one second varies by about ~300 loops/s, but it never goes below that, ie. I get between 95.3k - 95.6k iterations per second.

```
uint32_t prevMillis;
uint32_t numLoops = 0;

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("Begin...");
  delay(500);
  prevMillis = millis();
}

void loop() {
  if((millis() - prevMillis) >= 1000){
    Serial.println("Loops elapsed in 1s: " + String(numLoops));
    numLoops = 0;
    prevMillis = millis();
  }
  numLoops++;
}
```

ondabeach, 2018-05-23:
@WereCatf  thanks for your time in testing this.
What version of Arduino IDE are you using?
What 8266 core version are you using?
What module are you running the sketch on?

The starting iteration count in all instances is around 210000.

I ran your code on 2 different versions of NodeMCU and an esp-01 and they all exhibit the same monotonical pattern of iteration decramentation as outlined in my OP.

WereCatf, 2018-05-23:
@ondabeach Like I said, I am using the latest git-code. Try with that and see if the issue persists or not.

ondabeach, 2018-05-23:
@WereCatf 
As my OP states, I am using the latest core.
I am using IDE 1.8.5

What version of Arduino IDE are you using?
What module type are you using?

RudyFiero, 2018-05-23:
95888 at the start for me. 
  95338 about 11-12 hours later (now).

I am using 1.8.6 and git from last week.

#include <ESP8266WiFi.h>

```
long lastTime;
long loops = 0;

void setup() {
  Serial.begin(115200);
  ESP.eraseConfig();
  WiFi.mode( WIFI_OFF );
  WiFi.forceSleepBegin();
  }
void loop() {
  loops++;
  if (millis() >= lastTime + 1000) {
    Serial.println(String(loops));
    loops = 0;
    lastTime = millis();
  }
}

```

WereCatf, 2018-05-23:
@ondabeach IDE-version really shouldn't matter, since it's not the one doing the compiling, but 1.8.5. Also, ESP-12E.

ondabeach, 2018-05-23:
@WereCatf fair comment on the 2.4.1 as opposed to the latest git ver. I will download and try but I'd bet London to a brick that it won't make any difference.

I'd be interested to see the output of the first 100 seconds or so from your setup. As I intimated earlier, the first and last iteration counts of a 12 hour period don't tell much of a story.

ondabeach, 2018-05-23:
@RudyFiero I ran your code and below is the first 100 seconds of output:

A start and end iteration count does not demonstrate the pattern of decramentation you are experiencing.

225337
222689
222415
220198
219663
219662
219664
218023
216980
216981
216980
216981
216980
216980
216981
216396
214361
214363
214361
214362
214361
214361
214363
214361
214361
214362
214361
214363
214361
214361
214363
214361
213222
211807
211807
211807
211805
211806
211807
211805
211805
211807
211806
211807
211805
211806
211806
211806
211805
211808
211805
211807
211805
211806
211807
211805
211806
211807
211805
211803
211805
211806
211807
211805
211805
211810
209581
209311
209309
209310
209310
209310
209309
209311
209309
209311
209309
209310
209310
209310
209309
209311
209309
209311
209310
209309
209311
209309
209309
209311
209310
209310
209310
209309
209311
209309
209311
209311
209309
@devyte the decramentation pattern is not quite monotonical. The stable period increases in length with each phase of decramentation, which itself remains quite constant in that the number of iterations decreases by around the same amount (roughly 1%) over the space of only 3 seconds.

Very strange indeed.

I don't think others quite grasp the relevance of the finer details of the pattern of decramentation.

As I said earlier, I will not be using an 8266 module for the project I am currently working on. I do not need WiFi but the advertised speed of the 8266 and ESP32 modules (160mhz and 240mhz) had me going to them first. But, unfortuantely they do not deliver anything like the expected performance. A Maple Mini running at 72mhz can iterate the same loop at over 1.3mhz compared to the best I've achieved from an espressif module is around 220khz. The Maple Mini is also much more stable from the moment it starts.

This issue lost importance to me very early on, however I kept reporting on it as I thought it worth investigating. But if you want to put it in the too hard basket then feel free to close this issue.

On the other hand if you want to keep investigating then I am happy to put the time in to help as the espressif modules do contribute greatly to the success of my business.

WereCatf, 2018-05-23:
@ondabeach https://pastebin.com/zDi5SGVs -- as you can see, the issue is not present.

RudyFiero, 2018-05-23:
At first I thought this was a non-issue. But I did wonder what would happen over the period of a month or year. I might do a test like this with a battery backed up ESP and see where it goes.

ondabeach, 2018-05-23:
I had to kick down some pretty thick doors to get people to take me seriously when I reported the PUYA flash chip issue.

There are some very clever people around here, sometimes though, they can't see the forest for the trees.

Fortunately however, I'm nothing if not persistent :)

WereCatf, 2018-05-23:
I was wrong. I just got back to checking my log and I'm seeing this issue, too. Ain't got any timestamps, so I can't say how long it took for it to start manifesting, but yes, the number of loop-iterations does slowly decline over time.

torntrousers, 2018-05-23:
_...the decramentation pattern is not quite monotonical. The stable period increases in length with each phase of decramentation..._ - Dude! Where's my dictionary...

I've tried your original and some of the other code posted in this issue and the results I get seem pretty stable. With you original code I get values around 182161 which is quite a lot more than the numbers you get which seems odd.

WereCatf, 2018-05-23:
@torntrousers You are probably running your ESP at 160MHz instead 80MHz, aren't you?

torntrousers, 2018-05-23:
I've tried both. The original bug report doesn't mention the speed used.

RudyFiero, 2018-05-23:
> With your original code I get values around 182161 which is quite a lot more than the numbers you get which seems odd.

I found that elevation makes a difference in performance. 
.
.
(kidding)

lrmoreno007, 2018-05-23:
I can reproduce the problem with this chinese NodeMCU (esp12e):
[image]

This config:

[image]

And this sketch:

```
#include <ESP8266WiFi.h>

uint32_t lastTime;
long loops = 0;
uint32_t seconds = 0;
long first;
long lowest = 2147483647;
long highest;

void setup() {
  Serial.begin(115200);

  ESP.eraseConfig();
  WiFi.mode( WIFI_OFF );
  WiFi.forceSleepBegin();

  while (!Serial); // wait for serial attach
  Serial.println("starting...");

  lastTime = millis();
}

void loop() {
  loops++;

  uint32_t currentTime = millis();
  uint32_t delta = currentTime - lastTime;
  if  (delta > 1000) {
    seconds++;
    Serial.print("LOOPS ");
    Serial.print(loops);
    Serial.print(" SECONDS ");
    Serial.print(seconds);
    if (seconds == 1) first = loops;
    if (loops > highest) highest = loops;
    if (loops < lowest) lowest = loops;
    Serial.print(" FIRST ");
    Serial.print(first);
    Serial.print(" LOW ");
    Serial.print(lowest);
    Serial.print(" HIGH ");
    Serial.println(highest);

    loops = 0;
    lastTime = currentTime;
  }
}
```

This is the result after about 5200 seconds:
LOOPS 94794 SECONDS 5217 FIRST 95162 LOW 94657 HIGH 95295

and here a graphic:
[image]

As an observation, I get less loops / seconds than all of you, even with the original sketch.

lrmoreno007, 2018-05-23:
I have found a pattern, every 1199 seconds a fall spike occurs. For the big drop peak I have not found a pattern yet.

What happens every 1199-1200 seconds?

tonton81, 2018-05-23:
hook up wireshark, maybe a keep-alive?

tonton81, 2018-05-23:
well you never know if a bug can cause the wifi stack to fully shut down or not. :)

lrmoreno007, 2018-05-23:
(7 * 1199) + 1 (first one) = 8394 seconds

LOOPS 94770 SECONDS 8391 FIRST 95162 LOW 94639 HIGH 95295
LOOPS 94770 SECONDS 8392 FIRST 95162 LOW 94639 HIGH 95295
LOOPS 94770 SECONDS 8393 FIRST 95162 LOW 94639 HIGH 95295
**LOOPS 94629 SECONDS 8394 FIRST 95162 LOW 94629 HIGH 95295**
LOOPS 94771 SECONDS 8395 FIRST 95162 LOW 94629 HIGH 95295
LOOPS 94770 SECONDS 8396 FIRST 95162 LOW 94629 HIGH 95295
LOOPS 94770 SECONDS 8397 FIRST 95162 LOW 94629 HIGH 95295

is repeated

ondabeach, 2018-05-23:
@tonton81 read the thread more thoroughly. The problem occurs with WiFi turned off.

ondabeach, 2018-05-23:
As @devyte said, this problem might not be in the 8266 core. It might be in the espressif SDK. It would be good if someone can see if the problem happens when using IDF and not the core.

RudyFiero, 2018-05-23:
11 hours ago I started one that also printed out the heap, and there has been no change in that.

ondabeach, 2018-05-24:
@RudyFiero can you restart the sketch and post the first 100 seconds or so of output.

lrmoreno007, 2018-05-24:
I have updated the graph up to 20177 seconds (5.6 hours) and there has been a rise in the loops.

This may indicate that at the end it stabilizes within a range, drawing a graph in the form of a saw blade.

I must already cut this test, tomorrow I will try to do it longer, I will decrease the sampling to 15-20 seconds and I will try to use another different module to discard hardware issues (at least in the pattern of peaks)

I will update the previous post to not extend this.

ondabeach, 2018-05-24:
@lrmoreno007 I started a run last night which ran for about 2 hours before I went to bed. At that time it was stable at around 103500. This morning when I got up it was stable at around 105800, so I can confirm your finding that it does not keep decrementing.

It will be very helpful for the core developers if we can come up with a long term pattern, which seems now to be a sawtooth with monotonic fall (and possibly also rise) periods.

RudyFiero, 2018-05-24:
95888  initial
 95338  11 hours after start
 95392  25 hours after start

============================
 first 100 seconds 
```
1
95855
95885
95888
95890
95890
95890
95889
95889
95896
95888
95888
95890
95889
95889
95888
95889
95889
95888
95889
95890
95888
95889
95888
95889
95889
95888
95888
95890
95888
95890
95888
95889
95889
95889
95889
95890
95888
95890
95891
95889
95889
95889
95889
95891
95888
95889
95889
95889
95889
95889
95889
95890
95888
95890
95889
95888
95890
95888
95889
95888
95886
95890
95889
95888
95889
95888
95889
95889
95889
95890
95890
95887
95890
95888
95888
95889
95888
95890
95888
95887
95890
95888
95888
95888
95888
95891
95887
95887
95890
95888
95887
95889
95888
95890
95886
95889
95889
95888
95887
95890
95888
95889
95887
95888
95890
95887
95888
95889
95888
95889
95887
95888

```

RudyFiero, 2018-05-24:
@Irmoreno007 

> As an observation, I get less loops / seconds than all of you, even with the original sketch.

I am getting much the same as you are. Other than the setup, I am running the original sketch.

ondabeach, 2018-05-24:
@RudyFiero thanks Rudy, sorry I forgot to ask earlier, could you also post the sketch that produced that output. I'll try running the exact same code and see what I get. If yours is stable and mine not then we just need to find what is different between out systems.

RudyFiero, 2018-05-24:
```
#include <ESP8266WiFi.h>

long lastTime;
long loops = 0;

void setup() {
  Serial.begin(115200);
  ESP.eraseConfig();
  WiFi.mode( WIFI_OFF );
  WiFi.forceSleepBegin();
  delay( 1 );

}
void loop() {
  loops++;
  if (millis() >= lastTime + 1000) {
    Serial.println(String(loops));
//Serial.print("   ");
//Serial.println(ESP.getFreeHeap());
    loops = 0;
    lastTime = millis();
  }
}
```

devyte, 2018-05-24:
Please try to confirm that pattern, it is suggestive of certain types of search algorithms. I have to wonder about the long repetition period, though. For a system this small I'd expect a much shorter oscillation period. Then again, maybe it isn't really long, but rather the long oscillation is actually aliasing due to measuring at 1s, which is low frequency (consider a video recording of a car wheel in motion: in the video it can seem as though it is rotating slowly, even though in reality it's going very fast).
One way to test that is to play with the measuring time, e.g. measure at 0.5s, or 0.333s, or 0.1s, etc. If it is aliasing of a faster underlying change, then the apparent oscillation period should change.
I'm thinking of mem alloc/free as the culprit for the symptom. One place is inside the String object. You could try printing the C way with printf. There is probably another alloc/free when scheduling the loop/loop returns. I don't think that can be avoided.
@d-a-v wasn't there a way to get debug msgs for malloc/free calls?

Whatever else, having confirmation that the drop isn't monotonical, but that instead there is an oscillation, is a good thing.

torntrousers, 2018-05-24:
Came across this just now, vaguely relevant with the mention here of speed comparison to the Maple Mini - https://hilo90mhz.com/arduino-esp32-esp8266-101-speed-test-comparison-chart/

ascillato, 2018-05-24:
Hi,

I also test the code from @RudyFiero on a NodeMCU, and can confirm this strange behaviour.

**Code:**

```
#include <ESP8266WiFi.h>

long lastTime;
long loops = 0;

void setup() {
  Serial.begin(115200);
  ESP.eraseConfig();
  WiFi.mode( WIFI_OFF );
  WiFi.forceSleepBegin();
  delay( 1 );

}
void loop() {
  loops++;
  if (millis() >= lastTime + 1000) {
    Serial.println(String(loops));
//Serial.print("   ");
//Serial.println(ESP.getFreeHeap());
    loops = 0;
    lastTime = millis();
  }
}
```

**Compilation:**

[image]

**Output:** (from arduino serial plotter)

[image]

_Y axis is loops per second and X axis is seconds_

Note that each speed drop happens in the double of time than the previous drop.
I tested several times (power cycle and other times with reset button) and is exactly the same pattern each time. 

This is very odd. Any clues?

tonton81, 2018-05-24:
maybe it has something to do with Strings since they are known for dynamic allocation madness. Why are you casting when just printing returns same result?

tonton81, 2018-05-24:
try not using (String)? I mentioned this when nobody was listening. millis() should also be stored as uint32_t, not “long”, for the sake of portability.

ascillato, 2018-05-24:
Hi,

I test the **modified code** from @tonton81 and can **still confirm** this strange behaviour.

**Code:**

```
#include <ESP8266WiFi.h>

uint32_t lastTime;
uint32_t loops = 0;

void setup() {
  Serial.begin(115200);
  ESP.eraseConfig();
  WiFi.mode( WIFI_OFF );
  WiFi.forceSleepBegin();
  delay( 1 );

}
void loop() {
  loops++;
  if (millis() >= lastTime + 1000) {
    Serial.println(loops);
//Serial.print("   ");
//Serial.println(ESP.getFreeHeap());
    loops = 0;
    lastTime = millis();
  }
}
```

**Compilation:**

[image]

**Output:** (from arduino serial plotter)

[image]

_Y axis is loops per second and X axis is seconds_

Note that each speed drop happens in the double of time than the previous drop.
I tested several times (power cycle and other times with reset button) and is exactly the same pattern each time. The device is cold and with every power cycle, the loops value is exactly the same.

This is very odd. Any clues?

tonton81, 2018-05-24:
So now this rules out primitive types and String class

ascillato, 2018-05-24:
@igrr 

Any clues why is this ESP8266 behavior ?

tonton81, 2018-05-24:
Next, serial class should be looked into. Your resetting lastTime to millis() only after the print, could be serial latency issue?

tonton81, 2018-05-24:
using the corrected code with ascillato, latest github core, IDE 1.8.5, heres where it starts on an adafruit huzzah...

`95818
95894
95890
95891
95892
95892
95891
95890
95891
95892
95890
`

tonton81, 2018-05-24:
heres a longer result:

95896
95891
95891
95891
95890
95890
95892
95890
95891
95891
95891
95892
95889
95891
95892
95891
95890
95891
95891
95891
95890
95890
95892
95890
95891
95891
95890
95891
95890
95892
95890
95891
95891
95890
95889
95891
95889
95890
95891
95889
95891
95891
95889
95890
95890
95888
95889
95889
95891
95890
95889
95890
95890
95890
95890
95889
95891
95889
95888
95891
95889
95890
95891
95889
95891
95888
95889
95890
95890
95889
95890
95889
95891
95888
95889
95891
95891
95889
95891
95890
95892

[21 more replies]