esp8266/Arduino #4757: ESP8266 OTA serial monitor

kingstonheath, 2018-05-23:
Hi all,
Question from a newbie on ESP's that is driving me crazy.  I've built ArduinoOTA into a couple of sketches on my ESP8266's (Wemos D1R2).  I can see the network port in Arduino IDE and the OTA sketch uploads work well.  I cannot, however, get a serial monitor (Putty or Termite) working OTA.  No problems with them working when the boards are connected by cable.  I have scoured several forums and my understanding is that I need to select the same serial port in Putty or Termite as when the board is connected by cable (COM4 in my case).  When I do this, I get an error saying "Unable to open connection to COM4" which makes sense to me as nothing is connected to it.
Clearly something basic I have missed in my searching and hoping someone can help me out.  I have also made sure the baud matches what is set in the sketch (115200)

Cheers
Steve

kingstonheath, 2018-05-23:
Thanks for the response.  It didn't make sense to me either so that answers one question.  
Is it possible to monitor the board serial output OTA with Putty or Termite and, if so, is there a tutorial somewhere on how to set it up?

WereCatf, 2018-05-23:
No, not possible. Serial-output is just that, it only goes to the serial-port. If you want to use e.g. PuTTY to monitor your ESP8266-device, you should look into implementing e.g. Telnet-server on it and connect to that with PuTTY and then writing any messages out over Telnet instead of serial.

lrmoreno007, 2018-05-23:
If you had read and filled in the issue template, you would have noticed that it says: "I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there."

Obviously you have not read it because you would have seen this "http://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html"

Please close this issue, it's not a help forum.

devyte, 2018-05-23:
Not the right place to ask such questions. Closing.

closed 2018-05-23 by devyte