esp8266/Arduino #4761: ESP8266httpUpdate missing function for SPIFFS update

beegee-tokyo, 2018-05-24:
ESP8266httpUpdate class offers
```
t_httpUpdate_return update(const String& host, uint16_t port, const String& uri = "/",
                               const String& currentVersion = "");
```
which allows to send some information to a server script ([example](https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#http-server)->Advanced updater)

For SPIFFS update there is no corresponding function.
Or did I miss something?

caverna, 2018-05-24:
https://github.com/esp8266/Arduino/pull/3732

beegee-tokyo, 2018-05-24:
@caverna I am not talking about ESP8266HTTPUpdateServer but about ESP8266httpUpdate which is updatinf firmware and SPIFFS from an external server.

caverna, 2018-05-25:
Yeah, I know, but look at this (typeof(Update) == "ESP8266httpUpdate"):
```
size_t spiffsSize = ((size_t) &_SPIFFS_end - (size_t) &_SPIFFS_start);
_setUpdaterError();
if (!Update.begin(spiffsSize, U_SPIFFS)){//start with max available size
    if (_serial_output) Update.printError(Serial);
}
```

That could lead you to something, don't?

beegee-tokyo, 2018-05-25:
@caverna maybe, thanks for the tip. But I already ended up to work with my own ESP to Server communication to verify if an update is required.