esp8266/Arduino #4762: Suport NodeMCU ESP8266 - 32MB Flash

laercionit, 2018-05-24:
It recently hit NodeMCU 12-E with 32MB of Flash.

This allows you to use SPIFFS for more complex projects.
I think we need to adjust boards.txt to support this new Flash / SPIFFS modeling

Example:
https://www.amazon.de/nodemcu-V3-Lua-esp8266-32-MB-Internet-Development/dp/B078X8NJ3P

Deimos1994, 2018-05-24:
Hmm.. 32MB? I thought it supports up to 16MB...
How can it be possible? quote from the datasheet:
> ESP8266EX uses external SPI flash to store user programs, and supports up to 16 MB
memory capacity theoretically.

https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf

laercionit, 2018-05-24:
From what I saw the manufacturer is the robotdyn.
They make boards with upgrades.
https://robotdyn.com/catalog/arduino/boards/wifi-nodem-esp8266-32m-flash-ch340g.html

Deimos1994, 2018-05-24:
@laercionit It doesn't really matter who makes a board since the chip is designed by espressif, and espressif's datasheet says it supports only 16MB (128Mbits).
My opinion: I think sellers intentionally write it is 32MB (which is wrong should be 32Mb) just to sell those boards to people that don't pay much attention.

laercionit, 2018-05-25:
I have some cards from other models of this same company, they do some very well made things. They have an Arduino Mega with ESP8266 integrated.

Unfortunately using Generic Esp8266 I can not upload the code via USB.

This board follows the same NodeMCU reference. Selecting the NodeMCU only has version 1.0 that is limited to 4MB of memory.

To perform tests I will have to change the board.txt in NODEMCU to support 16MB equal to board generic.

Would someone have patience to guide me in how to make this change?

devyte, 2018-05-25:
Please be aware that with 16MB flash, spiffs is already in trouble wrt performance

WereCatf, 2018-05-25:
@laercionit It's 4MB flash, ie. 32 megabits, not 32 megabytes. Just look at the SPI-Flash on the board, it clearly says Winbond 25q32fvsig, which is a 32Mb-part ( https://www.winbond.com/resource-files/w25q32fv%20revi%2010202015.pdf )

The Chinaman either doesn't know or doesn't care about the difference between MB and Mb -- always keep this in mind when buying stuff like this.

devyte, 2018-05-25:
@WereCatf thanks for figuring that out!
Closing.

closed 2018-05-25 by devyte