esp8266/Arduino #4764: Reset on json.printTo();

mahmoud2802, 2018-05-26:
hello

void serializeEffects(String &jsonString, bool pretty, bool creds)
{
	LOG_PORT.println("///Serilaze Effect");
	// Create buffer and root object
	DynamicJsonBuffer jsonBuffer;
	JsonObject &json = jsonBuffer.createObject();
	json["count"] = Effect_count;
	// Effects
	JsonArray &EffectsJson = json.createNestedArray("effects");
	for (int i = 0; i < CountEffect; i++)
	{
		JsonObject &item = EffectsJson.createNestedObject();

		;
		item["id"] = effects_list[i].id;
		item["count"] = effects_list[i].count;
		item["name"] = effects_list[i].name.c_str();
		item["block"] = effects_list[i].block;
		item["speed"] = effects_list[i].speed;
		item["count_color_use"] = effects_list[i].count_of_coloruse;
		item["background_color"] = effects_list[i].background_color;
		item["direction"] = effects_list[i].direction;
		item["has_direction"] = effects_list[i].has_direction;
		item["has_block"] = effects_list[i].has_block;
		item["has_irancolor"] = effects_list[i].has_irancolor;
		item["has_specterum"] = effects_list[i].has_specterum;
		item["position"] = effects_list[i].position;
		item["color1"] = effects_list[i].color[0];
		item["color2"] = effects_list[i].color[1];
		item["color3"] = effects_list[i].color[2];
		item["color4"] = effects_list[i].color[3];
		item["color5"] = effects_list[i].color[4];
		item["color6"] = effects_list[i].color[5];
		item["color7"] = effects_list[i].color[6];
		item["color8"] = effects_list[i].color[7];
		item["color9"] = effects_list[i].color[8];
		item["color10"] = effects_list[i].color[9];
		EffectsJson.set(i, item);

	}
	
		json.printTo(jsonString);

	LOG_PORT.println("Serilaize Done");

}
i have a problem. when set Count to CountEffect I don't have any problem and works good.
but if CountEffect set 10 the program reset on 		json.printTo(jsonString);.
effects_list array size is 30.
can anyone help me to solve this problem?
Thanks

Pablo2048, 2018-05-26:
Probably no - it has nothing to do with Arduino Core - use the forum @ esp8266.com (why you didn't read the issue template?).

devyte, 2018-05-26:
Not the right place to ask, this is the issue tracker for the core.
Closing.

closed 2018-05-26 by devyte