esp8266/Arduino #4765: When a second NodeMCU ESP8266 is powered up, NEITHER will send readings to MQTT server

fritserasmus, 2018-05-27:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				NodeMCU ESP8266
- Core Version:			[(#4736) / installed 2018/05/27]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[NodeMCU 0.0 (ESP-12 Module)]
- Flash Mode:			[4M(1M SPIFFS]
- Flash Size: 		        [4MB]
- lwip Variant:                 [v2 Prebuilt (MSS=536]
- Reset Method:		[nodemcu]
- Flash Frequency:	        [80Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:               [115200]

### Problem Description

Detailed problem description goes here.
I have two Node-MCU's and will be adding two more.
UBUNTU 16.04 Server is my MQTT server also running mySQL

I have been using the same IDE for both temp sensors: DS18B20 but changing the topic to be different so I can have separated INSERT SQL Scripts.

When only one Noe-MCU is connected either via PC or directly from power it will send the reading out. It will show on the Serial monitor and MQTT SUB and update mySQL table. This is the rule for either of the two Node-MCU's connected.
The IP Address stays the same when connected to PC or direct to power supply
(I checked with (Advanced IP Scanner to make sure I have the unit by confirming MAC Addresses)
As soon as the second  Node-MCU starts up, NO Temperature readings reaches the Serial port or MQTT or mySQL

With Node-MCU's connected in turn for a a minute it will update with the correct record that distinguish between the different sensors. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

// DS18B20_WITH_ESP8266_NODEMCU_Upload_OWN_MQTT_FER_V2_G1

#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <OneWire.h>
#include <DallasTemperature.h>
#include <Streaming.h>

#define ONE_WIRE_BUS            D4      // DS18B20 pin

const char* ssid = "myaccesspoint";
const char* password = "accesspointpw";
const char* mqtt_server = "192.168.1.110";
const char* mqtt_username = "mqttuser";
const char* mqtt_password = "0000";
const char* mqtt_topic = "/18561/geyser1/temp";

WiFiClient espClient;
PubSubClient client(espClient);

OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature DS18B20(&oneWire);

char temperatureString[6];

void setup() {
  // setup serial port
  Serial.begin(115200);

  // setup WiFi
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);

  // setup OneWire bus
  DS18B20.begin();
}

void setup_wifi() {
  delay(300);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Attempt to connect
    if (client.connect("ESP8266Client", mqtt_username, mqtt_password)) {
      Serial.println("connected");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 2 seconds before retrying
      delay(2000);
    }
  }
}

float getTemperature() {
  Serial << "Requesting DS18B20 temperature..." << endl;
  float temp;
  do {
    DS18B20.requestTemperatures(); 
    temp = DS18B20.getTempCByIndex(0);
    delay(500);
  } while (temp == 85.0 || temp == (-127.0));
  return temp;
}

void loop() {
  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  float temperature = getTemperature();
  // convert temperature to a string with two digits before the comma and 2 digits for precision
  dtostrf(temperature, 2, 2, temperatureString);
  // send temperature to the serial console
  Serial << "Sending temperature: " << temperatureString << endl;
  // send temperature to the MQTT topic
  client.publish(mqtt_topic, temperatureString);

  delay(500);

}

### Debug Messages

```
None
```

WereCatf, 2018-05-27:
This is not the place to ask questions like this, this is for reporting bugs in the Arduino-core for the ESP8266. Use esp8266.com for general questions and help.

That said, your problem is pretty simple: you're starting your Nodemcus as both AP and STA and so they end up trying to connect to one-another. Use WiFi.mode(WIFI_STA); before WiFi.begin(ssid, password); to make them act as plain STA.

Pablo2048, 2018-05-27:
@fritserasmus please don't be so ironic - in the issue template you have been pointed to Issue policy ( https://github.com/esp8266/Arduino/blob/master/POLICY.md ) did you read this? First sentence in the issue template after "delete below" is "If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like **esp8266.com**." so correct direction where to ask has been already given.
IMO: if you have tried @WereCatf 's suggestion and wifi is not connected, then there is problem with correct wifi configuration somewhere in your sketch.

devyte, 2018-05-27:
The explanation and channels are in the issue POLICY doc found  in the root of this repo, and in the issue template, which you saw when opening this issue, and which also links to the issue POLICY doc. Be aware that this is a community effort, so the official channels are community channels.
Closing.

closed 2018-05-27 by devyte