esp8266/Arduino #4766: Using the WebSockets library for the GSM SIM800 modem

Djaison, 2018-05-27:
Hello! I'm using the wonderful library arduino WebSockets (Links2004) on ESP32. I want to use a GSM modem to connect my client running on ESP32 with a remote server via Websocket. The SIM800 modem has its own TCP / IP stack and involves working with AT commands. But this mode of operation does not fit.
However, it can be switched to point to point over serial (PPPoS) mode. I'd like to use the TCP Arduino stack with the WebSockets library. Question: Is it possible to connect a websockets connection to a serial interface as the simplest solution? Or do I need to write the TCP Client and Server class for the SIM800 and then include them in the arduino WebSockets library? (I'm new to C ++ (6 months +))

P.S. It would be nice if **@Links2004** pays attention to my question :-)
Thank you in advance!

WereCatf, 2018-05-27:
> ESP32

This is the ESP8266's issue-tracker, not the ESP32's. Secondly, this is for reporting bugs, not for asking general questions. For those, go to esp32.com and close this issue.

Djaison, 2018-05-27:
Ok. Thank You

closed 2018-05-27 by Djaison

VinodVermaVR, 2023-04-28:
After implementing the necessary changes, I was able to successfully resolve the issue. My code is now working seamlessly in both sending and receiving data through WebSocket and SIM800 module. As per the WebSocket protocol, I made sure to establish the initial handshake before sending any data. This helped to ensure that the data transmission was in accordance with the WebSocket standard.
If you require any further assistance or have any questions, feel free to contact me at [vinodvermare@gmail.com](mailto:vinodvermare@gmail.com)