esp8266/Arduino #4769: WDT reset with softwareserial

TLS1000, 2018-05-28:
Hi, I downloaded the latest git, and I'm using the softwareserial(ESP-version) and modbusmaster library.

Problem is that the ESP does a WDT reset every 10s or so:

Soft WDT reset
ctx: cont 
sp: 3ffef980 end: 3ffefcb0 offset: 01b0

I see the pulsetrain goin out GPIO14 so the libs seem to work. What would be causing the reset?

My code:
```
#include <ESP8266WiFi.h>;
#include <SoftwareSerial.h>
#include <ModbusMaster.h>

#define Serial2TxControl 5   //RS485 Direction control 
#define RS485Transmit    HIGH
#define RS485Receive     LOW
ModbusMaster node;

void preTransmission()
{
  //digitalWrite(MAX485_RE_NEG, 1);
  //digitalWrite(MAX485_DE, 1);
  digitalWrite(Serial2TxControl,RS485Transmit);
}

void postTransmission()
{
  //digitalWrite(MAX485_RE_NEG, 0);
  //digitalWrite(MAX485_DE, 0);
  digitalWrite(Serial2TxControl,RS485Receive);
}

SoftwareSerial mySerial= SoftwareSerial(12,14);

void setup() {
  pinMode(Serial2TxControl, OUTPUT);
  Serial.begin(19200);
  Serial.print("Serial2TXcontrol line: "); Serial.println(Serial2TxControl);
  //Serial.swap();
  mySerial.begin(19200);
  node.begin(4, mySerial);
  node.preTransmission(preTransmission);
  node.postTransmission(postTransmission);
  
}

int value = 0;

void loop() {
  uint8_t result;
  uint16_t data[6];
  result = node.readHoldingRegisters(15206,3);
  Serial.println(result,HEX);
  if (result == node.ku8MBSuccess)
  {
    
    Serial.println(node.getResponseBuffer(0));
    Serial.println(node.getResponseBuffer(1));
    Serial.println(node.getResponseBuffer(2));
    
  }
  delay(1000);
}
```

TLS1000, 2018-05-29:
hi, I went a little deeper on the working of the software wathdog of the ESP8266.
Problem is the waiting on a response from a modbus slave (espacially during testing with no slave connected), during that time the software wathdog fires because it thinks the program hangs.

you can solve it elegantly by disabling the watchdog during modbus reads:
ESP.wdtDisable();
  result = node.readHoldingRegisters(15206,3);
  ESP.wdtEnable(1);

I have no more soft WDT resets anymore

closed 2018-05-29 by TLS1000

mkucukiba, 2018-10-14:
> merhaba, ESP8266'nın yazılım wathdog'unun çalışmasına biraz daha derinden gittim. 
> Sorun, modbus slave'in (özellikle slave bağlı olmayan test sırasında) bir yanıt beklenmesidir, bu sırada yazılım wathdog programın durduğunu düşündüğü için patlar.
> 
> modbus okumaları sırasında watchdogu devre dışı bırakarak zarif bir şekilde çözebilirsiniz: 
> ESP.wdtDisable (); 
> result = node.readHoldingRegisters (15206,3); 
> ESP.wdtEnable (1);
> 
> Artık daha yumuşak WDT sıfırlarım yok

i am working for 10 day.
thanks TLS1000, this solution wonderfully

HarpreetHeera, 2025-02-04:
> hi, I went a little deeper on the working of the software wathdog of the ESP8266. Problem is the waiting on a response from a modbus slave (espacially during testing with no slave connected), during [...]

Thank you so much. Above solution works like a charm. Really appreciate.