esp8266/Arduino #4771: ESP8266httpUpdate SHA-256 support

suculent, 2018-05-28:
I need to validate firmware updates using SHA hash instead of MD5. Unfortunately this required adding sha256 to the core and touching Update.cpp. 

This is my draft with example for valid/invalid hashes. 

https://github.com/suculent/Arduino/tree/suculent-esp8266httpupdate%2Bsha256?files=1

suculent, 2018-05-29:
All the changes are in following commit:
https://github.com/suculent/Arduino/commit/b53094633f971e8c74fdf122d6784297e5905260

But the question rather stands, if it's wise to include respective 3rd party code (sha256.c) and cause higher memory requirements to everyone just because of my personal requirements. 

In my opinion however, it would be wise to support stronger hash than MD5 and also provide SHA-256 functions globally in case it's not already there and I've just missed it.

earlephilhower, 2018-05-29:
There are sha256 methods in axTLS and BearSSL (and probably in ROM somewhere, too), so you don't really need to pull in another code source for the hashing, I think.  As to the memory requirement, maybe the best thing to do is subclass the updater and override the hashing so that way it won't pull in the code unless you instantiate the sha256 hashing object.  Or, refactor the updater to use a callback or take another object to perform the operation.  You can make the default callback call MD5 to allow existing code to run w/o modification.

suculent, 2018-05-29:
I didn't want to break existing Updater API, however I don't like the md5 being explicitly in method names. If it was hash, it would make sense to subclass it and implement variants (MD5, SHA-256, etc...)

That's why I've added the setHash (etc.) methods that use MD5 and/or SHA-256 based on length (and giving space to deprecate public MD5 methods in future). The SHA object should be instantiated only in respective case already.

earlephilhower, 2018-05-29:
> That's why I've added the setHash (etc.) methods that use MD5 and/or SHA-256 based on length (and giving space to deprecate public MD5 methods in future). The SHA object should be instantiated only in respective case already.

@suculent, I thought you were concerned about PROGMEM space, not RAM.  The problem with having an if/else branch there is that the compiler can't know you're using one or the other so will need to instantiate both codepaths.  It's probably not a big deal, but there were howls of pain when the stdlib went up a few KB in ROM to support printf(%f)/etc. a few releases ago so be warned. :)

You're right about the API.  All these hashes I've seen have a "start()", "while (dataavailable) feed()", "result()".  In BearSSL there are something like 10 different hash options (to support certificate signing) and they all have the same "vtable" entries like above.

suculent, 2018-05-29:
I’ve been expecting all of this. The thing is, I am more a designer and high-level programmer and I have no experience subclassing in C++ at all. My scene was mostly Objective-C and Swift in the means of OOP where this would be a piece od cake for me (without using Google of course).

I’m not afraid of that, just that proposed solution suits my needs and I was expecting to have it rejected in this state. I was interested in how to do it properly but all I can do now is have a C++ class being a proper lib for Arduino which is a great help to my project tree. 

Odesláno z iPhonu

suculent, 2018-05-29:
One more thing: where xactly does the compiler know?

Not a pre-processor #define. 

I would expect to have a heterogeneous container like (id) in Obj-C or Any in Swift where you can point it to any object conforming respective space and assigning later lazily. 

I need to know how to instantiate subclasses based on array of static callbacks that will check if the input conforms to respective “plugin” format. 

Class Hash would call those callbacks in order to find out, which subclass is appropriate to be instantiated to process such data. Like a plugins. There would be MD5 plugin, SHA-256 plugins pointing to axtls or mbedtls or bear depending on symbol definedness. Ugh. This is so big spine it could be implemented throughout. 

But my business is the https://thinx.cloud IoT network where we do not interfere users code in order to make it work. 

Odesláno z iPhonu

earlephilhower, 2018-05-29:
Not rejecting, only noting!  Haven't looked through the code yet.

The compiler knows using static graph analysis.  If there is a function call anywhere in the graph, that function better be linked in or you'll have an unresolved global in the final pass.

````
if (mode==md5) {
  return CallMD5();
} else if (mode==sha256) {
  return CallSHA256();
}
````
has to have both MD5 and SHA256 included unless mode is a constant (that the compiler can prove by a local "const int mode=md5" or a #define).

Your plugin talk sounds just like subclassing a virtual base class, but if it has anything like the above code it'll end up including both code paths, of course.  Two separate subclasses, `class HashSHA256 : public Hash { virtual start/feed/result override; }` and `class HashMD5 : public Hash { virtual start/feed/result override;  }` would suffice, and then the users of the hashing function would just take a `class Hash *` that you pass in somewhere.

suculent, 2018-05-29:
Well, my current solution works for me only and therefore I do consider it rejected, until I’ll be able to do this Dependency Injection makeover. The pattern is not common in Arduino and will cause a lot of buzz and misunderstanding, but makes sense. There will be hashing functions deprecating and coming over next years... anyone could add s blockchain hash. 

Anyway, it’s not on my public roadmap now. 

Odesláno z iPhonu

suculent, 2018-05-31:
@earlephilhower Would you be able to prepare an empty architecture example for this Updater update? I will fill the implementations and link axtls/bearssl when available, but I'm not able to design the subclassing/builder interface barebone properly.

d-a-v, 2018-05-31:
@suculent here are details from what @earlephilhower is thinking about:

Hash.h:
```
class Hash
{
    virtual void start () = 0;
    virtual void feed (const void* data, size_t size) = 0;
    virtual const char* result (size_t* size) = 0;
};
```

MD5Hash.h:
```
class MD5Hash: public Hash
{
    virtual void start () override;
    virtual void feed (const void* data, size_t size) override;
    virtual const char* result (size_t* size) override;
};
```

SHA256Hash.h: same as MD5Hash.h

MD5Hash.cpp and SHA256Hash.cpp both implement their respective class.
```
#include <MD5Hash.h>

void MD5Hash::start ()
{
    // code
}
...
```

disclaimer: this is c/c++ mix, @devyte may propose a templated view of the thing, or a more c++ containered way of dealing with args.

devyte, 2018-06-02:
Pretty please, no more virtual inheritance, the ESP's mem can't take it any more.
Similar objects, no virtualness. Then, template whichever objects, methods, or functions that either instantiate a hash object or receive one as argument.

earlephilhower, 2019-07-25:
@suculent, have you been following the repo?  We now have cryptographically signed binary support which goes one step better than a simple SHA256 and I think solves your main concern.

Can you give a look, and if it looks usable for your use case (others seem to be using already as we've had some really cool PRs to make it standard compliant, even) then could you close this issue?

suculent, 2019-07-25:
Sure, thanks for heads up. I'm now digging into some modem libs so I haven't notice there's 2.5.2 already out. Closing.

closed 2019-07-25 by suculent