esp8266/Arduino #4776: use UDP to send data to iOS and data's lost sometimes

tunted, 2018-05-30:
I try to use ESP8266 to send a data of about 40 bytes to iOS through a UDP port. At the beginning, everything works fine, after sometimes data is lost. This issue does not occur very often but it makes my device not reliable.
Does any body meet this problem?
Thank you for your time.

earlephilhower, 2018-05-30:
UDP stands for "Unreliable Datagram Protocol."  Emphasis on **unreliable**.  There is no retry on error or missed packets.  Your app **must** handle losses gracefully if you choose to use UDP.

If you want reliable, please use a real TCP link.

I'm not seeing a problem here.  If you're getting massive loss, sure (but you will need to post a MCVE sketch), but you've said it's very infrequent and not provided a way to reproduce it.

Closing as not a problem.

closed 2018-05-30 by earlephilhower

tunted, 2018-05-30:
Sorry for not explaining clearly about my problem.
I use UDP library and I think it's has some issues here.
When iOS do not receive the packet, I tried to send packet again and again and it still doesn't come to iOS device.
Packet will reach to device if I try more times, I don't even know why.
I think that the problem may come from iOS system