esp8266/Arduino #4777: Exception 29 NodeMCU v2

dbischoff777, 2018-05-30:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				NodeMCU v2 ESP8266
- Core Version:			[2.4.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[NodeMCU 1.0 ESP-12E]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I'm running the Sketch below. After a fresh Flash of besaid Sketch everything works just fine. As soon as powersupply gets disconnected und reconnected again i get Exception 29 at the exact same point over and over until i flash the Sketch anew.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
#include <Wire.h>
#include <ESP8266WebServer.h>
#include <WiFiClient.h>
#include <ESP8266WiFi.h>
#include <PCF8574.h>
#include "LEDController.h"

// ID und Passwort des WiFi Netzes
const char* ssid = "dlink";
const char* password = "MJojij123321";

// Web Server des ESP initialisieren
ESP8266WebServer server(80);
// Ein Objekt vom Typ PCF8574 mit Namen PCF_20 erstellen -> PCF_20(0x20) -> Konstruktor
PCF8574 PCF_20(0x20);
PCF8574 PCF_21(0x21);
PCF8574 PCF_22(0x22);
PCF8574 PCF_23(0x23);
PCF8574 PCF_24(0x24);
PCF8574 PCF_25(0x25);
PCF8574 PCF_26(0x26);
PCF8574 PCF_27(0x27);

LEDController controller;
  
// Funktion die abgerufen wird sobald der Nutzer den ESP per fester IP ansteuert
void handleRoot() {
  // Sende dem Nutzer eine Nachricht sollte der Aufruf erfolgreich sein
  server.send(200, "text/plain", String("Hello from esp8266! Usage: navigate to") +
              String(" /setleds?led1=XX&led2=YY changing XX/YY to ON or OFF."));
}

// Funktion zum schalten der LED's wenn /setleds per URL genutzt wird
void handleSetLeds() {

[... 239 lines cut ...]
```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 23 results
0x4020f1ac: glue2esp_ifup at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 757 (discriminator 1)
0x4021681b: ip_chksum_pseudo at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/inet_chksum.c line 542
0x40212d1f: udp_sendto_if_src at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/udp.c line 463
0x4020f25d: netif_sta_status_callback at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 303
0x40212810: netif_set_ipaddr at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 711
0x4021229d: pbuf_free_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/pbuf.c line 1306
0x4021283e: netif_set_netmask at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 711
:  (inlined by) netif_set_addr_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/netif.c line 359
0x40214d60: dhcp_release_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/dhcp.c line 1314
0x40214db3: dhcp_coarse_tmr at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/dhcp.c line 423
0x402106d0: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x40100690: free at C:\Users\dbisc\OneDrive\Dokumente\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\umm_malloc/umm_malloc.c line 1739
0x4010100f: ppCalFrameTimes at ?? line ?
0x402106d0: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 165
0x402106de: cyclic_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 171
0x40104d7e: wdt_feed at ?? line ?
0x4021079c: sys_check_timeouts at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c line 351
0x4022ed54: ets_timer_handler_isr at ?? line ?
0x4022ed61: ets_timer_handler_isr at ?? line ?
0x4022eda6: ets_timer_handler_isr at ?? line ?
0x40208747: loop_task at C:\Users\dbisc\OneDrive\Dokumente\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/core_esp8266_main.cpp line 57
```

dbischoff777, 2018-05-30:
Update a few minutes later....

Changing lwIP from v2 Lower Memory to v1.4 Higher Bandwidth (Tools -> lwIP Variant) solved the issue. The exception is not thrown anymore. Regardless of the power supply being disc- and reconnected to the NodeMCU v2. The Sketch runs stable now. Hope this helps someone else...

peace

closed 2018-05-30 by dbischoff777

d-a-v, 2018-05-30:
2.4.1 has known issues with static IP. It would help if you tried latest git.