esp8266/Arduino #4781: How to use ESPAVRISP with USB instead of Wi-Fi

dev-ahmed-ibrahim, 2018-06-01:
I have implemented successfully the Wi-Fi programming to ATMEL AVR MCUs, but I would like to ask how to modify this library in order to support USB programming instead of Wi-Fi ?

Thanks in advance.

dev-ahmed-ibrahim, 2018-06-01:
I use nodemcu that has USB on board, how to modify this library to send programming orders through usb ?

me-no-dev, 2018-06-01:
what is the point? you can use USB UART and program it the normal way

dev-ahmed-ibrahim, 2018-06-01:
I think that there is confused you, my point is I would like to program ATMEL ATmega16 on breadboard using ESPAVRISP but using usb cable to nodemcu not Wi-Fi , can I proceed with this, and what modifications needed to this library that supports only Wi-Fi programming.

WereCatf, 2018-06-01:
@dev-ahmed-ibrahim It's you who is confused. It's literally just a USB-to-serial - port, you can't do anything else with the USB-port.

WereCatf, 2018-06-01:
@dev-ahmed-ibrahim The serial-port on the ESP8266 is connected to the USB-port, so any communications you have with the ESP will show up on the serial-port. You'd have to modify the code to use a software-serial on some other pins and connect those to the ATmega16 for programming instead.

me-no-dev, 2018-06-01:
Also for you to use the USB to program the AVR, you will need to figure out a way to not reset the ESP when the UART is open by AVRDUDE, then you need to detect that USB connection is established so you can in turn manually reset the AVR and then figure out how to push the data to it (Serial or ICSP).
Overall not really making much sense. You will be better off desoldering the ESP8266 all together and just connecting the AVR to the USB Serial.

closed 2018-06-02 by devyte