esp8266/Arduino #4782: SoftwareSerial

MarcFinns, 2018-06-01:
### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.]
- Development Env:		[Arduino IDE 1.8.5]
- Operating System:		[WindowS]

### Settings in IDE

- Module:  			[Nodemcu AND OTHERS]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

The built-in SoftwareSerial library cannot be overridden by a newer version of the same library, implementing additional methods. This gives compile level errors.
The only way to work around is to tweak the ESP8266 install and delete the library.
I did not test every version, but it wasn't the case in the past.

devyte, 2018-06-07:
The SoftwareSerial library is not built-in, it is pulled from [that ]( https://github.com/plerup/espsoftwareserial) repo, and from what [I can tell](https://github.com/esp8266/Arduino/blob/73d36bbc80f8659488eacb3776e869ca96d2d2d5/package/build_boards_manager_package.sh#L64), we're using version 3.4.1, which is the latest release.

If what you need is to tweak the SoftwareSerial lib, then you can do that by
- installing the git version of this repo
- pulling in the SoftwareSerial manually (git clone into the libraries dir, make sure to clone into SoftwareSerial dir and not the default)
- making changes to your local copy of the SoftwareSerial lib

If what you want is to propose changes to the SoftwareSerial lib, then you should do that in the other repo, then after a new release of the lib is out request a version update here.

Closing.

closed 2018-06-07 by devyte

MarcFinns, 2018-06-07:
I can leave with a workaround, HOWEVER I would invite you to compile the examples provided with the SoftwareSerial library with the version you embed in the ESP8266 distribution. 
On Windows and Linux it will say that certain methods - introduced recently - are not implemented. for example void enableIntTx(bool on). So it is not the latest version. My 2c