esp8266/Arduino #4783: delay()  void promisc_cb(uint8_t *buf, uint16_t len)

stern0m1, 2018-06-01:
delay() in promiscuous callback causes a crash.

 wifi_set_promiscuous_rx_cb(promisc_cb);
void promisc_cb(uint8_t *buf, uint16_t len) {
delay(1)
}

Exception (9):
epc1=0x40104de0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000003 depc=0x00000000

ctx: sys 
sp: 3ffffd60 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff00:  00000000 00000000 00000020 40100ec6  
3fffff10:  4010500c 00234f61 3ffee9e0 00000000  
3fffff20:  3ffee320 3ffee9e0 3ffee8fc 00000000  
3fffff30:  00000000 00234e29 40201340 00000001  
3fffff40:  3ffee9e0 00000000 3fff06ae 3ffefc20  
3fffff50:  3ffee938 3ffee938 3ffefc20 40202485  
3fffff60:  0000002a 00000029 3ffee320 402012c0  
3fffff70:  40228f88 3ffee2f8 3ffee320 3ffe91a8  
3fffff80:  3fffdcc0 00000000 3fff0684 40221930  
3fffff90:  40221b1e 3fffdab0 3ffeebac 40202fdb  
3fffffa0:  3ffe91a8 40000f49 3fffdab0 40000f49  
<<<stack<<<

devyte, 2018-06-02:
Callbacks like in this case don't execute in the continuation context, but in the sys context (I.e. inside the sdk), and yield/delay can't be used in the sys context.
Closing.

closed 2018-06-02 by devyte

stern0m1, 2018-06-03:
Is there any workaround where I can get a delay without a timer?
Thanks